zephyrproject-rtos/zephyr #50947: stm32 static IPv4 networking in smp_svr sample application does not seem to work until a ping is received

nordicjm, 2022-10-04:
**Describe the bug**
Testing `samples/subsys/mgmt/mcumgr/smp_svr` for the `nucleo_f746zg` board with the `overlay-udp.conf` overlay enabled (note: there is no partition map for this board, which needs to be setup) results in an image that, once flashed, does work, but does not accept UDP connections until the device is pinged.
Output on board after boot and sending a command via mcumgr (using `mcumgr --conntype udp --connstring=[192.168.1.220]:1337 fs upload /tmp/file /lfs1/testfile`):
```
uart:~$ *** Booting Zephyr OS build zephyr-v3.2.0-115-gee5a56aad1ab  ***
[00:00:01.533,000] <dbg> net_ethernet: ethernet_init: (main): Initializing Ethernet L2 0x20016c9c for iface 0x20012714
[00:00:01.539,000] <inf> net_config: Initializing network
[00:00:01.539,000] <inf> net_config: Waiting interface 1 (0x20012714) to be up...
[00:00:02.029,000] <dbg> net_ethernet: carrier_on_off: (sysworkq): Carrier ON for interface 0x20012714
[00:00:02.029,000] <dbg> net_ethernet: ethernet_fill_header: (sysworkq): iface 0x20012714 src 02:80:E1:B6:33:E0 dst 33:33:00:00:00:14[00:00:02.029,000] <dbg> net_ethernet: ethernet_fill_header: (sysworkq): iface 0x20012714 src 02:80:E1:B6:33:E0 dst 33:33:00:00:00:14[00:00:02.030,000] <dbg> net_ethernet: ethernet_fill_header: (sysworkq): iface 0x20012714 src 02:80:E1:B6:33:E0 dst 33:33:FF:B6:33:E4[00:00:02.030,000] <dbg> net_ethernet: ethernet_fill_header: (sysworkq): iface 0x20012714 src 02:80:E1:B6:33:E0 dst 33:33:00:00:00:04[00:00:01.533,000] <dbg> net_ethernet: ethernet_init: (main): Initializing Ethernet L2 0x20016c9c for iface 0x20012714
[00:00:01.539,000] <inf> net_config: Initializing network
[00:00:01.539,000] <inf> net_config: Waiting interface 1 (0x20012714) to be up..[00:00:02.030,000] <inf> net_config: Interface 1 (0xp.
[00:00:02.029,000] <dbg> net_ethernet: carrier_on_off: (sysworkq): Carrier ON for interface 0x2001271[00:00:02.030,000] <inf> net_co04
[00:00:02.029,000] <dbg> net_ethernet: ethernet_fill_header: (sysworkq): iface 0x20012714 src 02:80:E1:B6:33:E0 dst 33:33:00:00:00:144
[00:00:02.029,000] <dbg> net_ethernet: ethernet_fill_header: (sysworkq): iface 0x20012714 src 02:80:E1:B6:33:E0 dst 33:33:00:00:00:144
[00:00:02.030,000] <dbg> net_ethernet: ethernet_fill_header: (sysworkq): iface 0x20012714 src 02:80:E1:B6:33:E0 dst 33:33:FF:B6:33:E0[00:00:02.031,000] <inf> littlefs: FS at flash-controller@40023c00:0x10000 is 2 0x8000-byte blocks with 512 cycle
[00:00:02.031,000] <inf> littlefs: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
4
[00:00:02.030,000] <dbg> net_ethernet: ethernet_fill_header: (sysworkq): iface 0x20012714 src 02:80:E1:B6:33:E0 dst 33:33:00:00:00:0}[00:00:02.031,000] <wrn> littlefs: can't mount (LFS -84); formatting
[00:00:02.226,000] <inf> net_config: IPv6 address: 2001:db8::1
4
[00:00:02.030,000] <inf> net_config: Interface 1 (0x20012714) coming u[00:00:02.226,000] <inf> net_config: IPv6 address: 2001:db8::1
p
[00:00:02.030,000] <inf> net_config: IPv4 address: 192.168.1.22[00:00:02.429,000] <inf> littlefs: /lfs1 mounted
0
[00:00:02.030,000] <dbg> net_ethernet: ethernet_fill_header: (main): iface 0x20012714 src 02:80:E1:B6:33:E0 dst 33:33:00:00:00:16 tyd[00:00:02.429,000] <inf> smp_sample: build time: Oct  4 2022 12:58:36
[00:00:02.430,000] <inf> smp_udp: Started (IPv4)
4
[00:00:02.031,000] <dbg> net_ethernet: ethernet_fill_header: (main): iface 0x20012714 src 02:80:E1:B6:33:E0 dst 33:33:FF:00:00:01 ty)4
[00:00:02.031,000] <inf> littlefs: LittleFS version 2.5, disk version 2.0
[00:00:02.031,000] <inf> littlefs: FS at flash-controller@40023c00:0x10000 is 2 0x8000-byte blocks with 512 cycle
[00:00:02.031,000] <inf> littlefs: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
[00:00:02.031,000] <err> littlefs: WEST_TOPDIR/modules/fs/littlefs/lfs.c:1234: Corrupted dir pair at {0x0, 0x1}
[00:00:02.031,000] <wrn> littlefs: can't mount (LFS -84); formatting
[00:00:02.226,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:02.226,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:02.429,000] <inf> littlefs: /lfs1 mounted
[00:00:02.429,000] <inf> smp_udp_sample: Network connected
[00:00:02.429,000] <inf> smp_sample: build time: Oct  4 2022 12:58:36
[00:00:02.430,000] <inf> smp_udp: Started (IPv4)
[00:00:02.430,000] <inf> smp_udp: Started (IPv6)
uart:~$ [00:00:03.030,000] <dbg> net_ethernet: ethernet_fill_header: (sysworkq): iface 0x20012714 src 02:80:E1:B6:33:E0 dst 33:33:004[00:00:04.031,000] <dbg> net_ethernet: ethernet_fill_header: (sysworkq): iface 0x20012714 src 02:80:E1:B6:33:E0 dst 33:33:00:00:00:04[00:00:03.030,000] <dbg> net_ethernet: ethernet_fill_header: (sysworkq): iface 0x20012714 src 02:80:E1:B6:33:E0 dst 33:33:00:00:00:04[00:00:04.031,000] <dbg> net_ethernet: ethernet_fill_header: (sysworkq): iface 0x20012714 src 02:80:E1:B6:33:E0 dst 33:33:00:00:00:04uart:~$ [00:00:21.026,000] <dbg> net_ethernet: ethernet_recv: (rx_q[0]): iface 0x20012714 src CC:96:A0:4B:AB:D5 dst FF:FF:FF:FF:FF:F0[00:00:21.026,000] <dbg> net_ethernet: ethernet_recv: (rx_q[0]): ARP packet from CC:96:A0:4B:AB:D5 received
[00:00:21.026,000] <dbg> net_ethernet: ethernet_recv: (rx_q[0]): iface 0x20012714 src CC:96:A0:4B:AB:D5 dst FF:FF:FF:FF:FF:FF type 00[00:00:21.026,000] <dbg> net_ethernet: ethernet_recv: (rx_q[0]): ARP packet from CC:96:A0:4B:AB:D5 received
uart:~$ [00:00:27.276,000] <dbg> net_ethernet: ethernet_recv: (rx_q[0]): iface 0x20012714 src CC:96:A0:4B:AB:D5 dst FF:FF:FF:FF:FF:F0[00:00:27.276,000] <dbg> net_ethernet: ethernet_recv: (rx_q[0]): ARP packet from CC:96:A0:4B:AB:D5 received
[00:00:27.276,000] <dbg> net_ethernet: ethernet_recv: (rx_q[0]): iface 0x20012714 src CC:96:A0:4B:AB:D5 dst FF:FF:FF:FF:FF:FF type 00[00:00:27.276,000] <dbg> net_ethernet: ethernet_recv: (rx_q[0]): ARP packet from CC:96:A0:4B:AB:D5 received
```
(Nothing from the mcumgr client ever appears) Output when I then ping the device (First 3 pings from host time out):
```
uart:~$ [00:01:19.861,000] <dbg> net_ethernet: ethernet_recv: (rx_q[0]): iface 0x20012714 src 0C:37:96:21:F9:5D dst FF:FF:FF:FF:FF:F0[00:01:19.861,000] <dbg> net_ethernet: ethernet_recv: (rx_q[0]): ARP packet from 0C:37:96:21:F9:5D received
[00:01:19.861,000] <dbg> net_ethernet: ethernet_fill_header: (rx_q[0]): iface 0x20012714 src 02:80:E1:B6:33:E0 dst 0C:37:96:21:F9:5D4[00:01:19.863,000] <dbg> net_ethernet: ethernet_recv: (rx_q[0]): iface 0x20012714 src 0C:37:96:21:F9:5D dst 02:80:E1:B6:33:E0 type 08[00:01:19.863,000] <dbg> net_ethernet: ethernet_ll_prepare_on_ipv4: (rx_q[0]): Found ARP entry, sending pkt 0x20020614 to iface 0x204[00:01:19.863,000] <dbg> net_ethernet: ethernet_fill_header: (rx_q[0]): iface 0x20012714 src 02:80:E1:B6:33:E0 dst 0C:37:96:21:F9:5D4[00:01:19.861,000] <dbg> net_ethernet: ethernet_recv: (rx_q[0]): iface 0x20012714 src 0C:37:96:21:F9:5D dst FF:FF:FF:FF:FF:FF type 00[00:01:19.861,000] <dbg> net_ethernet: ethernet_recv: (rx_q[0]): ARP packet from 0C:37:96:21:F9:5D received
[00:01:19.861,000] <dbg> net_ethernet: ethernet_fill_header: (rx_q[0]): iface 0x20012714 src 02:80:E1:B6:33:E0 dst 0C:37:96:21:F9:5D4[00:01:19.863,000] <dbg> net_ethernet: ethernet_recv: (rx_q[0]): iface 0x20012714 src 0C:37:96:21:F9:5D dst 02:80:E1:B6:33:E0 type 08[00:01:19.863,000] <dbg> net_ethernet: ethernet_ll_prepare_on_ipv4: (rx_q[0]): Found ARP entry, sending pkt 0x20020614 to iface 0x204[00:01:19.863,000] <dbg> net_ethernet: ethernet_fill_header: (rx_q[0]): iface 0x20012714 src 02:80:E1:B6:33:E0 dst 0C:37:96:21:F9:5D4uart:~$ 
```
Output when I then send the same mcumgr command after the ping:
```
uart:~$ [00:02:12.034,000] <dbg> net_ethernet: ethernet_recv: (rx_q[0]): iface 0x20012714 src 0C:37:96:21:F9:5D dst 02:80:E1:B6:33:E0[00:02:12.037,000] <dbg> net_ethernet: ethernet_ll_prepare_on_ipv4: (mcumgr smp): Found ARP entry, sending pkt 0x20020614 to iface 04[00:02:12.037,000] <dbg> net_ethernet: ethernet_fill_header: (mcumgr smp): iface 0x20012714 src 02:80:E1:B6:33:E0 dst 0C:37:96:21:F94[00:02:12.034,000] <dbg> net_ethernet: ethernet_recv: (rx_q[0]): iface 0x20012714 src 0C:37:96:21:F9:5D dst 02:80:E1:B6:33:E0 type 00[00:02:12.037,000] <dbg> net_ethernet: ethernet_ll_prepare_on_ipv4: (mcumgr smp): Found ARP entry, sending pkt 0x20020614 to iface 04[00:02:12.037,000] <dbg> net_ethernet: ethernet_fill_header: (mcumgr smp): iface 0x20012714 src 02:80:E1:B6:33:E0 dst 0C:37:96:21:F94uart:~$ 
```

**Expected behavior**
For static IPv4 networking to work out of the box.

**Impact**
Annoying to me, I imagine this would be a showstopper for anyone wanting to use this board however.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.15.0
 - Commit SHA or Version used: 0ec243b65ee0ae07cf8f4bec897be0d7955fde25

carlescufi, 2022-10-04:
cc @rlubos

nordicjm, 2022-10-04:
Just had another check and does seem to be STM32 specific, seems the lag on the nordic was caused by a test fs_mgmt change, reverted that and now works instantly on nordic, long delay on stm32, sorry @erwango

nordicjm, 2022-10-04:
Issue seems to be MAC address of the device, the STM32 ethernet MAC address is changing on each boot because of `CONFIG_ETH_STM32_HAL_RANDOM_MAC`:
```
arp -a
? (192.168.1.220) at <incomplete> on enp0s20f0u2u1i5
arp -a
? (192.168.1.220) at 02:80:e1:1b:55:2a [ether] on enp0s20f0u2u1i5
arp -a
? (192.168.1.220) at 02:80:e1:57:6b:45 [ether] on enp0s20f0u2u1i5
```
Should this not be using some internal ID use at the MAC address? Doesn't seem very flexible to either have it random or use static Kconfig values for 3 bytes

erwango, 2022-12-12:
> Should this not be using some internal ID use at the MAC address? Doesn't seem very flexible to either have it random or use static Kconfig values for 3 bytes

Looking to `ethernet.yaml`, here are the possible settings that are widly used among vendors:
```
    local-mac-address:
      type: uint8-array
      description: Specifies the MAC address that was assigned to the network device
    zephyr,random-mac-address:
      type: boolean
      description: |
        Use a random MAC address generated when the driver is initialized.
        Note that using this choice and rebooting a board may leave stale
        MAC address in peers' ARP caches and lead to issues and delays in
        communication.  (Use "ip neigh flush all" on Linux peers to clear
        ARP cache.)

        It is driver specific how the OUI octets are handled.

        If set we ignore any setting of the local-mac-address property.
```
So, this isn't STM32 specific.
I'm ok, to provide a uid based configuration, but I'd like @tbursztyka confirming this is the good direction

closed 2023-01-09 by carlescufi