zephyrproject-rtos/zephyr #50948: SSD1306+lvgl sample fails to display

zpm1066, 2022-10-04:
**Describe the bug**
I'm able to build the sample "zephyrproject/zephyr/samples/subsys/display/lvgl" for a nrf52840 board using an overlay. The program builds fine but a connected generic OLED SSD1306 128x64 display (also a ssd1306 Adafruit FeatherWing OLED - 128x64) fails to display anything. 
The sample code hasn't been modified.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 nrf52840 board (particle_xenon)
 
 - What have you tried to diagnose or workaround this issue?
 In debug, it appears that the program crashes within lv_task_handler().
 
 - Is this a regression? 
 Unable to determine, as I cannot get the SSD1306 + lvgl to work with several Zephyr versions.

**To Reproduce**
Steps to reproduce the behavior:
1. Use the sample code "zephyrproject/zephyr/samples/subsys/display/lvgl" as is. 
2. Use the proj.conf and particle_xenon.overlay below.

particle_xenon.overlay:
/ {
	chosen {
		zephyr,display = &ssd1306;
	};
};

&pinctrl {
		i2c0_default: i2c0_default {
			group1 {
				psels = <NRF_PSEL(TWIM_SDA, 0, 26)>,
					<NRF_PSEL(TWIM_SCL, 0, 27)>;
			};
		};
	
		i2c0_sleep: i2c0_sleep {
			group1 {
				psels = <NRF_PSEL(TWIM_SDA, 0, 26)>,
					<NRF_PSEL(TWIM_SCL, 0, 27)>;
				low-power-enable;
			};
		};
};

&gpio0{
	status = "okay";
};

&gpio1{
	status = "okay";
};

&i2c0 {
	compatible = "nordic,nrf-twim";
	status = "okay";
	pinctrl-0 = <&i2c0_default>;
	pinctrl-1 = <&i2c0_sleep>;
    pinctrl-names = "default", "sleep";

	status = "okay";
	ssd1306: ssd1306@3c {
		compatible = "solomon,ssd1306fb";
		reg = <0x3c>;
		label = "SSD1306";
		width = <128>;
		height = <64>;
		segment-offset = <0>;
		page-offset = <0>;
		display-offset = <0>;
		multiplex-ratio = <63>;
		segment-remap;
		com-invdir;
		prechargep = <0x22>;
	};
};

proj.conf:

CONFIG_MAIN_STACK_SIZE=2048

CONFIG_I2C=y
CONFIG_I2C_NRFX=y
CONFIG_LOG=y

CONFIG_DISPLAY=y
CONFIG_DISPLAY_LOG_LEVEL_ERR=y

CONFIG_SSD1306=y
CONFIG_SSD1306_DEFAULT=y

CONFIG_LVGL=y
CONFIG_LV_MEM_CUSTOM=y
CONFIG_LV_USE_LOG=y
CONFIG_LV_USE_LABEL=y
CONFIG_LV_USE_BTN=y
CONFIG_LV_USE_IMG=y
CONFIG_LV_FONT_MONTSERRAT_14=y

CONFIG_LV_Z_BITS_PER_PIXEL=1
CONFIG_LV_Z_MEM_POOL_NUMBER_BLOCKS=8
CONFIG_APP_LINK_WITH_LVGL=y

*Expected behavior**
I expected the lvgl sample to work with the ssd1306 display and show the "Hello world!" text in the middle of the display.

**Impact**
Unable to use SSD1306 on several key projects with lvgl.

**Logs and console output**
Program builds without any errors.

**Environment (please complete the following information):**
macOS Big Sur & Catalina
Zephyr RTOS v3.1 and 3.2

**Additional context**
The same SSD1306 display with exactly the same connections works with nrf52840 and other boards when using SSD1306 with cfb.

zpm1066, 2022-10-04:
@laurenmurphyx64, @henrikbrixandersen, @jfischer-no, @brgl 
I'm also going to create an account at lvgl.com to see if they can also investigate, as I don't see anything wrong with the Zephyr code. 

I have several types of displays (ST7789v, ili9341 - 1.14" to 2.8") that all work just fine with Zephyr + lvgl. SSD1306 works fine with Zephyr on it's own. There may be a SSD1306 binding issue with lvgl.

I'll be glad to assist with further debugging of SSD1306/lvgl. Thanks.

jfischer-no, 2022-10-06:
> Zephyr RTOS v3.1 and 3.2

I cannot confirm, works fine for me using `west build -b nrf52840dk_nrf52840 samples/subsys/display/lvgl/ -- -DSHIELD=ssd1306_128x64`. Moving to discussion since it is likely configuration issue.