zephyrproject-rtos/zephyr #50970: SAME54_xpro network driver not attached

antonaltx, 2022-10-04:
**Describe the bug**
GSAM interface did not work in net samples

Platform :Win10Pro , blinky sample works fine
west build -b always atsame54_xpro samples\net\dhcpv4_client
west flash

after boot apps on target:
uart:~$ net iface
uart:~$

No interfaces shown .
All commits from "Support Ethernet on SAM E5x #23300" present in code.

laurenmurphyx64, 2022-10-04:
Hi! Thanks for filing an issue, could you fill in more information following the template [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/bug_report.md)?

cfriedt, 2022-10-04:
@antonaltx - unsolicited tip, but you can embed code / preformatted text between an opening and closing sequence of 3 back-ticks

See https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code

antonaltx, 2022-10-04:
Output from target app:
*** Booting Zephyr OS build zephyr-v3.2.0-132-g6bbd2b2d06f3  ***

[00:00:00.000,000] <err> net_if: There is no network interface to work with!
[00:00:00.006,000] <inf> net_dhcpv4_client_sample: Run dhcpv4 client
uart:~$

nandojve, 2022-10-05:
Hi @antonaltx ,

I don't have this board. I tested using [sam_v71_xult](https://docs.zephyrproject.org/latest/boards/arm/sam_v71_xult/doc/index.html) board with [TagoIO IoT Cloud HTTP](https://docs.zephyrproject.org/latest/samples/net/cloud/tagoio_http_post/README.html) Sample. The result log you can see below. I recommend you compare `prj.conf` because sometimes maybe missing some config.

```
[00:00:00.000,000] <inf> i2c_sam_twihs: Device i2c@40060000 initialized
[00:00:00.000,000] <inf> i2c_sam_twihs: Device i2c@40018000 initialized
uart:~$ *** Booting Zephyr OS build zephyr-v3.2.0-238-gb73286ca8355  ***
[00:00:00.255,000] <err> mdio_sam: transfer timedout mdio
[00:00:00.510,000] <err> mdio_sam: transfer timedout mdio
[00:00:00.765,000] <err> mdio_sam: transfer timedout mdio
[00:00:01.020,000] <err> mdio_sam: transfer timedout mdio
[00:00:01.021,000] <inf> eth_sam: MAC: XX:XX:XX:XX:XX:XX
[00:00:01.021,000] <inf> eth_sam: Queue 0 activated
[00:00:01.021,000] <inf> eth_sam: Queue 1 set to idle
[00:00:01.021,000] <inf> eth_sam: Queue 2 set to idle
[00:00:01.026,000] <inf> net_config: Initializing network
[00:00:01.026,000] <inf> net_config: Waiting interface 1 (0x20400670) to be up...
[00:00:03.628,000] <inf> phy_mii: PHY (0) Link speed 100 Mb, full duplex
[00:00:03.628,000] <inf> eth_sam: Link up
[00:00:03.628,000] <inf> net_config: Interface 1 (0x20400670) coming up
[00:00:03.628,000] <inf> net_config: Running dhcpv4 client...
[00:00:05.917,000] <inf> net_dhcpv4: Received: 192.168.178.32
[00:00:05.917,000] <inf> net_config: IPv4 address: 192.168.178.32
[00:00:05.917,000] <inf> net_config: Lease time: 864000 seconds
[00:00:05.917,000] <inf> net_config: Subnet: 255.255.255.0
[00:00:05.917,000] <inf> net_config: Router: 192.168.178.1
[00:00:05.917,000] <inf> tagoio_http_post: TagoIO IoT - HTTP Client - Temperature demo
[00:00:05.917,000] <inf> tagoio_http_post: Temp: 87
[00:00:16.238,000] <inf> tagoio_http_post: Temp: 39
[00:00:26.829,000] <inf> tagoio_http_post: Temp: 58
uart:~$
```

antonaltx, 2022-10-05:
Hi @nandojve ,
I have try different platform(STM32H743) and it works fine.
Problem is for same54_xpro. It does not see network iface. It is for tagoio_http_post.
I did not change any prj.conf .

Errors:

 *** Booting Zephyr OS build zephyr-v3.2.0-132-g6bbd2b2d06f3  ***

[00:00:00.000,000] <err> net_if: There is no network interface to work with!
[00:00:00.006,000] <inf> net_config: Initializing network
[00:00:00.006,000] <inf> net_config: Waiting interface -1 ((nil)) to be up...
uart:~$ net iface
uart:~$ device list
devices:
- eic@40002800 (READY)
- gpio@41008180 (READY)
- gpio@41008100 (READY)
- gpio@41008080 (READY)
- gpio@41008000 (READY)
- random@42002800 (READY)
- sercom@41012000 (READY)
[00:00:30.007,000] <err> net_config: Timeout while waiting network interface
[00:00:30.007,000] <err> net_config: Network initialization failed (-115)
[00:00:30.007,000] <inf> tagoio_http_post: TagoIO IoT - HTTP Client - Temperature demo
[00:00:30.007,000] <inf> tagoio_http_post: Temp: 81
[00:00:39.308,000] <err> tagoio_http_post: Could not resolve dns, error: -3
[00:00:39.308,000] <inf> tagoio_http_post: No connection available.
uart:~$
[00:00:49.308,000] <inf> tagoio_http_post: Temp: 86
uart:~$

nandojve, 2022-10-05:
Would you mind to check this patch:

```
diff --git a/drivers/ethernet/Kconfig.sam_gmac b/drivers/ethernet/Kconfig.sam_gmac
index 7763292633..1ced23776f 100644
--- a/drivers/ethernet/Kconfig.sam_gmac
+++ b/drivers/ethernet/Kconfig.sam_gmac
@@ -6,7 +6,8 @@
 menuconfig ETH_SAM_GMAC
        bool "Atmel SAM Ethernet driver"
        default y
-       depends on DT_HAS_ATMEL_SAM_GMAC_ENABLED
+       depends on DT_HAS_ATMEL_SAM_GMAC_ENABLED || \
+                  DT_HAS_ATMEL_SAM0_GMAC_ENABLED
        select NOCACHE_MEMORY if ARCH_HAS_NOCACHE_MEMORY_SUPPORT
        select MDIO
        help
```

antonaltx, 2022-10-06:
Hi @nandojve,
Patch corrected ethernet on SAM54_xpro. I will test more with different samples.
But so far looks good.
Thank you.
Now I see correct output from tagoio_http_post on SAM54_xpro:
uart:~$ *** Booting Zephyr OS build zephyr-v3.2.0-132-g6bbd2b2d06f3  ***
[00:00:00.255,000] <err> mdio_sam: transfer timedout mdio
[00:00:00.510,000] <err> mdio_sam: transfer timedout mdio
[00:00:00.765,000] <err> mdio_sam: transfer timedout mdio
[00:00:01.020,000] <err> mdio_sam: transfer timedout mdio
[00:00:01.020,000] <inf> eth_sam: MAC: 02:.........................
[00:00:01.020,000] <inf> eth_sam: Queue 0 activated
[00:00:01.026,000] <inf> net_config: Initializing network
[00:00:01.026,000] <inf> net_config: Waiting interface 1 (0x200003c0) to be up...
[00:00:04.070,000] <inf> phy_mii: PHY (0) Link speed 100 Mb, full duplex
[00:00:04.070,000] <inf> eth_sam: Link up
[00:00:04.070,000] <inf> net_config: Interface 1 (0x200003c0) coming up
[00:00:04.070,000] <inf> net_config: Running dhcpv4 client...
[00:00:11.538,000] <inf> net_dhcpv4: Received: 192.168.1.13
[00:00:11.538,000] <inf> net_config: IPv4 address: 192.168.1.13

antonaltx, 2022-10-06:
tagoio_http_post: TagoIO IoT - HTTP Client  on SAME54_xpro works.

PR #51053 (2022-10-10): drivers: eth: sam0: Fix missing devicetree dependency
The #48905 introduce a regression on gmac ethernet driver. The GMAC driver matches two compatibles. This add missing compatible for SAM0 variant.

Fixes: #50970.
commit 5c484e28 (2022-10-10): drivers: eth: sam0: Fix missing devicetree dependency

The #48905 introduce a regression on gmac ethernet driver. The GMAC
driver matches two compatibles. This add missing compatible for SAM0
variant.

Fixes: #50970.
closed 2022-10-10 by MaureenHelm