zephyrproject-rtos/zephyr #50975: Add OpenSSF Scorecard Action

pnacht, 2022-10-04:
**Is your enhancement proposal related to a problem? Please describe.**
Open-source supply-chain attacks are [increasing every year][sonatype]. Beyond the infamous [SolarWinds][solarwinds] and [Codecov][codecov] attacks, there have also been multiple smaller but significant supply-chain disruptions such as [left-pad][leftpad],  [colors/faker.js][faker], [coa/rc][coa] and [ua-parser-js][ua].

**Describe the solution you'd like**
I'm working on behalf of Google and the [Open Source Security Foundation][ossf] (also a Linux Foundation project) to help essential open-source projects improve their supply-chain security. Given Zephyr's significant position in the RTOS space, the OpenSSF has identified it as one of the 100 most critical open-source projects.

Would you consider adopting an OpenSSF tool called [Scorecards][sc]? Scorecards runs dozens of automated security [checks][checks] to help maintainers better understand their project's supply-chain security posture. It is developed by the OpenSSF, [in partnership with GitHub][sc-blog].

I see Zephyr already has the OpenSSF Best Practices badge (gold, very nice!) and even created an entire [project](https://github.com/zephyrproject-rtos/zephyr/projects/7) a while ago to get it to silver. It even had a PR to improve an aspect of the project's posture based on Scorecards feedback a while back (#47074).

However, to further assist in this effort, the OpenSSF has also developed the [Scorecard GitHub Action][sc-gha]. It effectively tries to perform a continuous monitoring of the project's security posture. This is in no way a replacement of your current Best Practices badge; the Action simply tries to objectively inform the project of possible improvements to its security posture.

The Action is very lightweight and runs on every change to the repository's main branch. The results of its checks are available on the project's [security dashboard](https://github.com/zephyrproject-rtos/zephyr/security), and include suggestions on how to solve any issues (see examples below). The Action does not run or interact with any workflows, but merely parses them to identify possible vulnerabilities. This Action has been adopted by 1800+ projects already.

Would you be interested in a PR which adds this Action? Optionally, it can also publish your results to the OpenSSF REST API, which allows a [badge][badge] with the project's score to be added to its README.

![Code scanning dashboard with multiple alerts, including Code-Review and Token-Permissions][img-security]

![Detail of a Token-Permissions alert, indicating the specific file and remediation steps][img-detail]

[badge]: https://openssf.org/blog/2022/09/08/show-off-your-security-score-announcing-scorecards-badges/
[checks]: https://github.com/ossf/scorecard#scorecard-checks
[coa]: https://www.bleepingcomputer.com/news/security/popular-coa-npm-library-hijacked-to-steal-user-passwords/
[codecov]: https://blog.gitguardian.com/codecov-supply-chain-breach/
[faker]: https://www.bleepingcomputer.com/news/security/dev-corrupts-npm-libs-colors-and-faker-breaking-thousands-of-apps/
[leftpad]: https://qz.com/646467/how-one-programmer-broke-the-internet-by-deleting-a-tiny-piece-of-code/
[ossf]: https://openssf.org/
[sc]: https://github.com/ossf/scorecard
[sc-blog]: https://github.blog/2022-01-19-reducing-security-risk-oss-actions-opensff-scorecards-v4/
[sc-gha]: https://github.com/ossf/scorecard-action
[solarwinds]: https://en.wikipedia.org/wiki/2020_United_States_federal_government_data_breach
[sonatype]: https://www.sonatype.com/resources/state-of-the-software-supply-chain-2021
[ua]: https://www.bleepingcomputer.com/news/security/popular-npm-library-hijacked-to-install-password-stealers-miners/

[img-security]: https://user-images.githubusercontent.com/15221358/190184391-84ca1844-259a-4b3b-9c86-74adadbea7f1.png
[img-detail]: https://user-images.githubusercontent.com/15221358/190184600-ee8d3b39-077e-416a-8711-1b5fb01cf0b3.png

nashif, 2022-11-18:
> Would you be interested in a PR which adds this Action? Optionally, it can also publish your results to the OpenSSF REST API, which allows a [badge](https://openssf.org/blog/2022/09/08/show-off-your-security-score-announcing-scorecards-badges/) with the project's score to be added to its README.

sure, please submit a PR.

pnacht, 2022-11-18:
@nashif please see the linked PR #52370.

PR #52370 (2024-08-09): Add the OpenSSF Scorecard GitHub Action
Closes #50975.

As per the linked issue, this PR adds the OpenSSF Scorecard GitHub Action, which monitors the project's supply-chain security posture and suggests improvements through the Security Dashboard.

I've also taken the liberty of adding a badge to the project's README.md. However, this badge is strictly optional and I'll gladly remove it, just say the word!
closed 2024-08-09 by nashif