zephyrproject-rtos/zephyr #51007: Improve process aound feature freeze exceptions

nashif, 2022-10-05:
Follow up to escalation path and post freeze exceptions, ideas for improving the process:

- Any changes raised for exception need to be complete, approved and ready for merge. (also approved by assignee)
- Changes need to be first discussed and brought to the attention of the release management working group (Escalation to the TSC should not be the first step). Maintainers of the given subsystem that the change touches should be involved.
- Reestablish a CCB (Change Control Board) as the second step in the escalation process to deal with exceptions for a release. Define CCB structure, ie. 4-5 TSC members.
- As the final step, a change can be escalated to the TSC and a vote is called.

mbolivar-nordic, 2022-10-26:
Process WG:

- @nashif: this is related to an exception request we received to merge a PR in the v3.2 feature freeze. The idea is that there should be an escalation path where the TSC is not the first place we raise requests for exceptions. Gaps here are to document the escalation process and establish a decision making body that can deal with these issues before the TSC.
- @mbolivar-nordic so you're looking concretely for changes to https://docs.zephyrproject.org/latest/project/release_process.html#stabilization-phase that we will hold release managers to
- Take it to the TSC - no objections to the proposal or its language

henrikbrixandersen, 2022-11-02:
Where can I found the proposed changes to the stabilisation phase documentation?

mbolivar-nordic, 2022-11-02:
Approved at TSC

closed 2022-11-02 by mbolivar-nordic