zephyrproject-rtos/zephyr #51048: Firmware Upgrade Issue with Mcumgr On STM32H743 controller

vipulkute-eaton, 2022-10-06:
I am working on the Firmware upgrade with mcumgr on STM32H743 controller.
I have flash the signed primary and signed secondary image and using mcumgr command I am able to see the file list. 
mcumgr command : mcumgr image list --conntype serial --connstring "COM7,baud=115200"

But when I try to test or confirm the secondary image, i am getting error.
mcumgr command : mcumgr --conntype serial --connstring "COM7,baud=115200" image test [hash of the secondary image]
Response: Error 1

After debugging I observed that for STM32H743 controller flash driver only accept write flash address value divisible by 32.
During image test or image confirm, image trailer of secondary image needs to be updated. During this process when code 
tries to update the magic number in the flash memory, the address of the magic number field is not divisible by 32 and  
due to this code returns the error. 
Code fails inside the function flash_stm32_valid_range() in file flash_stm32h7x.c

This issue only exist for STM32H743 controller as i have already tested the firmware upgrade on STM32F429 with same steps and it works fine.  For stm32h7 controller there is difference in flash layout and sector size compare to STM32F429 controller. 

Does Firmware upgrade support on STM32H743 controller ? Is there any change needs to be done for STM32H743 controller to support the firmware upgrade?

erwango, 2022-10-07:
@vipulkute-eaton If the issue is around creation of the image, can you check if using `--align 32` when calling imgtool can help situiation?
There's a dedicated zephyr option for this `-DCONFIG_MCUBOOT_EXTRA_IMGTOOL_ARGS="--align 32"`.
Note that you might need to escape the command: `-DCONFIG_MCUBOOT_EXTRA_IMGTOOL_ARGS=\"--align\ 32\"`

aunsbjerg, 2022-10-08:
@vipulkute-eaton I believe what you are seeing is related to [this](https://github.com/mcu-tools/mcuboot/issues/713) mcuboot issue.

vipulkute-eaton, 2022-10-20:
> @vipulkute-eaton I believe what you are seeing is related to [this](https://github.com/mcu-tools/mcuboot/issues/713) mcuboot issue.

Yes it was related to same issue but is there any solution provided in any release?

vipulkute-eaton, 2022-10-20:
In main.py inside imgtool folder I dont see there is 32byte alignment choice present. 
@click.option('--align', type=click.Choice(['1', '2', '4', '8']), required=True)

utzig, 2022-10-20:
I tested both MCUboot upgrades and MCUmgr after this commit https://github.com/zephyrproject-rtos/mcuboot/commit/66cdd46b4494be95ea549db811bad242b253c51e, on a Nucleo-H753ZI and everything worked correctly. So you just need to be sure you are after that commit, or using `main`. I think there was no MCUboot release after that PR was merged. Also afair you don't need to add extra parameters to `imgtool`, since `west` already fetches the alignment from the DTS, if using `west sign`...

carlescufi, 2022-10-25:
Closing based on [this comment](https://github.com/zephyrproject-rtos/zephyr/issues/51048#issuecomment-1286121606). Please reopen if it's still an issue.

vipulkute-eaton, 2022-10-26:
@utzig can you please confirm your Zephyr release version on which you have tested the firmware upgrade?. Bootloader side i will clone the commit ID as highlighted by you.

utzig, 2022-10-26:
> @utzig can you please confirm your Zephyr release version on which you have tested the firmware upgrade?. Bootloader side we will clone the commit ID as highlighted by you.

Since I am not working with embedded systems anymore, I removed my local Zephyr clone, and didn't think of writing down the version I tested. I only remember it was just after the merge of the fix commit I pointed above. Anyway I suggest you just go with `main` as is, because if this stopped working for any reason, or you find some other issue, `main` is what will have to be fixed, in which case just reopen this issue.