zephyrproject-rtos/zephyr #51057: Retrieve gpios used by a device (pinctrl)

danielstuart14, 2022-10-07:
After hours researching, I think I found out that with pinctrl we aren't able to get the gpio a device is using. Is that correct?
Prior to pinctrl, I was able to get the RX and TX pin from a Serial device using its tx-pin and rx-pin property, for example.
Does anyone knows if this is still possible with pinctrl?

I'm currently developing a LIN bus driver and I need access to the RX and TX GPIO from the UART device so I can read/write the BREAK symbol. It would be nice if the driver was able to get the gpio/pin used by the UART without needing to be specified.

henrikbrixandersen, 2022-10-10:
Which SoC? nRF? And I believe you are asking for the pin used, not the GPIO used (as a UART does not use GPIOs).

gmarull, 2022-10-10:
If you need such feature, you can create a new compatible that _extends_ your uart device with `tx-gpios`, `rx-gpios` similar to how certain I2C drivers do for recovery (you can use them as a reference).

closed 2022-10-10 by gmarull

danielstuart14, 2022-10-11:
@gmarull I'll take a look, thank you!