zephyrproject-rtos/zephyr #51060: 10-bit addressing not supported by I2C slave driver for STM32 target

AshwiniMShinde-eaton, 2022-10-07:
I am working on I2C master and slave example codes to enable 10-bit addressing. I am using STM32H743ZI nucleo board. In slave example code, the issue is that i2c_slave_register() function doesn't support 10-bit addressing.
After adding the following line in drivers/i2c/i2c_ll_stm32_v2.c file, this issue got fixed.

`	if( data->slave_cfg->flags == I2C_SLAVE_FLAGS_ADDR_10_BITS )
	{
		LL_I2C_SetOwnAddress1(i2c, config->address, LL_I2C_OWNADDRESS1_10BIT);
		printk("\nSlave registered with 10-bit address");
	}
	else
	{
		LL_I2C_SetOwnAddress1(i2c, config->address << 1, LL_I2C_OWNADDRESS1_7BIT);
		printk("\nSlave registered with 7-bit address");
	}`

erwango, 2022-10-07:
@AshwiniMShinde-eaton Thanks  for sharing this.
Would you mind raising a PR directly ?

AshwiniMShinde-eaton, 2022-10-10:
> @AshwiniMShinde-eaton Thanks for sharing this. Would you mind raising a PR directly ?

Created PR https://github.com/zephyrproject-rtos/zephyr/pull/51119