zephyrproject-rtos/zephyr #51062: lora_recv_async receives empty buffer after multiple receptions on sx12xx

ivanwagner, 2022-10-07:
**Describe the bug**
When a second call is being made to lora_recv_async (using a sx12xx LoRa module) it actually returns an empty buffer of data reception. The length of the received buffer is actually correct but the contents are filled with zeros.

**To Reproduce**
Steps to reproduce the behavior:
To reproduce this target you can use any board which mounts an sx12xx such as:
- rm1xx_dvk
- 96b_wistrio
- b_l072z_lrwan1
- rak4631_nrf52840

**Expected behavior**
I would expect the correct received buffer.

**Impact**
It makes the lora_recv_async not usable.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain (e.g Zephyr SDK 0.14
 - v3.1

ivanwagner, 2022-10-07:
It leads down to resetting Rx before calling the the user callback. I've opened a PR here: https://github.com/zephyrproject-rtos/zephyr/pull/51063

martinjaeger, 2022-10-11:
Duplicate of https://github.com/zephyrproject-rtos/zephyr/pull/48765 and already fixed.

closed 2022-10-11 by martinjaeger