zephyrproject-rtos/zephyr #51070: ModuleNotFoundError: No module named 'elftools'

WajahatRiaz, 2022-10-07:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/51069

<div type='discussions-op-text'>

<sup>Originally posted by **WajahatRiaz** October  7, 2022</sup>
**Describe the bug:**
I'm trying to build a simple blinky application for my stm32f103rb board but I'm getting the **ModuleNotFoundError: No module named 'elftools'** error over and over again. I tried it for different boards and QEMU. The trivial solutuon of installing **pyetools** 
also didn't work because it's already installed. All efforst are in vain. I used the following guide to setup the zephyr project for my Windows based host. 

[https://docs.zephyrproject.org/latest/develop/getting_started/index.html](url)

**To Reproduce:**
`cd %HOMEPATH%\zephyrproject\zephyr`
`west build -p always -c -b nucleo_f103rb samples\basic\blinky`

**Expected behavior:**
A successful build.

**Impact**
Unable to build and run application.

**Logs and console output:**
[image]

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: Zephyr SDK
 - Version: latest

**Additional context:**
Someone mentioned that it may be some issue with the dependencies so I ran the following commands:
`cd %HOMEPATH%\zephyrproject\zephyr`
`pip3 install --user --upgrade -r scripts/requirements.txt`

All requirements were already installed and updated. Especially the **pyelftools**
[image]

</div>

closed 2022-10-11 by stephanosio

WajahatRiaz, 2022-10-13:
> Issues are not for questions/asking for help. Please keep using the discussion for that.
> 
> The upstream build environment is regularly tested and no such error has been observed. This is very likely an issue with your local environment set-up.

I actually pointed out some of the flaws in your documentation. I was unable to narrow down to a solution. Also, if you didn't know the answer you could have at least guided me.

stephanosio, 2022-10-13:
> I actually pointed out some of the flaws in your documentation. 

Where did you do that?

If you have followed the Getting Started Guide down to every line (i.e. run as admin, install without `--user`), this problem would not have happened.

Also, we use GitHub issues to keep track of bugs, enhancements, etc, and weekly go through them in a meeting. That is why this was converted to a discussion -- because in its form, it is neither a bug nor an enhancement.

WajahatRiaz, 2022-10-13:
> > I actually pointed out some of the flaws in your documentation. 
> 
> Where did you do that?
> 
> If you have followed the Getting Started Guide down to every line (i.e. run as admin, install without `--user`), this problem would not have happened.
> 
> Also, we use GitHub issues to keep track of bugs, enhancements, etc, and weekly go through them in a meeting. That is why this was converted to a discussion -- because in its current form, it is neither a bug nor an enhancement.

Installing with or without the `--user` had no effect. Alas, it's pointless to debate with you. Thank you for your time.