zephyrproject-rtos/zephyr #51073: Runtime locking correctness validator

desowin, 2022-10-07:
**Is your enhancement proposal related to a problem? Please describe.**
Sometimes a function expects that the caller holds specific lock (e.g. mutex). Zephyr currently doesn't have any runtime method of checking locking violations (e.g. caller does not hold mutex).

**Describe the solution you'd like**
Something similar to Linux [Runtime locking correctness validator](https://www.kernel.org/doc/Documentation/locking/lockdep-design.txt), especially the `lockdep_assert_held()`.

**Describe alternatives you've considered**
`__ASSERT_NO_MSG(mutex->owner == z_current());` but accessing the owner field directly seems wrong. Also, the locking correctness validation is something that should specifically have to be enabled.

**Additional context**
Linux kernel lockdep can do much more than just asserting that a lock is held, e.g. it can detect deadlock possibility before actually hitting the deadlock. The full lockdep functionality is probably not possible in Zephyr due to target device constraints, but just having equivalent to `lockdep_assert_held()` would already be helpful.

henrikbrixandersen, 2022-10-19:
CC: @andyross

cfriedt, 2022-10-20:
Just a couple of thoughts:
* Due to reaching into the mutex structure, it will likely require a new system call e.g. `__syscall bool z_mutex_is_locked_by(mutex*, thread*)`
* However, it should also be possible to enable / disable the feature with Kconfig (so that it is zero overhead when disabled)
* Again, however, fwir, syscalls always need to be present (i.e. not macros), so likely we will need a macro or static inline that wraps around the eventual syscall
E.g. 
```cpp
static inline void k_mutex_held(struct k_mutex *mu) {
  if (IS_ENABLED(CONFIG_MUTEX_CHECKING)) {
    __ASSERT(z_mutex_is_held_by(mu, k_current_get()), "msg..");
  }
}
```

andyross, 2022-10-20:
There's something like this in spinlocks already: CONFIG_SPIN_VALIDATE.  And in the almost-ready-I-swear Zync PR[1], there's a broadly equivalent CONFIG_ZYNC_VALIDATE.  It doesn't provide the ownership as an API though, because that requires storing the owner, which is a comparatively heavy operation for an RTOS.

[1] Check the andyross/zephyr/zynctmp branch if you want to see the most recent work.  It has dozens of WIP patches that need to be squashed still, but is much closer to releasable state.

luisacicolini, 2022-12-13:
@andyross thank you very much for your comments. I took some time to dig deeper into the theory of runtime validators and locking classes and to look at your repository. As far as I've seen, Zync contains some relevant modifications with respect to the original Zephir kernel, would it be more helpful if I worked on the validator directly on Zync?  Thanks :)

Finder16, 2026-09-16:
I took a look at the current codebase, and it seems that some related functionality has already been added since this issue was opened.

Spinlocks have validation support through `CONFIG_SPIN_VALIDATE`, and mutexes now have `CONFIG_MUTEX_DEADLOCK_DETECT`, which detects deadlock cycles by following mutex ownership chains.

At the same time, the Bluetooth host defines its own `BT_DEV_LOCK_ASSERT()` and directly checks `bt_dev.lock.owner == k_current_get()`. That seems like a concrete example where a common mutex ownership assertion, similar to `lockdep_assert_held()`, would still be useful.

Rather than starting with a full lock dependency graph, would it make sense to first provide a common debug/assertion primitive for verifying that the current thread owns a given `k_mutex`, and then consider integration with the existing spinlock validation and mutex deadlock detection later?

I’m also curious whether this should remain a simple mutex-specific assertion, or be designed as a more general validation interface that could eventually cover other locking primitives.