zephyrproject-rtos/zephyr #51079: samples: tests: Improve accuracy of required dependencies in networking tests

katgiadla, 2022-10-07:
**Problem description:**
Majority of the networking testcases require any network interface - this requirement is specified in testsuite YAML by using keyword _netif_.
On the boards, which support this feature, it is possible to build and/or execute test configuration.

Some test suites assume that specific networking technology will be used ([example for Ethernet](https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/net/bridge/testcase.yaml) and [for OpenThread](samples/net/openthread/coprocessor)). 

Unfortunatelly, these technologies are not accurate specified. It means, samples/tests are build and/or execute on the boards, which does not support specified technology but support any network interface.

**Solution:**
Clarify, which network feature is required in testsuites/sample by changing field _depends_on_.
e.g. [example for Ethernet](https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/net/bridge/testcase.yaml)

Currently:
` depends_on: netif`

Change into:
` depends_on: eth`

rlubos, 2024-02-21:
For samples. I've opened https://github.com/zephyrproject-rtos/zephyr/pull/69295 to narrow the testing scope for the specialized ones (mostly Ethernet stuff, the only OT speicifc is the linked one).

As for the tests, I've looked them over but I'm not really sure we should change anything there. All of the tests we have (unless I missed something) do not really depend on the actual hardware, as they operate on fake interfaces they create.

jukkar, 2024-02-22:
Also after #69244 the tests should run without any board specific Ethernet interfaces.

PR #69295 (2024-03-06): net: Add more specific dependencies for specialized samples
Certain samples  are supposed to demonstrate technology-specific features, therefore their dependencies in sample.yml could be narrowed to verify build only on boards which support that specific technology (like Ethernet or OpenThread).

Resolves #51079
closed 2024-03-06 by aescolar