zephyrproject-rtos/zephyr #51094: RISC-V support in Zephyr v2.7 (LTS2)

npitre, 2022-10-07:
## Introduction

Companies wishing to release products based on Zephyr prefer to base 
their work on an LTS branch. The latest LTS available is the v2.7 branch 
whose RISC-V support is severely lacking.

### Problem description

The latest LTS branch still dates back to a year ago. At the time, the
RISC-V support was rather poor and subpar. In the mean time, RISC-V 
support in Zephyr has hugely progressed. Without those post-LTS changes
the latest LTS is almost unusable for actual products based on RISC-V,
especially RV64 SMP configurations.

### Proposed change

Backport latest RISC-V support to the v2.7 branch in order to make
decent RISC-V support available in at least one LTS branch.

## Detailed RFC

The work to make RISC-V full-featured and stable in the v2.7 LTS branch 
has already been done. Backported patches fall into 3 categories:

- backport of fixes for CI test flaws the RISC-V + SMP configuration exposed
  (about 10 commits)

- generic Zephyr minor fixes and API clarifications relied upon by the 
  RISC-V code (4 commits: 3 backports and 1 original)

- RISC-V specific changes with no impact on any other architectures 
  (about 55 commits, almost all backports)

### Proposed change (Detailed)

Here's the complete list of required commits:

- riscv: linker: fix __rom_region_end in non-XIP system

- riscv: linker: fix PMP support on non-XIP system.

- gen_app_partitions: add .sdata/.sbss section into app_smem

- riscv: linker: fix duplicate symbol when CONFIG_USERSPACE=y

- riscv: linker: move snippets-ram-sections.ld to ram start

- tests: disable division-by-zero test in RISC-V arch

- arch: riscv: pmp: Fix 64-bit compatibility of pointer size

- arch: riscv: pmp: Fix RV64 compatibility of register size

- arch: riscv: pmp: Fix is_user_mode in RV64

- riscv: fix non-standard assembly of RISC-V

- arch/riscv: Implement arch_curr_cpu()

- arch/riscv: Boot secondary CPUs for SMP support

- arch/riscv: Add IPI support

- riscv: use simplest asm expression when possible

- riscv: fix wrong access width in assembly code

- riscv: better abstraction for register-wide load/store opcodes

- riscv: better abstraction for register-wide FP load/store opcodes

- riscv: exception code mega simplification and optimization

- riscv: irq_offload: simpler implementation

- riscv: implement CONFIG_IRQ_OFFLOAD_NESTED

- riscv: align struct __esf properly

- sched: formalize the passing of NULL to z_get_next_switch_handle()

- riscv: implement arch_switch()

- riscv: isr.S: optimize FP regs save/restore decision

- riscv: isr.S: dedicate a register to &current_cpu

- riscv: isr.S: compute _current_cpu using CPU number on SMP

- riscv: _isr_wrapper: get rid of the ASSUME_EQUAL() macro

- riscv: Introduce RISCV_ALWAYS_SWITCH_THROUGH_ECALL

- arch: riscv: ensure fcsr is cleared on thread start or FPU enable

- riscv: pmp: new implementation

- riscv: pmp: new stackguard implementation

- riscv: pmp: add new usermode support

- riscv: pmp: switch over to the new implementation

- riscv: pmp: remove previous implementation

- riscv: pmp: rename CONFIG_PMP_SLOT

- riscv: pmp: work around a QEMU bug

- riscv: pmp: make PMP debug display more comprehensive

- riscv: clarify stack size and alignment parameters

- riscv: drop user stack guard area when using separate privileged stacks

- riscv: fix IRQ stack guard location

- riscv: report stack overflow errors correctly

- riscv: rationalize PMP related Kconfig options

- arch: riscv: Remove outdated comment

- riscv: stop preserving the tp register needlessly

- riscv: TLS-based arch_is_user_context() implementation

- riscv: pmp: work around another QEMU bug

- riscv: pmp: properly initialize per-thread m-mode/u-mode entry array

- riscv: linker: align end of ROM area to a "page" boundary for QEMU

- scripts: gen_syscalls: fix access validation size on extra params array

- scripts: gen_syscalls: add missing --split-type case

- tests/kernel/fpu_sharing: Run test with MP_NUM_CPUS=1

- tests/kernel/smp: Add SMP switch torture test

- tests: kernel/smp: wait for threads to exits between tests

- tests: lifo_usage: make it less susceptible to SMP races

- riscv: smp: fix secondary cpus' initial stack

- riscv: pmp: fix stackguard when used on SMP

- riscv: PMP-based stack guard is incompatible with stack sentinel

- riscv: fix crash resulting from touching the initial stack's guard area

- drivers/timer: Use correct timer for CPU in SMP Risc-V

- boards/riscv: Add qemu_riscv32_smp and qemu_riscv64_smp

- riscv: smp: update the qemu_riscv32/64 configs

- tests/kernel/mem_protect: Check for thread_userspace_local_data

- tests: mem_protect: enlarge heap size of RISCV64

- tests: mem_protect: avoid allocating K_OBJ_MSGQ in userspace.

- tests: mem_protect: ensure allocated objects are initialized

- tests: Disable HW stack protection for some mpu tests

- z_thread_mark_switched_*: use z_current_get() instead of k_current_get()

- riscv: increase stack size for tests on RV64

### Dependencies

This should have no impact on non RISC-V targets.

### Concerns and Unresolved Questions

There is no official process for this kind of request.

It is unclear to me how to create PRs for the v2.7 branch and not the main branch.

Strictly speaking, "decent" RISC-V support wasn't available in LTS2 so 
there is no bug per se to fix. In essence this is more about adding 
new support to a "stable" branch. However the proposed changes are 
almost entirely confined to RISC-V specific files so the risk to existing 
LTS2 users is close to none.

## Alternatives

Wait for the next LTS release. However this is still far away and 
the date isn't yet predictable. This doesn't suit industry players well.

henrikbrixandersen, 2022-10-10:
We usually do not backport added functionality, only bugfixes. I would be especially cautious seeing the amount of changes proposed here.

cfriedt, 2022-10-11:
Agree with what Brix says. However, I would also support backports to ensure that LTS is both stable and includes RISC-V bugfixes.

@npitre has highlighted a number of issues in the testsuite - that should be fine.

One item that would be quite important to highlight here, is that @npitre has demonstrated a massive performance issue between LTSv2 and `main` on RISC-V. IIRC, It was 1-2 orders of magnitude which is huge problem in LTSv2 - enough to consider it a bug.

My position on SMP support is that it is an acceptable feature to backport, since
* it does not change any actual Zephyr API (if anything, brings RISC-V closer to Zephyr API)
* is isolated to the RISC-V architecture
* there is at least 1 RISC-V SoC with SMP support in `v2.7-branch` (IIRC, need to verify)
* we want product manufacturers to base their products on LTS
* able to bridge the gap between LTSv2 and LTSv3 is a legitimate expectation

My main criteria to accept this the RISC-V SMP feature would be:
a) demonstrated stability (zero regressions + soak period)
b) TSC approval

cfriedt, 2022-10-21:
@npitre - could you push a branch to your GitHub and drop a link here? Alternatively, even a draft PR would be good. @pgielda would like to do some hardware validation as well.

npitre, 2022-10-21:
You may find the v2.7_riscv_smp branch at https://github.com/npitre/zephyr

npitre, 2022-10-27:
... also in PR #51727.

nashif, 2023-10-24:
This was discussed in the TSC and consensus was not to add new architecture features to a released branch, a topic branch was created that became stale without progress (branch was removed). So closing. Please reopen if you disagree.

closed 2023-10-24 by nashif