zephyrproject-rtos/zephyr #51105: esp32 wifi: http transmit rate is too slow

zoujixing, 2022-10-09:
ESP32 as wifi station mode to connect server by http client(zephyr/subsys/net/lib/http). Request to downlow file.The file size is about 1MB, need about 30minutes.
This time is too long.How to inrease download rate?Follow is relevant setting:

# Networking config
CONFIG_NET_TCP=y
CONFIG_NET_UDP=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
CONFIG_NET_SOCKETS_SOCKOPT_TLS=y
CONFIG_NET_CONNECTION_MANAGER=y
CONFIG_HTTP_PARSER=y
CONFIG_HTTP_PARSER_URL=y
CONFIG_HTTP_CLIENT=y

CONFIG_WIFI=y
CONFIG_WIFI_ESP32=y
CONFIG_WIFI_ESWIFI_THREAD_PRIO=5
CONFIG_ESP32_WIFI_EVENT_TASK_PRIO=7
CONFIG_NETWORKING=y
CONFIG_NET_L2_ETHERNET=y
CONFIG_NET_IPV6=n
CONFIG_NET_IPV4=y
CONFIG_NET_DHCPV4=y
CONFIG_NET_MGMT=y
CONFIG_NET_MGMT_EVENT=y
CONFIG_ESP32_WIFI_STA_RECONNECT=n
CONFIG_NET_RX_STACK_SIZE=6144

ssharks, 2022-10-09:
Sounds like a bad tcp implementation. In the native tcp stack, there was a similar issue, but that has been fixed by

maweicoltsmart, 2022-10-10:
@ssharks
esp32 wifi compile error
branch：main
```
west update
west blobs fetch hal_espressif
west build -p auto -b esp32 samples/net/wifi/
```

```
[1/11] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/colt/zephyrproject/code/zephyr/build/zephyr/zephyr_pre0.map 
: && ccache /home/colt/zephyr-sdk-0.15.0/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/xtensa/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/colt/zephyrproject/code/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/subsys/net/libsubsys__net.a  zephyr/subsys/net/l2/ethernet/libsubsys__net__l2__ethernet.a  zephyr/subsys/net/l2/wifi/libsubsys__net__l2__wifi.a  zephyr/subsys/net/ip/libsubsys__net__ip.a  zephyr/subsys/random/libsubsys__random.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/wifi/libdrivers__wifi.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/entropy/libdrivers__entropy.a  zephyr/drivers/timer/libdrivers__timer.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  modules/mbedtls/libmodules__mbedtls.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/home/colt/zephyr-sdk-0.15.0/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.1.0"  -L/home/colt/zephyrproject/code/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -no-pie  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -T/home/colt/zephyrproject/code/modules/hal/espressif/zephyr/esp32/src/linker/esp32.rom.alias.ld  -T/home/colt/zephyrproject/code/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.ld  -T/home/colt/zephyrproject/code/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.api.ld  -T/home/colt/zephyrproject/code/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.libgcc.ld  -T/home/colt/zephyrproject/code/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.redefined.ld  -T/home/colt/zephyrproject/code/modules/hal/espressif/zephyr/esp32/../../components/soc/esp32/ld/esp32.peripherals.ld  -Wl,--wrap=k_calloc  -Wl,--wrap=k_malloc  -lnet80211  -lcore  -lpp  -lcoexist  -lphy  -lrtc  -lgcc  -L/home/colt/zephyrproject/code/modules/hal/espressif/zephyr/esp32/../blobs/lib/esp32 && cd /home/colt/zephyrproject/code/zephyr/build/zephyr && /usr/bin/cmake -E echo
/home/colt/zephyrproject/code/modules/hal/espressif/zephyr/esp32/../blobs/lib/esp32/libnet80211.a: file not recognized: file format not recognized
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/colt/zephyrproject/code/zephyr/build
```

ssharks, 2022-10-10:
Did you run `west espressif update`, to de the binary blobs?

maweicoltsmart, 2022-10-10:
@ssharks
```
west espressif update                                                                     
usage: west espressif [-h] [-b BAUD] [-p PORT] [-e ELF] {monitor}
west espressif: error: argument command: invalid choice: 'update' (choose from 'monitor')
```

ssharks, 2022-10-10:
Sorry, should have referred to the latest docs: west blobs fetch hal_espressif

ssharks, 2022-10-10:
See https://docs.zephyrproject.org/latest/boards/xtensa/esp32/doc/index.html

maweicoltsmart, 2022-10-10:
@ssharks

I did this at the beginning, but compilation errors still occur

sylvioalves, 2022-10-10:
@zoujixing which Zephyr version/revision are you working with? I was able to run zperf and download data with ~10MBit/s. Let me now so we can optimized TCP buffers and IRAM content.  
Also, trying increase this values if applicable:  
```c
CONFIG_NET_BUF_DATA_SIZE=1024
CONFIG_NET_TCP_MAX_RECV_WINDOW_SIZE=20000
```

sylvioalves, 2022-10-10:
> Sounds like a bad tcp implementation. In the native tcp stack, there was a similar issue, but that has been fixed by #46584. Espressif is using binary blob to implement the TCP stack, so little change of fixing it yourself. The best you can do is to try and make a wireshark log.

@ssharks, ESP32 uses Zephyr's native TCP stack, not esp-idf. It would be esp-idf net stack if using offloaded Wi-Fi network and external ESP32 board with esp-at.

zoujixing, 2022-10-11:
> @zoujixing which Zephyr version/revision are you working with? I was able to run zperf and download data with ~10MBit/s. Let me now so we can optimized TCP buffers and IRAM content. Also, try to increase this values if applicable:
> 
> ```c
> CONFIG_NET_BUF_DATA_SIZE=1024
> CONFIG_NET_TCP_MAX_RECV_WINDOW_SIZE=20000
> ```

@sylvioalves I use V3.1 branch version.

zoujixing, 2022-10-12:
@ssharks @sylvioalves I add [#46584)](https://github.com/zephyrproject-rtos/zephyr/pull/46584) to v3.1 release version.Now the download rate is 1MB/1minute. Any other suggest to increase rate?

ssharks, 2022-10-12:
You might want to use zephyr 3.2.0 instead as a ton of tcp fixes have been done recently. On of the major contributors is the buffer size and the number of packets in flight. Increasing CONFIG_NET_BUF_DATA_SIZE is one option, but increasing the number of packets instead would be my choice, as it is more efficient with small packets. See https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/net/socket/tcp/prj.conf for some inspiration. If you are not memory constraint, try 128 or more tx buffers for example.

ssharks, 2022-10-12:
Also increase CONFIG_NET_PKT_TX_COUNT, accordingly

ssharks, 2022-10-12:
> @ssharks @sylvioalves I add [#46584)](https://github.com/zephyrproject-rtos/zephyr/pull/46584) to v3.1 release version.Now the download rate is 1MB/1minute. Any other suggest to increase rate?

Anyway, you improved a 30 fold!

ssharks, 2022-10-12:
A part that also significantly helped with me was https://github.com/zephyrproject-rtos/zephyr/pull/46348. Especially since I was doing rather small writes.

`CONFIG_NET_PKT_RX_COUNT=32
CONFIG_NET_PKT_TX_COUNT=32
CONFIG_NET_BUF_RX_COUNT=128
CONFIG_NET_BUF_TX_COUNT=128`

And remove the `CONFIG_NET_BUF_DATA_SIZE` setting, to get the default 128 bytes. In this case there can be a maximum of 32 packets in flight. The default transmit window is than 128*128/3 = 5.4 kB. Suppose a round trip of 100 ms, you are bound to 54 kB/s maximum. With 1 MB/s you are at 16.6 kB/s. 
To significantly increase even further you have to increase CONFIG_NET_BUF_TX_COUNT further or reduce the latency. Normally the MTU should go to 1280 bytes/message, which means CONFIG_NET_PKT_TX_COUNT should be >= CONFIG_NET_BUF_TX_COUNT/10. If you want to prioritize upload the CONFIG_NET_BUF_RX_COUNT and CONFIG_NET_BUF_TX_COUNT can be asymmetrical.

sylvioalves, 2022-10-13:
@zoujixing, let us know once you try out those suggestion above.

zoujixing, 2022-10-17:
@ssharks @sylvioalves 
Add [#46348](https://github.com/zephyrproject-rtos/zephyr/pull/46348) and use following setting:
`
#CONFIG_NET_BUF_DATA_SIZE=512
CONFIG_NET_PKT_RX_COUNT=32
CONFIG_NET_PKT_TX_COUNT=32
CONFIG_NET_BUF_RX_COUNT=320
CONFIG_NET_BUF_TX_COUNT=320
`
The download rate is about 1MB/13s.

ssharks, 2022-10-17:
Looks decent, is this sufficient to close to the issue?

closed 2022-10-17 by zoujixing

tiwarivivek, 2022-12-12:
@sylvioalves 
you wrote --
>  I was able to run zperf and download data with ~10MBit/s.

where to get zperf ref application for esp32 board with appropreate configs?

/sample/net/zperf doesnt have esp32 in /boards, and even after adding (similar as sample/net/wifi), i needed to change default linker directive for esp32 as otherwise dram0_1_seg starts 'overflowing' with many kbs.

other changes for a succesfull linking/running zperf with esp32 wifi interface involved -- 

1. reducing CONFIG_ESP_HEAP_MEM_POOL_REGION_1_SIZE to 1024
2. in /zephyrproject/zephyr/soc/xtensa/esp32/linker.ld, moving noinit from below lib:section from dram0_1_seg  to dram0_0_seg 
- libzephyr.a:log_core
- libkernel.a:init
- libkernel.a:system_work_q
- libsubsys__net__ip.a:net_mgmt
- libsubsys__net__ip.a:net_tc

But i am not sure if these changes can negatively impact performance or working of the app.

For example--
In my persuit to enable zperf for esp32 wifi interface, with changes explained above -- I am seeing some random FATAL error 0 related to 'system_work_q', and this exception / halt was NOT seen in /net/wifi/sample.
i am not sure if its related to mem segment movement i did (libkernel.a:system_work_q) or may be increaseing some stack config somewhere will help if its related to stack corruption etc!!

As, it seems, as per your comment here,  you are already internally using a zperf test on esp32, It will be good if the same (with appropriate configs) is provided as sample or reference! For a beginner, it will save lots of effort experimenting etc!