zephyrproject-rtos/zephyr #51133: Bluetooth: audio: Sink ASE does not go to IDLE state

MariuszSkamra, 2022-10-10:
**Describe the bug**
After the Release operation initiated by remote device, the ASE stuck in Releasing state.

**To Reproduce**
Steps to reproduce the behavior:
1. Reproducible with Android AOSP Master on Pixel 6
2. Start unicast server
3. Establish connection and play some music then stop playing
4. After a while, the Android sends Release command
5. Observe the issue

```
> ISO Data RX: Handle 2 plen 132                                                                                                                                                                                                               #799 66.436000
        Packet boundary flag: 0x02
        Timestamp flag: 1
        ISO data length: 128
          Timestamp: 60561283
          Packet seq. nb.: 305
          SDU length: 120
          Packet status flags: 0x00
> ACL Data RX: Handle 0 flags 0x02 dlen 10                                                                                                                                                                                                     #800 66.529600
      ATT: Write Command (0x52) len 5
        Handle: 0x0010
          Data: 080101
        ASCS: Audio Stream Endpoint Control Point (Handle: 0x0010)
        Release
        Number of ASEs: 1
        ASE ID: 1
< ACL Data TX: Handle 0 flags 0x00 dlen 12                                                                                                                                                                                                     #801 66.529800
      ATT: Handle Value Notification (0x1b) len 7
        Handle: 0x0010
          Data: 0801010000
        ASCS: Audio Stream Endpoint Control Point (Handle: 0x0010)
        Operation response: Release
        Number of ASEs: 1
         ASE ID: 1
          Response code: Success
< HCI Command: Disconnect (0x01|0x0006) plen 3                                                                                                                                                                                                 #802 66.530100
        Handle: 2
        Reason: Remote User Terminated Connection (0x13)
< ACL Data TX: Handle 0 flags 0x00 dlen 9                                                                                                                                                                                                      #803 66.530100
      ATT: Handle Value Notification (0x1b) len 4
        Handle: 0x0013
          Data: 0106
        ASCS: Audio Stream Sink Endpoint (Handle: 0x0013)
        ASE ID: 1
        Releasing
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                                                                                      #804 66.530300
      Disconnect (0x01|0x0006) ncmd 1
        Status: Success (0x00)
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                                                                                         #805 66.726100
        Num handles: 1
        Handle: 0
        Count: 1
> ACL Data RX: Handle 0 flags 0x02 dlen 10                                                                                                                                                                                                     #806 66.824500
      ATT: Handle Value Notification (0x1b) len 5
        Handle: 0x006f
          Data: 020300
        TBS: Call State (Handle: 0x006f)
        Unknown parser
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                                                                                         #807 66.922700
        Num handles: 1
        Handle: 0
        Count: 1
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                                                                                                 #808 66.928900
        Status: Success (0x00)
        Handle: 2
        Reason: Connection Terminated By Local Host (0x16)
> ACL Data RX: Handle 0 flags 0x02 dlen 10                                                                                                                                                                                                     #809 67.217700
      ATT: Write Request (0x12) len 5
        Handle: 0x004b
          Data: 0401b6
        VCS: Volume Control Point (Handle: 0x004b)
          Change counter: 23
          Volume setting: 0
          Set absolute volume
< ACL Data TX: Handle 0 flags 0x00 dlen 10                                                                                                                                                                                                     #810 67.218000
      ATT: Handle Value Notification (0x1b) len 5
        Handle: 0x0048
          Data: b60002
        VCS: Volume State (Handle: 0x0048)
          Volume setting: b6
          Mute: 0
          Change counter: 2
< ACL Data TX: Handle 0 flags 0x00 dlen 5                                                                                                                                                                                                      #811 67.218100
      ATT: Write Response (0x13) len 0
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                                                                                         #812 67.315900
        Num handles: 1
        Handle: 0
        Count: 1
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                                                                                         #813 67.365100
        Num handles: 1
        Handle: 0
        Count: 1
= bt: bt_ascs: No callback for stopped set                                                                                                                                                                                                          66.929100
= bt: bt_ascs: Could find ASCS based on ISO 0x2000c7e8 with ACL 0xe529                                                                                                                                                                              66.929100
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                                                                                                 #814 70.314600
        Status: Success (0x00)
        Handle: 0
        Reason: Remote User Terminated Connection (0x13)
```

Thalley, 2022-10-11:
@MariuszSkamra It is hard to tell what's going on without timestamps, as it also looks like the ACL disconnects at the end. 

But you are stating that the ASE never reaches the idle state on the device?

MariuszSkamra, 2022-10-11:
Yes. I've checked the current main and it looks like this particular issue have been fixed. However, there are 2 issues anyway:

1. `Released` notification is sent twice
2. `bt_audio_stream->released` callback is not called

```
> ISO Data RX: Handle 2 plen 132                                                                                                                                                                                                               #849 63.694900
        Packet boundary flag: 0x02
        Timestamp flag: 1
        ISO data length: 128
          Timestamp: 57942564
          Packet seq. nb.: 334
          SDU length: 120
          Packet status flags: 0x00
> ACL Data RX: Handle 0 flags 0x02 dlen 10                                                                                                                                                                                                     #850 69.174000
      ATT: Write Command (0x52) len 5
        Handle: 0x0010
          Data: 080101
        ASCS: Audio Stream Endpoint Control Point (Handle: 0x0010)
        Release
        Number of ASEs: 1
        ASE ID: 1
< ACL Data TX: Handle 0 flags 0x00 dlen 12                                                                                                                                                                                                     #851 69.174200
      ATT: Handle Value Notification (0x1b) len 7
        Handle: 0x0010
          Data: 0801010000
        ASCS: Audio Stream Endpoint Control Point (Handle: 0x0010)
        Operation response: Release
        Number of ASEs: 1
         ASE ID: 1
          Response code: Success
< HCI Command: Disconnect (0x01|0x0006) plen 3                                                                                                                                                                                                 #852 69.174500
        Handle: 2
        Reason: Remote User Terminated Connection (0x13)
< ACL Data TX: Handle 0 flags 0x00 dlen 9                                                                                                                                                                                                      #853 69.174600
      ATT: Handle Value Notification (0x1b) len 4
        Handle: 0x0013
          Data: 0106
        ASCS: Audio Stream Sink Endpoint (Handle: 0x0013)
        ASE ID: 1
        Releasing
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                                                                                      #854 69.174700
      Disconnect (0x01|0x0006) ncmd 1
        Status: Success (0x00)
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                                                                                         #855 69.370100
        Num handles: 1
        Handle: 0
        Count: 1
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                                                                                         #856 69.566700
        Num handles: 1
        Handle: 0
        Count: 1
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                                                                                                 #857 69.573000
        Status: Success (0x00)
        Handle: 2
        Reason: Connection Terminated By Local Host (0x16)
< ACL Data TX: Handle 0 flags 0x00 dlen 9                                                                                                                                                                                                      #858 69.573300
      ATT: Handle Value Notification (0x1b) len 4
        Handle: 0x0013
          Data: 0106
        ASCS: Audio Stream Sink Endpoint (Handle: 0x0013)
        ASE ID: 1
        Releasing
< ACL Data TX: Handle 0 flags 0x00 dlen 9                                                                                                                                                                                                      #859 69.573500
      ATT: Handle Value Notification (0x1b) len 4
        Handle: 0x0013
          Data: 0100
        ASCS: Audio Stream Sink Endpoint (Handle: 0x0013)
        ASE ID: 1
        Idle state
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                                                                                         #860 69.665000
        Num handles: 1
        Handle: 0
        Count: 1
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                                                                                         #861 69.714200
        Num handles: 1
        Handle: 0
        Count: 1
= bt: bt_ascs: No callback for stopped set                                                                                                                                                                                                          69.573100
```

MariuszSkamra, 2022-10-11:
And that is the case, `ep->stream == NULL`

Thalley, 2023-02-13:
@MariuszSkamra Are you looking at this?

Thalley, 2023-06-05:
@MariuszSkamra any idea of whether this is still an issue?

MariuszSkamra, 2023-06-06:
I don't think so, it shall be fixed now.

closed 2023-06-06 by Thalley