zephyrproject-rtos/zephyr #51142: TestPlan generation not picking up tests missing "test_" prefix

asemjonovs, 2022-10-10:
**Describe the bug**
If a developer misses adding "test_" prefix to their `ZTEST(foo, bar)` the resulting twister.json indicates `null` status for the test.
```
                {
                    "identifier":"foo.bar",
                    "execution_time":"0.00",
                    "status":null
                },
```

**To Reproduce**
Steps to reproduce the behavior:
1. Create ZTEST(foo, bar) test case // don't add prefix for the test bar
3. Run test

**Expected behavior**
I'd expect a PASS or FAIL status for the test

**Impact**
annoyance

**Logs and console output**
```
===================================================================
START - bar
 PASS - bar in 0.000 seconds
===================================================================
TESTSUITE foo succeeded
===================================================================
RunID: 9dfc67b98373aaf928e2e1827871a1e5
PROJECT EXECUTION SUCCESSFUL
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK

smrtos, 2022-10-13:
I added below tests to `tests/kernel/mutex/mutex_api/src/test_mutex_apis.c`

```
ZTEST(mutex_api, dummy)
{
	ztest_test_pass();
}
```

And run with twister:
```
twister -p qemu_x86 -vv -T tests/kernel/mutex/mutex_api
```

And got this:

> ERROR   - /home/ming/sources/zephyrproject/zephyr/tests/kernel/mutex/mutex_api/src/test_mutex_apis.c: Found a test that does not start with test_
> ERROR   - /home/ming/sources/zephyrproject/zephyr/tests/kernel/mutex/mutex_api: can't load (skipping): /home/ming/sources/zephyrproject/zephyr/tests/kernel/mutex/mutex_api/src/test_mutex_apis.c: Found a test that does not start with test_
>

asemjonovs, 2022-10-13:
I think this is partially related to https://github.com/zephyrproject-rtos/zephyr/issues/49746.  Where our directory structure doesn't follow the typical `<project>/src` format.  Try 

1. mv tests/ztest/base/src/ tests/ztest/base/src_bug
2. modify tests/ztest/base/CMakeLists.txt (rename src to src_bug)
3. add `ZTEST_USER(framework_tests, foo) {}` to `tests/ztest/base/src_bug/main.c`
4. Run `./scripts/twister -c -T tests/ztest/base/ -p native_posix`

Observe in generated `twister.json`
```
                {
                    "identifier":"testing.ztest.base.verbose_0.foo",
                    "execution_time":"0.00",
                    "status":null
                },
```

asemjonovs, 2022-10-13:
Also another scenario where this is not caught is when the `ZTEST_SUITE` is defined in another file.
File1:
```
ZTEST_SUITE(foo, NULL, NULL, NULL, NULL, NULL);
```
File2:
```
ZTEST(foo, bar) {}
```

Do we still need this source file parsing with the recent change to parse the symbol files?

nashif, 2022-10-13:
file parsing is ignored on new ztest api tests and is replaced with data from the binary, so I am not sure why you are getting those...

closed 2022-12-27 by github-actions[bot]