zephyrproject-rtos/zephyr #51146: Running test: drivers:  disk: disk_access with  RAM disk  fails

albertofloyd, 2022-10-11:
Exploring option to enable RAM filesystem, attempted running RAM disk test case.
The test fails for certain sectors.
Is this a configuration error? 

SUITE FAIL -   0.00% [disk_driver]: pass = 0, fail = 2, skip = 0, total = 2 durs
 - FAIL - [disk_driver.test_read] duration = 0.032 seconds
 - FAIL - [disk_driver.test_write] duration = 0.041 seconds

**To Reproduce**
Steps to reproduce the behavior:

1. Modify test code to enable RAM disk 
```
- CONFIG_DISK_DRIVER_SDMMC=y
+CONFIG_DISK_DRIVER_RAM=y
+CONFIG_DISK_RAM_VOLUME_SIZE=16

```
3. cd tests/drivers/disk/disk_access$
4. west build -c -p -b mec172xevb_assy6906
5. west_flash

Running TESTSUITE disk_driver
===================================================================
Disk reports 32 sectors
Disk reports sector size 512
START - test_read
Testing reads of 8 sectors
E: Sector 39 is outside the range 32
Testing reads of 1 sectors
Testing reads of 29 sectors
E: Sector 16 is outside the range 32

    Assertion failed at WEST_TOPDIR/zephyr_albertofloyd/tests/drivers/disk/disk)
Failed to read from mid disk sector
 FAIL - test_read in 0.032 seconds
===================================================================
START - test_write
Testing writes of 8 sectors
E: Sector 39 is outside the range 32
Testing writes of 1 sectors
Testing writes of 29 sectors
E: Sector 16 is outside the range 32

    Assertion failed at WEST_TOPDIR/zephyr_albertofloyd/tests/drivers/disk/disk)
Failed to write to mid disk sector
 FAIL - test_write in 0.041 seconds
===================================================================
TESTSUITE disk_driver failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -   0.00% [disk_driver]: pass = 0, fail = 2, skip = 0, total = 2 durs
 - FAIL - [disk_driver.test_read] duration = 0.032 seconds
 - FAIL - [disk_driver.test_write] duration = 0.041 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION FAILED

hunterhu3000, 2022-10-12:
The issue can be reproduced on my side, I'll take a look.

hunterhu3000, 2022-10-12:
Add some log:
```
START - test_read

@@@ test_read: SECTOR_COUNT1 = 8
Testing reads of 8 sectors
@@@@ test_sector_read: Testing reads of 8 sectors
@@@@ test_sector_read: disk_sector_count = 32, num_sectors = 8
@@@@@ test_sector_read: sector = 8, num_sectors = 8
@@@@ test_sector_read: disk_sector_count - 1 = 31, num_sectors = 8
E: Sector 39 is outside the range 32
@@@ test_read: SECTOR_COUNT2 = 1
Testing reads of 1 sectors
@@@@ test_sector_read: Testing reads of 1 sectors
@@@@ test_sector_read: disk_sector_count = 32, num_sectors = 1
@@@@@ test_sector_read: sector = 15, num_sectors = 1
@@@@ test_sector_read: disk_sector_count - 1 = 31, num_sectors = 1
@@@ test_read: SECTOR_COUNT3 = 29
Testing reads of 29 sectors
@@@@ test_sector_read: Testing reads of 29 sectors
@@@@ test_sector_read: disk_sector_count = 32, num_sectors = 29
@@@@@ test_sector_read: sector = -13, num_sectors = 29
E: Sector 16 is outside the range 32

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/disk/disk_access/src/main.c:102: test_sector_read: (rc not equal to 0)
Failed to read from mid disk sector
 FAIL - test_read in 0.096 seconds
```

This is because disk_sector_count and num_sectors are uint32, so the MAX function think -13 is bigger than 0. We need to change the form of MAX.

PR #51221 (2022-10-19): tests: drivers: disk: Replace the MAX function with if and else
As the parameter of "MAX" is uint32, the nagetive number will be bigger than 0. The MAX function does not work as expected. So change it to if/else form.

Fixes #51146