zephyrproject-rtos/zephyr #51149: Esp32 wifi compilation error

maweicoltsmart, 2022-10-11:
branch：main
sdk：zephyr-sdk-0.15.1
ubuntu20.04 intel x86

Run：
```
west update
west blobs fetch hal_espressif
west build -p auto -b esp32 samples/net/wifi/
```
The compilation error message is as follows：
```
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/colt/zephyrproject/code/zephyr/build/zephyr/zephyr_pre0.map 
: && ccache /home/colt/zephyr-sdk-0.15.1/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/xtensa/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/colt/zephyrproject/code/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/subsys/net/libsubsys__net.a  zephyr/subsys/net/l2/ethernet/libsubsys__net__l2__ethernet.a  zephyr/subsys/net/l2/wifi/libsubsys__net__l2__wifi.a  zephyr/subsys/net/ip/libsubsys__net__ip.a  zephyr/subsys/random/libsubsys__random.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/wifi/libdrivers__wifi.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/entropy/libdrivers__entropy.a  zephyr/drivers/timer/libdrivers__timer.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  modules/mbedtls/libmodules__mbedtls.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/home/colt/zephyr-sdk-0.15.1/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.1.0"  -L/home/colt/zephyrproject/code/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -no-pie  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -T/home/colt/zephyrproject/code/modules/hal/espressif/zephyr/esp32/src/linker/esp32.rom.alias.ld  -T/home/colt/zephyrproject/code/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.ld  -T/home/colt/zephyrproject/code/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.api.ld  -T/home/colt/zephyrproject/code/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.libgcc.ld  -T/home/colt/zephyrproject/code/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.redefined.ld  -T/home/colt/zephyrproject/code/modules/hal/espressif/zephyr/esp32/../../components/soc/esp32/ld/esp32.peripherals.ld  -Wl,--wrap=k_calloc  -Wl,--wrap=k_malloc  -lnet80211  -lcore  -lpp  -lcoexist  -lphy  -lrtc  -lgcc  -L/home/colt/zephyrproject/code/modules/hal/espressif/zephyr/esp32/../blobs/lib/esp32 && cd /home/colt/zephyrproject/code/zephyr/build/zephyr && /usr/bin/cmake -E echo
/home/colt/zephyrproject/code/modules/hal/espressif/zephyr/esp32/../blobs/lib/esp32/libnet80211.a: file not recognized: file format not recognized
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/colt/zephyrproject/code/zephyr/build
```

sylvioalves, 2022-10-11:
@maweicoltsmart I can't reproduce this at all. Would you clean the blobs `west blobs clean` and fetch again?

maweicoltsmart, 2022-10-12:
@sylvioalves 
It's my own fault

closed 2022-10-12 by maweicoltsmart