zephyrproject-rtos/zephyr #51165: tools/fiptool/fiptool: Permission denied

gmarull, 2022-10-11:
**Describe the bug**

Observed in CI the following error:

```
make: execvp: tools/fiptool/fiptool: Permission denied
make: *** [Makefile:1453: /__w/zephyr/zephyr/twister-out/fvp_base_revc_2xaemv8a_smp_ns/samples/application_development/out_of_tree_driver/sample.drivers.out_of_tree/tfa/fvp/release/fip.bin] Error 127
make: Leaving directory '/__w/zephyr/modules/tee/tf-a/trusted-firmware-a'
ninja: build stopped: subcommand failed.
```

**To Reproduce**

```
west build -b fvp_base_revc_2xaemv8a_smp_ns samples/application_development/out_of_tree_driver -p
```

(not always reproducible, CI re-run may pass)

**Expected behavior**

CI build **always** succeeds.

**Impact**

CI failures

**Logs and console output**

See above.

**Environment (please complete the following information):**
 - OS: CI image
 - Commit SHA or Version used 147d9cb1272b39e119025953acdb3d19ba52e45e

**Additional context**

https://github.com/zephyrproject-rtos/zephyr/actions/runs/3225514752/jobs/5277945064

SgrrZhf, 2022-10-12:
Sure, I'll take a look at it.

SgrrZhf, 2022-10-13:
We got some conclusions.

`fiptool` will be generated and placed in `zephyr-base/modules/tee/tf-a/trusted-firmware-a/tools/fiptool`. It won't be built when `fiptool` exist, but in the CI environment, several build jobs will be started simultaneously, this may cause a 'fiptool` permission error.

Now we are trying to fix it.

PR #52054 (2022-11-08): modules: tfa: update tf-a revision
Fix #51165, zephyr needs fiptool build out-of-tree.

Fixes #51165
commit b17118bc (2022-11-08): modules: tfa: update tf-a revision

Fix #51165, zephyr needs fiptool build out-of-tree.
closed 2022-11-08 by povergoing