zephyrproject-rtos/zephyr #51174: Bluetooth: l2cap needs check rx.mps when le_recv

yajunzhi, 2022-10-11:
**Describe the bug**
When a le dynamic channel connected on l2cap, and the mtu value is bigger than BUF_ACL_RX_SIZE, mps value is smaller than BUF_ACL_RX_SIZE. Then receiving l2cap packet from remote device, it could be received, but expect discard this too large packet according protocol.

So, in l2cap_chan_le_recv() function, should add some code to check the buffer length, refer the following,

 static void l2cap_chan_le_recv(struct bt_l2cap_le_chan *chan,
                               struct net_buf *buf) {
    ...
    **if (buf->len > chan->rx.mps) {
        BT_ERR("Invalid mps length");
        bt_l2cap_chan_disconnect(&chan->chan);
        return;
    }**

    /* Check if segments already exist */
    if (chan->_sdu) {
        l2cap_chan_le_recv_seg(chan, buf);
        return;
    }
    ...
}

carlescufi, 2022-10-11:
@yajunzhi feel free to open a Pull Request to fix this.

kruithofa, 2022-11-16:
May be related to #52228

jori-nordic, 2022-11-30:
Thanks for the code snippet @yajunzhi ! I opened a PR (https://github.com/zephyrproject-rtos/zephyr/pull/52631) to fix this issue. 
Closing it as duplicate, since https://github.com/zephyrproject-rtos/zephyr/issues/52228 has more details.

PR #52631 (2022-11-30): Bluetooth: Host: l2cap: disconnect when PDU > MPS
The Bluetooth spec states that a K-frame exceeding the device's MPS should trigger a disconnection of the channel.

Fixes two PTS tests that verify this behavior:
L2CAP/LE/CFC/BV-27-C
L2CAP/ECFC/BV-34-C

The current behavior is a silent truncation, which is not very nice for the user, even if the spec would allow it.

Fixes #52228 
Fixes #51174
closed 2022-11-30 by jori-nordic