zephyrproject-rtos/zephyr #51211: POSIX Roadmap for LTSv3

cfriedt, 2022-10-12:
## Introduction

This is an umbrella issue that will outline planned work to be done in the POSIX subsystem prior to LTSv3 (Ideally complete by v3.4). 

### Problem description

There was a lapse in maintainership of the POSIX subsystem for a period of time which slowed integration with the rest of the Zephyr codebase and addition of new POSIX features.

Currently, the most visible items on the TODO list are below.

* Remove the `zephyr/posix/time.h` header and `#include_next` workaround
  * This header and associated declarations are typically provided by the libc
* Support automatic pthread stack and object allocation
  * i.e. ensure that `pthread_create()` works with a `NULL` `pthread_attr_t *`
* Clean up layering issues between network and POSIX headers
  * Network headers currently define a number of POSIX symbols, macros, etc without proper namespacing
* The POSIX Architecture and POSIX Subsystem are currently at odds
  * `CONFIG_ARCH_POSIX` and `CONFIG_POSIX_API` are mutually exclusive
  * this is primarily due to conflicting symbols in the C global namespace
  * ideally, we would also like to be able to test Zephyr POSIX subsystem functions under `native_posix`
* Remove `CONFIG_NET_SOCKETS_POSIX_NAMES`
  * this is mainly needed for testing network library code on `native_posix`

### Proposed change

* Remove the `zephyr/posix/time.h` header and `#include_next` workaround
  * Work libc maintainers to ensure that the correct declarations are in `<time.h>`
* Support automatic pthread stack allocation and object creation
  * Add support for thread stack / object allocation to Zephyr
  * This can be configured to allocate from a system thread pool and / or from the heap
* Clean up layering issues between network and POSIX headers
  * Properly namespace Zephyr net symbols and constants
  * Make the POSIX subsystem should be a thin wrapper around net counterparts
* The POSIX Architecture and POSIX Subsystem are currently at odds
  * Use `dlopen()` to open / link to the native libc, thus avoiding pollution of the global C namespace
  * Properly abstract calls to the underlying "HAL" from within `native_posix` drivers and libraries
  * Remove `posix_cheats.h`
* Remove `CONFIG_NET_SOCKETS_POSIX_NAMES`
  * unnecessary once Zephyr POSIX library code can be used on `native_posix`

## Detailed RFC

### Proposed change (Detailed)

### Dependencies

TODO: links to issues  / PRs, etc

### Concerns and Unresolved Questions

## Alternatives

carlescufi, 2023-01-10:
Architecture WG:

- @cfriedt shows a presentation with a roadmap update for POSIX support
- @nashif asks about the scope of POSIX support, which profiles will be supported?

romkell, 2023-01-10:
- @cfriedt: when ever implementing dynamic features in the scheduler / core code (not yet defined safety scope) please consider safety where dynamic memory management and using heaps is problematic to get TÜV certification.
But I remember that we had that discussion earlier already and solved that by layering the dynamic part on top of the core code which can be passed statically OR dynamically allocated objects such as stack buffers, thread contexts etc. what so ever. I feel you are aware of that.

cfriedt, 2023-01-10:
> * @cfriedt: when ever implementing dynamic features in the scheduler / core code (not yet defined safety scope) please consider safety where dynamic memory management and using heaps is problematic to get TÜV certification.
Absolutely

>   But I remember that we had that discussion earlier already and solved that by layering the dynamic part on top of the core code which can be passed statically OR dynamically allocated objects such as stack buffers, thread contexts etc. what so ever. I feel you are aware of that.

Yeah, so the suggested approach from an earlier PR was to keep thread stack allocation separate and selectable, so that will still be the case. However, the proposed kernel-level API will also allow
* a static, pre-allocated pool of thread stacks to be used
* dynamic, heap-allocated thread stacks to be used
* both of the above (with a configurable preference)
* and of course neither

romkell, 2023-01-10:
> Yeah, so the suggested approach from an earlier PR was to keep thread stack allocation separate and selectable, so that will still be the case. However, the proposed kernel-level API will also allow
    - a static, pre-allocated pool of thread stacks to be used
    - dynamic, heap-allocated thread stacks to be used
    - both of the above (with a configurable preference)
    - and of course neither

Like it!

cfriedt, 2023-02-05:
Would be kind of nice to be able to use `socketpair()` without networking. I wonder if something can be borrowed from `eventfd`.. 🤔

abrodkin, 2023-06-04:
@cfriedt speaking of POSIX PSE profile test-suites (in the original issue description above), it might be a bit more correct to refer to "embedded versions" of those, such as:
* [VSX4-PSE](https://www.opengroup.org/testing/testsuites/vsx4pse.htm) - Embedded core POSIX system interfaces tests
* [VSPH-PSE](https://www.opengroup.org/testing/testsuites/vsthpse.htm) - Embedded threads tests
* [VSRT-PSE](https://www.opengroup.org/testing/testsuites/vsrtpse.htm) - Embedded realtime tests

cfriedt, 2023-06-04:
> ## Introduction
> 
> This is an umbrella issue that will outline planned work to be done in the POSIX subsystem prior to LTSv3 (Ideally complete by v3.4). 
> 
> Note: a full list of issues to be considered in the roadmap is [here](https://github.com/zephyrproject-rtos/zephyr/issues?q=is%3Aissue+is%3Aopen+label%3A%22Area%3A+POSIX%22+-label%3A%22RFC%22).
> 
> ### Problem description
> 
> There was a lapse in maintainership of the POSIX subsystem for a period of time which slowed integration with the rest of the Zephyr codebase and addition of new POSIX features.
> 
> Currently, the most visible items on the TODO list are below.
> 
> * Plan for formally supporting Real-Time and Embedded Profiles of [IEEE Std 1003.13-2003](https://ieeexplore.ieee.org/document/1342418) (a.k.a. [PSE52](http://get.posixcertified.ieee.org/docs/pse52-2003.html))
>   * https://github.com/zephyrproject-rtos/zephyr/issues/17706
>   * [Conformance Test Suites](https://www.opengroup.org/testing/testsuites/vspse52-2003.html):
>     * [VSX4-PSE](https://www.opengroup.org/testing/testsuites/vsx4.htm) POSIX [IEEE 1003.1](https://ieeexplore.ieee.org/servlet/opac?punumber=4694974) (Note: test is designed for [IEEE 1003.1-2001](https://ieeexplore.ieee.org/document/974398))
>     * [VSTH-PSE](https://www.opengroup.org/testing/testsuites/vsth.htm) Threads
>     * [VSRT-PSE](https://www.opengroup.org/testing/testsuites/vsrt.htm) Real-Time
> * Use a more standardized and compatible set of POSIX headers
>   * https://github.com/zephyrproject-rtos/zephyr/issues/13787
>   * Touched-upon briefly [here](https://github.com/zephyrproject-rtos/zephyr/pull/51771#issuecomment-1295846694)
>   * Zephyr needs to define feature test macros. See [2.2 The Compilation Environment](https://pubs.opengroup.org/onlinepubs/9699919799/functions/V2_chap02.html#tag_15_02)
>   * We will likely be borrowing as much header detail as possible from Newlib for compatibility / maturity purposes.
>   * Potentially enable / disable POSIX feature test macros with Kconfig options
> * With `CONFIG_POSIX_API`, Ensure POSIX headers can be included as `#include <poll.h>` rather than `#include <zephyr/posix/poll.h>`
>   * #43998
> * Support automatic pthread stack and object allocation
>   * i.e. ensure that `pthread_create()` works with a `NULL` `pthread_attr_t *`
>   * #25973
> * Clean up layering issues between network and POSIX APIs
>   * Network headers currently define a number of POSIX symbols, macros, etc without proper namespacing
>   * #24966
> * The POSIX Architecture and POSIX Subsystem are currently at odds
>   * `CONFIG_ARCH_POSIX` and `CONFIG_POSIX_API` are mutually exclusive
>   * this is primarily due to conflicting symbols in the C global namespace
>   * ideally, we would also like to be able to test Zephyr POSIX subsystem functions under `native_posix`
>   * #45100
> * Remove `CONFIG_NET_SOCKETS_POSIX_NAMES`
>   * this is mainly needed for testing network library code on `native_posix`
> * Support for [`sysconf()`](https://pubs.opengroup.org/onlinepubs/9699919799/functions/sysconf.html)
> * Address individual enhancment issues and bugfixes
> 
> ### Proposed change
> 
> * Replace most of the Zephyr posix headers with sanitized Newlib headers
>   * Newlib's POSIX headers are quite mature and use the prescribed feature test macros already
>   * PicoLibc's POSIX headers were derived from Newlib's so there is commonality gain
>   * 3rd party toolchains and libc's often are derived from Newlib, so it's a safe bet moving forward.
>   * Provide POSIX feature test macros in Zephyr (possibly enabled / disabled via Kconfig)
>   - [x] https://github.com/zephyrproject-rtos/zephyr/pull/51781
>   - [x] https://github.com/zephyrproject-rtos/zephyr/pull/52003
>   - [x] https://github.com/zephyrproject-rtos/zephyr/pull/52087
>   - [x] https://github.com/zephyrproject-rtos/zephyr/pull/52173
>   - [x] https://github.com/zephyrproject-rtos/zephyr/pull/52313  
>   - [x] https://github.com/zephyrproject-rtos/zephyr/pull/52316
> * Support for `sysconf()`
>   - [ ] https://github.com/zephyrproject-rtos/zephyr/issues/56670
> * `CONFIG_POSIX_API` include issue fixed by
>   - [x] https://github.com/zephyrproject-rtos/zephyr/pull/51771
> * Support automatic pthread stack allocation and object creation
>   * Add support for thread stack / object allocation to Zephyr
>   * This can be configured to allocate from a system thread pool and / or from the heap
> * Clean up layering issues between network and POSIX APIs
>   - [x] https://github.com/zephyrproject-rtos/zephyr/issues/54996
>   * Properly namespace Zephyr net symbols and constants (e.g. `FD_SET()`)
>   * Make the POSIX subsystem should be a thin wrapper around net counterparts
>   * Move some implementations into POSIX or common layer (e.g. `poll()`)
>   * Restructure some system calls so that there is a proper symbol in POSIX (e.g. `read()`, `write()`, `ioctl()`) that then calls a system call.
> * The POSIX Architecture and POSIX Subsystem are currently at odds
>   * Use `dlopen()` to open / link to the native libc, thus avoiding pollution of the global C namespace
[... 41 lines cut ...]
> ┌──────────┴┐       ├───────────┐
> │           │       │           │
> │   POSIX   ├───────►   NET     │
> │           │       │           │
> └───────────┘       └───────────┘
> ```
> 
> I'm thinking the correct name for `??` would be something like `zfdio`
> 
> Diagrams made with https://asciiflow.com/
> 
[... cut ...]
> 
> ### Dependencies
> 
> - [ ] Clean up layering issues between network and POSIX APIs
> - [ ] The POSIX Architecture and POSIX Subsystem are currently at odds
> - [ ] Remove CONFIG_NET_SOCKETS_POSIX_NAMES
> 
> ### Concerns and Unresolved Questions
> 
> ## Alternatives
> 
> There was a previous massive PR that attempted to address a number of the goals at once (#43987), but it is impossible to review in the current state and needs to be broken down and simplified in separate individual PRs.
>

abrodkin, 2023-06-04:
@cfriedt you updated names. but references are still to non-PSE pages ;)

cfriedt, 2023-06-04:
> @cfriedt you updated names. but references are still to non-PSE pages ;)

Actually, all three are referenced from "Conformance Test Suites" already. I was just trying to provide a reference to the higher scope specifications as well.

In any case, the information is available.

cfriedt, 2024-06-18:
**TSC Attention Required**

Please refer to 

[POSIX System Interfaces](https://docs.zephyrproject.org/latest/services/portability/posix/conformance/index.html#posix-system-interfaces-required)
[POSIX Application Environment Profiles](https://docs.zephyrproject.org/latest/services/portability/posix/aep/index.html)

**Why should the PRs below be considered for approval?**

* A number of members of the Zephyr project (especially those in automotive, other safety-centric areas), depend on POSIX for product development for the next several years.
* Many features were part of the release plan.
* Not *all* features were finished on time, but a *huge* amount were, and some had approvals but did not get merged in time
* Better to not leave architecturally important PRs on the table

**What are the PR's we are asking to merge?**

Allow toolchains / platforms to override POSIX Option Groups via Kconfig (mainly for maintainers, libc compatibility)
- [ ] https://github.com/zephyrproject-rtos/zephyr/pull/74184

Add stubs for the `POSIX_SIGNALS` Option Group (we do not yet have processes in Zephyr, so complete to the best of our ability, mainly just adding stubs)
- [ ] https://github.com/zephyrproject-rtos/zephyr/pull/74436

Implement `POSIX_DEVICE_IO` Option Group (depended on mmap, so a bit delayed. lays groundwork for zvfs, extremely useful in terms of reducing api-layer dependency cycles)
- [ ] https://github.com/zephyrproject-rtos/zephyr/pull/73978

Implement `POSIX_FD_MGMT` Option Group (`zvfs_poll()`, `zvfs_select()`, extremely useful in terms of reducing api-layer dependency cycles)
- [ ] https://github.com/zephyrproject-rtos/zephyr/pull/73978

**What are we leaving behind?**

**System Interfaces (Base Definitions)**
* `_POSIX_REALTIME_SIGNALS` (i.e. signalling individual threads)
  * very, *very* close to being done, but didn't have quite enough cycles to finish by last week
  * requires kernel changes, which probably *should not be rushed*
  * RT signals are actually a relatively simple and clean interface, compared to full POSIX signals
  * Note: ISO C and full POSIX signals really should be implemented on top of kernel-level RT signals
  * `pthread_kill()` is the one function in `_POSIX_THREADS` that depends on RT Signals
* `_POSIX_THREAD_SAFE_FUNCTIONS`
  * also very, *very* close to being done, but has changes requested, dependencies on posix device io, etc
  * We get by without this today, and it will be fine to get it in after LTS

**PSE51 - Minimal Realtime System Profile**
* N/A this is feature complete 🎉  (with caveats above)

**PSE52 - Realtime Controller System Profile**
* `POSIX_FILE_SYSTEM`
  * `subsys/fs` requires better `fdtable` integration (i.e. using per-file-descriptor offset, rather than per-file offset)
  * depends on device io, fd mgmt, etc
* `_POSIX_TRACE*`
  * not a lot of demand, as Zephyr already has a pretty good tracing subsystem
  * had considered implementing as stubs, but not enough cycles

**PSE53 - Dedicated Realtime System Profile**
* quite a few things relating to multi-process support (pipe, signal jump, spawn, sporadic server)
* would rather just do this properly after LTSv3

**Why did these PRs not make it in for Feature Freeze?**

* Difficulty making parallel PRs (i.e. there were dependency PRs that took some time, so PRs needed to be serialized)
* Limited developer / reviewer bandwidth. The people who are kind enough to review POSIX PRs have limited bandwidth, as do we all. Of course, more help making PRs is always appreciated.
* Some Options in particular are beyond the scope of what is supported in Zephyr, so could be stubs (e.g. mutex priorities)

**Post-release Bug-fixes and Stability-fixes (backports)**

* Optional POSIX features should be guarded by feature test macros (e.g. `#if defined(_POSIX_PRIORITY_SCHEDULING)`)
* Use correct header names (e.g. `sys/features.h` instead of `posix_features.h`
* Mitigate type definition conflicts between Newlib / Picolibc / Minimal libc
* Merge `signal.h` and `time.h` into minimal libc to reduce conflicts between Newlib / Picolibc / Minimal libc
* Do NOT blanket-define `_POSIX_THREADS` (`lib/posix/options/CMakeLists.txt`)

cfriedt, 2024-07-10:
Updated comment to reflect that the the posix device_io, signals, and fd_mgmt features were reverted.

This puts LTSv3 in a kind of dilapidated state from the POSIX & libc perspective.

closed 2024-12-08 by cfriedt