zephyrproject-rtos/zephyr #51214: enc28j60 appears to be unable to correctly determine network state

nordicjm, 2022-10-12:
**Describe the bug**
Testing some network samples with an enc28j60 connected to an stm32 nucleo_f401re board seems to have a working network connection, but the actual state of the network does not seem to be reflected in the driver. I'm not sure if this is a driver fault or a fault elsewhere.

**To Reproduce**
enc28j60 connected to nucleo_f401re board with the follow dts:
```
&spi1 {
        pinctrl-0 = <&spi1_sck_pa5 &spi1_miso_pa6 &spi1_mosi_pa7>;
        pinctrl-names = "default";
        cs-gpios = <&gpiob 6 (GPIO_ACTIVE_LOW | GPIO_PULL_UP)>;
        status = "okay";

        eth_dev: eth_dev@0 {
                compatible = "microchip,enc28j60";
                reg = <0x0>;
                local-mac-address = [00 00 00 01 02 03];
                /* Errata B7/1 specifies min 8Mhz, 20MHz max according to RM */
                spi-max-frequency = <10000000>;
                int-gpios = <&gpiob 8 GPIO_ACTIVE_LOW>;
        };
};
```
Then load `samples/subsys/mgmt/mcumgr/smp_svr` with `overlay-udp.conf`. Let the board start without a network cable attached, the output seems to make it look like there is an active network connection:
```
*** Booting Zephyr OS build zephyr-v3.2.0-414-gb219b5cba721  ***
[00:00:00.002,000] <inf> eth_enc28j60: ENC28J60 Initialized
[00:00:00.002,000] <dbg> eth_enc28j60: eth_enc28j60_tx: pkt 0x2000c7e4 (len 106)
[00:00:00.003,000] <dbg> eth_enc28j60: eth_enc28j60_tx: Tx successful
[00:00:00.003,000] <dbg> eth_enc28j60: eth_enc28j60_tx: pkt 0x2000c7e4 (len 106)
[00:00:00.004,000] <dbg> eth_enc28j60: eth_enc28j60_tx: Tx successful
[00:00:00.004,000] <dbg> eth_enc28j60: eth_enc28j60_tx: pkt 0x2000c7e4 (len 78)
[00:00:00.005,000] <dbg> eth_enc28j60: eth_enc28j60_tx: Tx successful
[00:00:00.005,000] <dbg> eth_enc28j60: eth_enc28j60_tx: pkt 0x2000c7e4 (len 62)
[00:00:00.006,000] <dbg> eth_enc28j60: eth_enc28j60_tx: Tx successful
[00:00:00.012,000] <dbg> eth_enc28j60: eth_enc28j60_tx: pkt 0x2000c7e4 (len 106)
[00:00:00.013,000] <dbg> eth_enc28j60: eth_enc28j60_tx: Tx successful
[00:00:00.013,000] <dbg> eth_enc28j60: eth_enc28j60_tx: pkt 0x2000c7e4 (len 78)
[00:00:00.014,000] <dbg> eth_enc28j60: eth_enc28j60_tx: Tx successful
[00:00:00.014,000] <inf> smp_udp_sample: Network connected
[00:00:00.014,000] <inf> smp_sample: build time: Oct 12 2022 13:51:32
[00:00:00.014,000] <inf> smp_udp: Started (IPv4)
[00:00:00.014,000] <inf> smp_udp: Started (IPv6)

uart:~$ [00:00:01.006,000] <dbg> eth_enc28j60: eth_enc28j60_tx: pkt 0x2000c7e4 (len 70)
[00:00:01.007,000] <dbg> eth_enc28j60: eth_enc28j60_tx: Tx successful
[00:00:02.007,000] <dbg> eth_enc28j60: eth_enc28j60_tx: pkt 0x2000c7e4 (len 70)
[00:00:02.008,000] <dbg> eth_enc28j60: eth_enc28j60_tx: Tx successful
[00:00:01.006,000] <dbg> eth_enc28j60: eth_enc28j60_tx: pkt 0x2000c7e4 (len 70)
[00:00:01.007,000] <dbg> eth_enc28j60: eth_enc28j60_tx: Tx successful
[00:00:02.007,000] <dbg> eth_enc28j60: eth_enc28j60_tx: pkt 0x2000c7e4 (len 70)
[00:00:02.008,000] <dbg> eth_enc28j60: eth_enc28j60_tx: Tx successful
```
However, there is no cable plugged in, and no LEDs on the ethernet port. If an ethernet cable is then plugged in, it does not seem to detect anything (no additional output appears) but the device can communicate on the network:
```
mcumgr --conntype udp --connstring=[192.168.1.220]:1337 echo test
test

[00:00:47.286,000] <dbg> eth_enc28j60: enc28j60_read_packet: Received packet of length 60
[00:00:47.286,000] <dbg> eth_enc28j60: eth_enc28j60_tx: pkt 0x2000c7a0 (len 42)
[00:00:47.287,000] <dbg> eth_enc28j60: eth_enc28j60_tx: Tx successful
[00:00:47.289,000] <dbg> eth_enc28j60: enc28j60_read_packet: Received packet of length 60
[00:00:47.289,000] <dbg> eth_enc28j60: eth_enc28j60_tx: pkt 0x2000c7e4 (len 59)
[00:00:47.290,000] <dbg> eth_enc28j60: eth_enc28j60_tx: Tx successful
[00:00:47.286,000] <dbg> eth_enc28j60: enc28j60_read_packet: Received packet of length 60
[00:00:47.286,000] <dbg> eth_enc28j60: eth_enc28j60_tx: pkt 0x2000c7a0 (len 42)
[00:00:47.287,000] <dbg> eth_enc28j60: eth_enc28j60_tx: Tx successful
[00:00:47.289,000] <dbg> eth_enc28j60: enc28j60_read_packet: Received packet of length 60
[00:00:47.289,000] <dbg> eth_enc28j60: eth_enc28j60_tx: pkt 0x2000c7e4 (len 59)
[00:00:47.290,000] <dbg> eth_enc28j60: eth_enc28j60_tx: Tx successful
```

**Expected behavior**
Network status to be reflected

**Impact**
Annoyance

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.15.0
 - Commit SHA or Version used: b219b5cba721b43a2c4fe71f5845f0abc852e7f6

**Additional context**
enc28j60 board being used is by "LC STUDIO"

erwango, 2022-10-13:
@nordicjm Have you been able to test with a non STM32 target?

nordicjm, 2022-10-13:
Yes, it is not stm32 specific, it's the driver which does not seem to support link status notifications. In theory can be marked as enhancement but I marked it as bug because I would consider it a bug for a network interface to be saying it is sending packets when it actually isn't because there is no network connection

rlubos, 2022-10-19:
Ethernet drivers should report their carrier status (essentially meaning whether the cable is plugged in or not) with `net_eth_carrier_on/off()` API. The thing is, not all drivers do that, and for those who doesn't the network inteface state reporting will not work correctly (i. e. the interface will be marked as "UP" as soon as it was brought up by the administrator).  `enc28j60 ` is one of those drivers - the cable detection needs to be implemented in the driver.

The issue should be assigned to the driver maintainer, I don't have access to the hardware to verify potential improvements.

laurenmurphyx64, 2022-10-25:
@jvasanth1 Would you know someone from Microchip who can help on this issue?

PR #51729 (2022-10-31): drivers: ethernet: enc28j60: Added carrier status update functionality
Expanded the enc28j60 driver to update its carrier status whenever the Ethernet cable is plugged in or out. I used the enc424j600 driver as reference, as this driver already included this functionality. The driver will now take the interface offline whenever the cable is not plugged in and stop sending packets. The respective events are also generated for use in the application. I tested it locally on my nRF52840 based gateway board using the enc28j60 network chip. Everything worked as expected.

Fixes #51214
commit e35ce374 (2022-10-31): drivers: ethernet: enc28j60: Added carrier status update functionality

Expanded the enc28j60 driver to update its carrier status
whenever the Ethernet cable is plugged in or out. I used the
enc424j600 driver as reference, as this driver already
included this functionality. The driver will now take
the interface offline whenever the cable is not plugged in
and stop sending packets. The respective events are also
generated for use in the application. I tested it locally
on my nRF52840 based gateway board using the enc28j60 network
chip. Everything worked as expected.

Fixes #51214
closed 2022-10-31 by carlescufi