zephyrproject-rtos/zephyr #51233: samples/boards/up_squared/gpio_counter run failed

mengxianglinx, 2022-10-13:
**Describe the bug**
Test case in samples/boards/up_squared/gpio_counter runs failed.

**To Reproduce**
Steps to reproduce the behavior:
1. Change BIOS settings as samples/boards/up_squared/gpio_counter/src/main.c describes.
2. make
3. Boot up_squared board over network
4. See error
>>Checking Media Presence......
>>Media Present......
Downloading NBP file...
Succeed to download NBP file.
*** Zephyr EFI Loader ***
RSDP found at 0x79c34000
Zeroing 4816288 bytes of memory at 0x106000
Copying 32768 data bytes to 0x1000 from image offset
Copying 24576 data bytes to 0x100000 from image offset 32768
Copying 2110048 data bytes to 0x59dda0 from image offset 57344
Jumping to Entry Point: 0x1137 (48 31 c0 48 31 d2 48)
*** Booting Zephyr OS build zephyr-v3.2.0-412-g3d85ae37b6ee  ***
counter: 0x0

**Environment (please complete the following information):**
 - OS: (Linux)
 - Toolchain (Zephyr SDK 0.15.0)
 - Commit SHA or Version used 3d85ae37b6ee

enjiamai, 2022-11-04:
After given the HAT pin16 an input signal, I saw following result:

```
>>Checking Media Presence......
>>Media Present......
 Downloading NBP file...

  Succeed to download NBP file.
*** Zephyr EFI Loader ***
RSDP found at 0x79c34000
Zeroing 4816288 bytes of memory at 0x106000
Copying 32768 data bytes to 0x1000 from image offset 
Copying 24576 data bytes to 0x100000 from image offset 32768
Copying 2110048 data bytes to 0x59dda0 from image offset 57344
Jumping to Entry Point: 0x1137 (48 31 c0 48 31 d2 48)
*** Booting Zephyr OS build zephyr-v3.2.0-1233-g39bb5cd06a08 ***
counter: 0x0
counter: 0x1
counter: 0x2
counter: 0x3
counter: 0x4
counter: 0x5
counter: 0x6
counter: 0x7
counter: 0x8
counter: 0x9
counter: 0xa
counter: 0xb
counter: 0xc
counter: 0xd
counter: 0xe
counter: 0xf
counter: 0x0
counter: 0x1
counter: 0x2
counter: 0x3
counter: 0x4
counter: 0x5
counter: 0x6
counter: 0x7
counter: 0x8
counter: 0x9
counter: 0xa
counter: 0xb
counter: 0xc
counter: 0xd
counter: 0xe
counter: 0xf
counter: 0x0
counter: 0x1
counter: 0x2
```

It works when I manually trigger the input. I think this is a expected result, @dcpleung , could you please help to confirm? Thank you very much!

dcpleung, 2022-11-04:
I don't have an UP2 board to test this. But the output looks correct.

enjiamai, 2022-11-07:
Thanks for comments from @dcpleung . A manually input triggering works. So I think this is not a bug. Double checking with a output loopback.

enjiamai, 2022-11-14:
Close it because this is triggered successfully manually on UP2.

closed 2022-11-14 by enjiamai