zephyrproject-rtos/zephyr #51234: it8xxx2_evb: The testcase tests/kernel/sleep/failed to run.

Kunli123, 2022-10-13:
**Describe the bug**
The testcase tests/kernel/sleep/ failed to run on the it8xxx2_evb

**To Reproduce**
twister  -p it8xxx2_evb --device-testing --west-flash="../../itetool/loader.sh" --device-serial=/dev/ttyUSB0 -T tests/kernel/sleep/ -v

**Expected behavior**
A clear and concise description of what you expected to happen.

**Logs and console output**
```

***** delaying boot 1ms (per build configuration) *****
*** Booting Zephyr OS build zephyr-v3.2.0-404-ge852247de33f  (delayed boot 1ms) ***
Running TESTSUITE sleep
===================================================================
START - test_sleep
Kernel objects initialized
Test thread started: id = 0x80101108
Helper thread started: id = 0x80101060
Testing normal expiration of k_sleep()
Testing: test thread sleep + helper thread wakeup test
Testing: test thread sleep + isr offload wakeup test
Testing: test thread sleep + main wakeup test thread
Testing kernel k_sleep()
PASS - test_sleep in 2.035 seconds
===================================================================
START - test_sleep_forever
Kernel objects initialized
PASS - test_sleep_forever in 0.003 seconds
===================================================================
START - test_usleep
elapsed_ms = 2167
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sleep/src/usleep.c:95: sleep_test_usleep: (elapsed_ms <= UPPER_BOUND_MS is false)
overslept
FAIL - test_usleep in 21.683 seconds
===================================================================
TESTSUITE sleep failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -  66.67% [sleep]: pass = 2, fail = 1, skip = 0, total = 3 duration = 23.721 seconds
- PASS - [sleep.test_sleep] duration = 2.035 seconds
- PASS - [sleep.test_sleep_forever] duration = 0.003 seconds
- FAIL - [sleep.test_usleep] duration = 21.683 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: 9c955129b21eb68ec9b0d6d00d7d91ee
PROJECT EXECUTION FAILED

```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.15.0
 - Commit: b663008d0c03c025191bc67956feaa57

Kunli123, 2022-10-13:
I used the git bisect tool to find that there was a problem with the commite.
commite：[c32f376e9977d4904db86b72c4a4818fc7942071]

hunterhu3000, 2022-10-14:
see https://github.com/zephyrproject-rtos/zephyr/pull/47084
It seems the timer is still not accurate. The sleep time upper bound of this test case is 2000, sometimes the test result will exceed this value.
If we restore before  commite：[https://github.com/zephyrproject-rtos/zephyr/commit/c32f376e9977d4904db86b72c4a4818fc7942071], twister will pass, but the west will fail

hunterhu3000, 2022-10-14:
Maybe we need to further reduce the SYS_CLOCK_TICKS_PER_SEC value to 3800 or smaller.

hunterhu3000, 2022-10-25:
The issue is not seen after commit 3cef8e68df3ede8dc9b41e76953ba74f87b2da7a

closed 2022-10-25 by hunterhu3000