zephyrproject-rtos/zephyr #51294: ztest: Broken tests in main branch due to API-breaking change `ZTEST_FAIL_ON_ASSUME`

alwa-nordic, 2022-10-14:
The commit a40a2f5c5077d2349d8442a2bdd9b8551ebcc04e does not pass CI. It's now on main branch and is failing CI for other PRs. It should be reverted. It can be reapplied again with fixes for tests that it breaks with this API change.

That commit introduces a new config `ZTEST_FAIL_ON_ASSUME`. The config is `default y` and essentially makes the `zassume`-functions into `zassert`. But I though the whole point of assumptions is that tests can be skipped if they don't apply, so I don't understand the motivation for this change. I propose reverting and dropping this change.

To reproduce:
```bash
./scripts/twister -T tests/bluetooth/host/keys/bt_keys_get_addr
```

ppryga-nordic, 2022-10-14:
It blocks PR: #51256.

alwa-nordic, 2022-10-14:
> But I though the whole point of assumptions is that tests can be skipped if they don't apply, so I don't understand the motivation for this change.

I though some more, and it makes sense that there are different reasons for skipping a test.

It makes sense to skip tests that you know will fail if some other test fails. In those cases it makes sense to raise a warning if there are skipped tests but no failed ones. See the following the following example.

https://github.com/zephyrproject-rtos/zephyr/blob/2037b20db52a40d8639ea25647b362e4a4fa15e4/tests/bluetooth/uuid/src/test_bt_uuid_to_str.c#L33

But another reason to skip a test is that the test want's to say that it's not compiled with the needed kconfigs to be able to run. It would then make sense to run the test suite with different kconfigs and take the union of the results. A PASS and a SKIP on the same test should union to PASS.

I need to have a look at the failing test to see how it's using `assume`.

alwa-nordic, 2022-10-14:
It seems there is a PR that changes the test: https://github.com/zephyrproject-rtos/zephyr/pull/51293

ahmedmoheb-nordic, 2022-10-14:
> UPDATE: It seems the test that fails is using zassume in some entirely different way. It runs all tests twice and skips them exactly half the time. I'm not sure why it's set up like that. We may want to look into it.

This comment clarify what is different with that particular test https://github.com/zephyrproject-rtos/zephyr/pull/50633#discussion_r981153372

yperess, 2022-10-14:
Let me clarify, the point of assume is to avoid having too many things fail which makes it hard to know what the bug is. For example, in a test for some code with dependencies, one should use the assume API on calls to the dependency instead of asserts. It's a way of saying "I need this to work, but it's not my responsibility to test". The reason we're dating the run is because these gaps started happening silently. Someone would introduce a breaking change into code that wasn't tested. That code did however have an assume in another module. So the idea is that if you have failed assumptions, you do have a bug, is just somewhere else. If you have 0 failed tests, then there's a gap in testing.

I'll look into this issue today as my top priority.

yperess, 2022-10-14:
> I though some more, and it makes sense that there are different reasons for skipping a test.
>
> It makes sense to skip tests that you know will fail if some other test fails. In those cases it makes [...]

Correct, this change was specific for the assume API. You can still skip tests using ztest_test_skip() or the macro Z_TEST_SKIP_IFDEF(). Those will not fail the run. I would argue also that with the test API using the linker now, a test that is missing the Kconfig (statically known to not be applicable) should be in a separate .c file and just not even linked into the test binary.

alwa-nordic, 2022-10-14:
Ok. The above mentioned PR has now been merged. This issue is no longer needs a bugfix so I will close it.

closed 2022-10-14 by alwa-nordic

yperess, 2022-10-14:
> Ok. The above mentioned PR has now been merged. This issue is no longer needs a bugfix so I will close it.

Thanks, I just came into the office. What a nice surprise