zephyrproject-rtos/zephyr #51298: [wip]API: Retention register API

nvlsianpu, 2022-10-14:
**Is your feature request related to a problem? Please describe.**
Retention registers are registers which are able to retains values between system reboots or other events which causes system RAM invalidation. Usually there are just a few retention register in a SoC if any.

Retention registers are not a backup-RAM. Better to access the backup-RAM like the rest of system memory, which is most efective.

Retention API assumes that locations can be overwritten without page erases, meaning flash is excluded. Open question about atomicity of read/write operations

**Describe the solution you'd like**
It might be worth to introduce an API which is dedicated for retention register like entities.

There was also idea of using syscon API for that.
This API is very close to the existing syscon

**Describe alternatives you've considered**
There was also idea of using syscon API for that.

After Looking at the syscon API I see one problem with that - It is too low level. I'm expecting that we are using vendor provided hardware abstraction layer for manipulate peripherals MIMO register- syscon is an interface for manipulate registers for given device driver. In reality we won't (and it is unwanted IMO) give any possibility for manipulation on register level of all the peripheral - instead just a few like retention registers. Creating a fake syscon like register device might be an option, but real addressing isn't required for that. I'm doubting in that It will be proper solution.

With solution like that
We need however to express which retained register area to use to store expected information
``` DTS
zephyr,SOMETHIN-retention-addr = regret1
```

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

nvlsianpu, 2022-10-14:
@bjarki-trackunit I have nothing against high level abstraction. I think need to considered this having in mind that solution should give possibility for access given register (maybe only these within SoCs) without much overhead.

nordicjm, 2023-02-13:
@bjarki-trackunit Have a look at https://github.com/zephyrproject-rtos/zephyr/compare/main...nordicjm:zephyr:retention and see what you think, I've currently got 2 backends implemented, I have been trying to implement a `noload` RAM one but struggling on how to get it implemented in the linker, somehow it'd need to have memory at specific regions (for specific lengths) reserved and not sure how to go about that. I think I might have to have it whereby it goes as a separate dts node and the existing sram region and Kconfig is reduced properly.

nordicjm, 2023-04-27:
Fixed with #55125

closed 2023-04-27 by nordicjm