zephyrproject-rtos/zephyr #51317: Confusing license references in nios2f-zephyr

rahulmohang, 2022-10-16:
In the file, 'https://github.com/zephyrproject-rtos/zephyr/tree/main/soc/nios2/nios2f-zephyr/cpu/ghrd_10m50da.qpf', more than one license is referred to. 

License text:
"
Your use of Altera Corporation's design tools, logic functions
and other software and tools, and its AMPP partner logic
functions, and any output files from any of the foregoing
(including device programming or simulation files), and any
associated documentation or information are expressly subject
to the terms and conditions of the Altera Program License
Subscription Agreement, the Altera Quartus Prime License Agreement,
the Altera MegaCore Function License Agreement, or other
applicable license agreement, including, without limitation,
that your use is for the sole purpose of programming logic
devices manufactured by Altera and sold by Altera or its
authorized distributors. Please refer to the applicable
agreement for further details.
"

The text of the reference to licenses does not give a clear indication of the license.

alpianon, 2022-11-21:
My understanding is that some express license to redistribute the software should be granted by Altera to the Zephyr project

alpianon, 2023-01-27:
Is there any progress on this? Could you remove the stale label? Thanks!

alpianon, 2023-03-29:
still no progress on this?

alpianon, 2023-05-30:
is anyone taking care of this?

alpianon, 2023-07-31:
anyone?

closed 2023-10-15 by github-actions[bot]