zephyrproject-rtos/zephyr #51319: Additional MODBUS Function Codes

DaAwesomeP, 2022-10-17:
Currently the MODBUS server is limited to the following function codes:
https://github.com/zephyrproject-rtos/zephyr/blob/51af614f88eae21dc4baa3cbed03e1de81b053df/subsys/modbus/modbus_server.c#L960-L1002

This switch case calls functions to do some basic validation and eventually call user functions based on a struct here:
https://github.com/zephyrproject-rtos/zephyr/blob/ae9de202220ce51e1ed61157f73d3800f84c6b06/include/zephyr/modbus/modbus.h#L332-L361

Since there are many more possible function codes [in the MODBUS specification](https://www.modbus.org/specs.php) than are currently supported in Zephyr:
1. Should the callbacks for these additional codes be abstracted in some way so that the user can make use of them with their own validator and parser?
2. Should the rest of the function codes be implemented but in a way that the user can configure which function codes are actually supported by their application (similar to the `CONFIG_MODBUS_FC08_DIAGNOSTIC` option)?

I think the best option is 2), as this will match the current existing API. Users can expect basic MODBUS packet structure/address bounds checking for all of the currently implemented function codes, so it would be inconsistent to not offer this validation for some other function code handlers.

We are specifically interested in implementing function codes 20 (`0x14`), 21 (`0x15`), and 14 (`0x10E`) in our application for eventual remote firmware updates and status monitoring over RTU.

lindblandro, 2023-01-26:
I'd also prefer option 2 since it would also allow vendor specfic function codes to be defined by the application in the protocol defined ranges 65-72 and 100-110.. I have a long overdue branch that implements this for RTUs but I haven't really looked at TCP side yet. What confuses me are the "raw" callbacks that seem to be needed for Modbus TCP, but I can't really find any details about what they are or why they exist. In any case, adding a feature for allowing users to e.g. provide a list of `(fc, callback)` pairs should be fairly straightforward to implement. 

Maybe @jfischer-no can provide more details on the raw feature for TCP  and could user-definable codes be used to support some of the missing Modbus features?

jfischer-no, 2023-01-27:
>    Should the callbacks for these additional codes be abstracted in some way so that the user can make use of them with their own validator and parser?
 
For the client side we can probably expose API like mbc_send_cmd_custom(). Server side could use user FC table or a callback called if there is no match in the core FC table, maybe per interface.

>   Should the rest of the function codes be implemented but in a way that the user can configure which function codes are actually supported by their application (similar to the CONFIG_MODBUS_FC08_DIAGNOSTIC option)?

What is rest? Zephyr Modbus supports all mandatory FC. What is missing in your opinion?

> Maybe @jfischer-no can provide more details on the raw feature for TCP and could user-definable codes be used to support some of the missing Modbus features?

You can see RAW as an intermediate backend, like serial line. The motivation was not to pull all the TCP + TLS stuff into the Modbus support and keep code clean and simple. There are samples how to implement TCP server and TCP gateway using RAW interface. So RAW interface is not limited to TCP or sockets,  user has options to implement other transport backends using RAW and provide TLS if needed. FC handling is actually the other side of Modbus support and should not be mixed with backends.

DaAwesomeP, 2023-01-27:
> What is rest? Zephyr Modbus supports all mandatory FC. What is missing in your opinion?

@jfischer-no I am specifically interested in FCs 20 (0x14), 21 (0x15), and 14 (0x10E), but I think a way to specify custom sending handlers may cover everything most easily.

lindblandro, 2023-01-28:
> You can see RAW as an intermediate backend, like serial line. The motivation was not to pull all the TCP + TLS stuff into the Modbus support and keep code clean and simple. There are samples how to [...]

Ok that makes a lot of sense and I suspected it had something to do with that. The docs could clarify this a bit from an architectural standpoint as there's now RTU / RAW and the absence of explicit TCP backend jumps out of the code. The RAW backend at first sight seems to be relevant for implementing custom function codes, but it's actually a completely separate matter.

PR #54248 (2023-09-29): modbus: add support for defining custom functions
Enables support for custom function codes. Modbus specification allows vendor specific function codes in the range 65-72 & 100-110 [1] and this feature allows users to implement custom logic for those codes. Additionally, since the Zephyr Modbus stack doesn't implement all defined Modbus fcs this feature allows users to add support for codes outside the basic register reading / writing functionality offered by Zephyr.

Custom function codes can be added on a per-interface basis and the handler structures are allocated by the caller. The TCP ser [... cut]
closed 2023-09-29 by carlescufi