zephyrproject-rtos/zephyr #51342: Bluetooth ISO

carlst, 2022-10-17:
**Describe the bug**
Extra stream_sent callback to `bt_audio_stream_send` after seq_num 16-bit rollover.

This can cause client to attempt to send an additional data packet to ISO stream.

**To Reproduce**
Steps to reproduce the behavior:
1. Build `samples/bluetooth/broadcast_audio_source`
2. Run until seq_num exceeds 65536 (~11 minutes)
3. One extra stream_sent callback occurs

**Expected behavior**
Expect only a single callback for each seq_num in calls to `bt_audio_stream_send`

**Impact**
Low priority

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: Segger Embedded Studio
 - Board: nrf52840DK
 - Commit SHA b0e76dbf359da092ac65e980815a87902ab1958a

**Additional context**

May be related to:
- handling of 16-bit and > 16-bit packet sequence numbers in `subsys/bluetooth/controller/ll_sw/hci/hci.c` `hci_iso_handle` 
- packet acknowledgement generation
- #44244

Thalley, 2022-10-19:
I think this might be a controller issue, and not a host issue, as the host simply sends the PSN to the controller (cast from 32-bit to 16-bit). There is nothing in the host that should provoke this issue, and the sent callback is triggered on a `num_complete_packets` event from the controller. 

@cvinayak I think this may be more up your allay than mine, so I'm assigning this to you.

Thalley, 2022-10-19:
@carlst Can you provide information on how you detected the double callback? Any logs with timestamps or anything you can provide?

carlst, 2022-10-19:
@Thalley  Yes, please see attached files (with logging in hci.c and main.c) and logs. The issue is that `tx->payload_count` becomes set to 65535 twice in `subsys/bluetooth/controller/ll_sw/hci/hci.c` `hci_iso_handle()` when `stream->pkt_seq_num` rolls over, causing 2 back-to-back sent callbacks.

broadcast_audio_source API has recently changed so the attached log refers to commit 9f652ea04a6deba0b33c0b34b861195ae967cd7c and not previous commit.

[broadcast_audio_source_log_20221019.zip](https://github.com/zephyrproject-rtos/zephyr/files/9824166/broadcast_audio_source_log_20221019.zip)

Thalley, 2022-10-20:
Thanks @carlst I think that points towards a controller issue :)

cvinayak, 2022-10-26:
@carlst Please try using `<` instead of `<=`  in the below line:

https://github.com/zephyrproject-rtos/zephyr/blob/c707f1ed28a83d9a0715ca08743d0e2112a3fa95/subsys/bluetooth/controller/hci/hci.c#L5770-L5771

carlst, 2022-10-27:
@cvinayak No change when `<` is used instead of `<=`.

Handling 39-bit sequence numbers at and after 16-bit sequence number rollovers while allowing for variable Tx buffering may require some additional state information(?). I will check further.

cvinayak, 2022-10-27:
> @cvinayak No change when `<` is used instead of `<=`.
> 
> Handling 39-bit sequence numbers at and after 16-bit sequence number rollovers while allowing for variable Tx buffering may require some additional state information(?). I will check further.

Anyway the code there in hci.c is a workaround until ISOAL module is integrated, i.e. the correct thing to do is pass the ISO data from HCI into ISOAL and thereafter enqueue towards lower link layer. Currently, the code there is bypassing ISOAL and directly placing the ISO data with the hack to fill the packet counter.

carlst, 2022-10-27:
Sorry, I imagined this code had survived ISOAL integration. I will handle in the application for now, and close the issue.

cvinayak, 2022-11-02:
> Sorry, I imagined this code had survived ISOAL integration. I will handle in the application for now, and close the issue.

No worries, you are free to correct the implementation; ISOAL module upstream is feature complete (unframed and framed). I have just not gotten yet to integrating it.

I would take inspiration from how Connect ISO code path uses the ISOAL interface:

https://github.com/zephyrproject-rtos/zephyr/blob/3a3cfbb652b29e9f39a38fe6febf29e1570d32bc/subsys/bluetooth/controller/hci/hci.c#L5647-L5699

Re-opening this issue, so that Broadcast ISO uses ISOAL interfaces.

carlst, 2022-11-07:
PR #52034 is a draft. It is not yet complete, and fails the automatic PR checks. Currently in testing with broadcast_audio_source/broadcast_audio_sink samples.

cvinayak, 2022-11-08:
> PR #52034 is a draft. It is not yet complete, and fails the automatic PR checks. Currently in testing with broadcast_audio_source/broadcast_audio_sink samples.

Thanks for the PR. Will review it sometime this week.

Thalley, 2023-02-24:
@carlst @cvinayak is this issue still relevant?

carlst, 2023-02-24:
@Thalley @cvinayak No. This issue was subsumed by #52034

closed 2023-02-24 by carlst