zephyrproject-rtos/zephyr #51347: CSTD should not be set globally

felipe-iar, 2022-10-17:
**Is your enhancement proposal related to a problem? Please describe.**
As default, Zephyr assumes `c99` for the global `CSTD` property. This property is propagated to the compiler flags through `zephyr_compile_options()` in the top-level "CMakeLists.txt" file:

https://github.com/zephyrproject-rtos/zephyr/blob/0432d0a891dba55492bf45d2e89d7db6ce2f0636/CMakeLists.txt#L882-L888

The immediate side effect is that, if the compiler does not set `cstd` in the "compiler_flags.cmake" module, the option will be propagated to the compiler anyway.
https://github.com/zephyrproject-rtos/zephyr/blob/822c37745e189c3505c7dc45c405ee16135be1e7/cmake/compiler/compiler_flags_template.cmake#L46

The `c99` string alone in the compiler flags becomes recognized as a second source file to be compiled which causes a fatal error.

**Describe the solution you'd like**
Not every compiler provides an option for selecting previous C Language Standards. Portability-wise, this option should not be mandatory. Or `CSTD` should not be set globally.

Ideally the build scripts should detect which language support level is offered by the compiler.

**Describe alternatives you've considered**
For now, a (dirty) workaround was to trick CMake so `c99` becomes a symbol.
```cmake
set_compiler_property(PROPERTY cstd -DC_LANG_STD=)
```

**Additional context**
This relates to portability issues that were discovered recently in porting work for the IAR C/C++ Compiler.

felipe-iar, 2022-10-17:
@tejlmand
@stephanosio

vindicatorr, 2023-02-20:
I only want to hop in on this because I only just now discovered `CSTD`.
I was having build issues with `error: 'asm' undeclared (first use in this function)`.
I found `-std=c99` being set and didn't know how until I went exploring.
There was nothing in the documentation about it.
When I `set_property(GLOBAL PROPERTY CSTD $ENV{CSTD})` (CSTD=gnu99), I got past those errors.