zephyrproject-rtos/zephyr #51349: Turn power domains on/off directly

keith-zephyr, 2022-10-17:
**Is your feature request related to a problem? Please describe.**
In our application, we have some external temperature sensors that get turned off based on the power state of an external component.

[image]

In this application, the power rail turns off/on asynchronously, and is not under the control of the Zephyr application.

The Zephyr application does monitor the power good signal as a GPIO input.

We're trying to model this using Zephyr's power domains.  We would like to create a power domain for the external "power rail".  If the power rail turns off, we want to notify the temperature sensor drivers, so the drivers don't attempt to communicate with an un-powered device.

The current power domain examples seem to be structured for a bottom-up management of the power domain.  Each device in the power domain calls `pm_device_runtime_get()` while the device is use.  If all devices in the domain are idle, the domain state is changed to suspended.

We're currently getting around this by using `pm_device_runtime_get/put()`  in our power rail driver:

```
power_rail_gpio_callback()
{
    if (power_good) {
        pm_device_runtime_get();
    } else {
        pm_device_runtime_put();
    }
}

power_rail_pm_action()
{
	switch (action) {
	case PM_DEVICE_ACTION_RESUME:
		pm_device_children_action_run(TURN_ON);
	case PM_DEVICE_ACTION_SUSPEND:
		pm_device_children_action_run(TURN_OFF);
}
```

**Describe the solution you'd like**

While using `pm_device_runtime_get/put()` at the power domain level is working for now, I don't think this is an intended use case.  If a temperature sensor in the domain makes a call to `pm_device_runtime_get/put()`, things will break due to the counting nature of get/put.

An API that let us directly set the power domain state is one idea we've considered.

I'm open to other ideas on how to model this.

gmarull, 2022-10-18:
cc: @JordanYates

JordanYates, 2022-10-18:
> We're trying to model this using Zephyr's power domains. We would like to create a power domain for the external "power rail". If the power rail turns off, we want to notify the temperature sensor drivers, so the drivers don't attempt to communicate with an un-powered device.

Interesting use-case, what happens when the power rail goes down halfway through an I2C transaction? Is this even feasible to properly support?

dawidni, 2022-10-18:
Actually, in our case, the SCL and SDA lines are also pulled up to the Power Rail so the I2C transaction will just fail.

semihalf-niedzwiecki-dawid, 2022-10-18:
Actually, in our case, the SCL and SDA lines are also pulled up to the Power Rail so the I2C transaction will just fail.

JordanYates, 2022-10-19:
My suggestion would be to model it as a power domain that fails to power up if it isn't already on, something like:
```
atomic_t power_up;

power_rail_gpio_callback()
{
    if (power_good) {
        pm_device_children_action_run(TURN_ON);
    } else {
        /* Possibility for weird behaviours here? Child devices don't transition through SUSPENDED? */
        pm_device_children_action_run(TURN_OFF);
    }
    power_up = power_good;
}

power_rail_pm_action()
{
	int rc = 0;
	switch (action) {
	case PM_DEVICE_ACTION_TURN_ON:
	case PM_DEVICE_ACTION_TURN_OFF:
		return -ENOTSUP;
	case PM_DEVICE_ACTION_RESUME:
		if (!power_up) {
			return -EAGAIN;
		}
		break;
	default:
		break;
	}
	return rc;
}
```

PR #61166 (2023-11-01): drivers: power_domain: Introduce a gpio monitor driver
Power rails of some peripherals are controlled externally.
This is a case in embedded controllers, where the power of
some I2C devices are managed by the main application
processor.

To ensure that zephyr drivers access the devices where is
powered on, introduce a "monitoring" power domain. It works
by registering interrupt handler with gpio a pin, so that
when power state changes, it will notify relevant drivers.

Fixes: #51349
commit 06cfbd41 (2023-11-01): drivers: power_domain: Introduce a gpio monitor driver

Power rails of some peripherals are controlled externally.
This is a case in embedded controllers, where the power of
some I2C devices are managed by the main application
processor.

To ensure that zephyr drivers access the devices where is
powered on, introduce a "monitoring" power domain. It works
by registering interrupt handler with gpio a pin, so that
when power state changes, it will notify relevant drivers.

Additionaly add CONFIG_POWER_DOMAIN_INIT_PRIORITY to replace
harcoded init priority.

Fixes: #51349
closed 2022-12-19 by keith-zephyr