zephyrproject-rtos/zephyr #51351: I2C: ESP32 driver does not support longer clock stretching

martinjaeger, 2022-10-17:
**Describe the bug**

I'm trying to make the ESP32-C3 talk to a TI bq76952 battery management IC (full hardware design [here](https://github.com/LibreSolar/bms-c1)). Exactly the same firmware works fine with an STM32G0B1. With the ESP32-C3 almost all I2C communication attempts fail.

Below an example of a failed I2C transaction to read the device ID:

[image]

The same command looks like this with the STM32.

[image]

It can be observed that in both cases, the SCL is held low by the bq76952 towards the end (most probably because the device is busy with internal calculations). However, for the ESP32-C3 we suddenly see an 0xFF on the bus, which was not requested and does not belong there, whereas in case of the STM32 the communication continues after the clock stretching has finished.

**To Reproduce**

Custom board with ESP32-C3 and bq76952 is required, it will be difficult to reproduce with standard boards.

**Expected behavior**

The I2C communication handles clock stretching properly.

**Impact**

Communication with that chip does not work.

**Environment (please complete the following information):**

 - Commit SHA: f37ebf48ece1892771719321fca95156fd2e50c6

martinjaeger, 2022-10-17:
I found a hacky solution to increase the timeout and fix the issue. Will come up with a PR in the next days.

martinjaeger, 2022-10-18:
@laurenmurphyx64 why do you think this is an enhancement? With the driver as it is now it is not possible to talk to the chip via I2C at all, so that's clearly a bug in my opinion.

laurenmurphyx64, 2022-10-18:
@martinjaeger If I remember correctly, during the bug triage meeting this week folks were of the opinion that the expected behavior described in the bug was optional / not supported by Zephyr. It's possible we went too fast categorizing this one, though. @stephanosio Thoughts?

martinjaeger, 2022-10-18:
Yes, we don't have an API to configure the clock stretching / I2C timing behavior. But drivers should set values which are not configurable to conservative default values in such cases. Also configuration options can be added via devicetree without having a dedicated API (which I have done in the related PR).

closed 2022-10-31 by stephanosio

balancer2024, 2024-05-12:
hi @martinjaeger , I stumbled upon this post as I think I am experiencing similar issue. Problem is I am using the arduino Wire library on ESP32 to communicate to BQ76952. It seems applying a longer timeout on the Wire library is not helping. Can you please point me to the exact portion of your code where you applied the timeout so I can try to replicate on the arduino platform?

martinjaeger, 2024-05-13:
> hi @martinjaeger , I stumbled upon this post as I think I am experiencing similar issue. Problem is I am using the arduino Wire library on ESP32 to communicate to BQ76952. It seems applying a longer [...]

The commit with the reqired changes is linked above. The relevant part in the current driver implementation is here:
https://github.com/zephyrproject-rtos/zephyr/blob/961959fc5f2ba913a80af6e92ea493b373f90253/drivers/i2c/i2c_esp32.c#L270

PR #51405 (2022-10-31): drivers: i2c: esp32: set timeout to allow clock stretching
The ESP32 series MCUs allow to set a timeout which triggers an error if the SCL line is unchanged for the specified amount of time.

By default, the ESP-IDF HAL sets the timeout to an arbitrary value of 10 times the bus cycle.

This is not sufficient for chips like the TI bq76952, which pulls the SCL line low (clock stretching) for several 100 µs.

The timeout should also not be dependent on the chosen bitrate, as it is defined by the time a chip needs for internal calculation before it can provide requested data or continue communicat [... cut]
commit d5168a8d (2022-10-31): drivers: i2c: esp32: set timeout to allow clock stretching

The ESP32 series MCUs allow to set a timeout which triggers an error
if the SCL line is unchanged for the specified amount of time.

By default, the ESP-IDF HAL sets the timeout to an arbitrary value of
10 times the bus cycle.

This is not sufficient for chips like the TI bq76952, which pulls the
SCL line low (clock stretching) for several 100 µs.

The timeout should also not be dependent on the chosen bitrate, as it
is defined by the time a chip needs for internal calculation before it
can provide requested data or continue communica [... cut]