zephyrproject-rtos/zephyr #51359: Ext2 file system support

fkokosinski, 2022-10-18:
**Is your enhancement proposal related to a problem? Please describe.**
Adding support for the ext2 file system in Zephyr.

**Describe the solution you'd like**
Implementing a new file system in the Zephyr tree along with new tests and samples.
Implementation of ext2 that will be contributed is based on ext2 specification
and many other resources that describe it (see below).

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

### Planned PRs
- [ ] Moving common tests out of LittleFS
- [ ] Read-only file system
- [ ] Read-only file system – documentation
- [ ] Write operations for ext2
- [ ] Write operations for ext2 – documentation
- [ ] Tests in Zephyr framework
- [ ] Handling large storage devices
- [ ] Handling large files

### Resources
1. https://www.nongnu.org/ext2-doc/ext2.html
2. http://www.science.smith.edu/~nhowe/262/oldlabs/ext2.html
3. https://www.kernel.org/doc/html/latest/filesystems/ext2.html

nordicjm, 2022-10-19:
Any particular reason to pick ext2 over ext4?

fzdobylak, 2022-10-20:
There are few reasons why I have chosen the older version of ext.
1. It is simpler hence it is easier to implement from scratch. (It can be later updated to ext3 and ext4.)
2. It uses less resources. (In ext4 structures can be bigger, and we also have to save the data of the journaling system.) Many supported hardware devices have little amount of memory hence the simpler file system seems like a reasonable choice.
3. Journaling can be harmful for flash memory (which will be the first type of storage that ext2 is implemented for.)

nordicjm, 2022-10-20:
> Journaling can be harmful for flash memory (which will be the first type of storage that ext2 is implemented for.)

Note that ext4 supports a mode without using a journal

de-nordic, 2022-10-20:
What problem does the inclusion of ext2 to embedded device with limited RAM solves?

nashif, 2022-11-20:
new filesystem is a new feature request.

snehalv2002, 2023-06-20:
is anyone working on this

fkokosinski, 2023-06-20:
Hi @snehalv2002!

Yes, we have a PR that adds Ext2 to Zephyr here: https://github.com/zephyrproject-rtos/zephyr/pull/55152.

snehalv2002, 2023-06-20:
Is there any way I could contact you @fkokosinski or @fzdobylak to learn more about how you completed this. I want to try implementing another filesystem but I am new to this. Let me know if it would be possible to email/text/videocall.

kartben, 2023-08-21:
@fzdobylak looks like this can be closed, right?

fzdobylak, 2023-08-22:
@kartben Yes, the work we have planned is done. The features mentioned in the last two bullets are not implemented, but we don't have any plans to realise them in the near future because current implementation is enough in most use cases.

closed 2023-08-22 by fkokosinski