zephyrproject-rtos/zephyr #51364: ESP32 WIFI: when allocating system_heap to PSRAM(extern ram), wifi station can't connet to ap(indicate that ap not found)

zoujixing, 2022-10-18:
To reduce internal ram usage, I try to allocate system_heap to PSRAM(eg: k_malloc(), k_calloc()). Use folllow seeting:
`
CONFIG_HEAP_MEM_POOL_SIZE=0
CONFIG_ESP_HEAP_MIN_EXTRAM_THRESHOLD
`
This uses k_malloc in the heap_cap.c instead of k_malloc() in mempool.c.
Then wifi as station mode to connect a AP, but it fail and indicate not found ap(AP is existed).Whether the WIFI can not use PSRAM?

zoujixing, 2022-10-18:
The same problem occurs with the following configuration:
`
CONFIG_HEAP_MEM_POOL_SIZE=50000
CONFIG_ESP_HEAP_MIN_EXTRAM_THRESHOLD=1024
`

sylvioalves, 2022-10-18:
Hi @zoujixing, I don't recommend you to do so as this would result in unpredictable behavior. I recommend moving a few network stack and wifi buffers using the flag below:  

```c
CONFIG_ESP_SPIRAM=y
CONFIG_ESP32_WIFI_NET_ALLOC_SPIRAM=y
```
Let me know if this is enough for you RAM usage.

sylvioalves, 2022-10-24:
@zoujixing, Are you able to use 3.2 instead 3.1? If not, I can share a branch with the necessary changes.

zoujixing, 2022-10-26:
Hi @sylvioalves,
I'm trying to use v3.2 instead 3.1 but need time. Please tell me the necessary changes on v3.1.Thanks.

closed 2023-01-09 by github-actions[bot]