zephyrproject-rtos/zephyr #51369: Bluetooth: hci: h4: weak bt_hci_transport_setup down and h4_close function missing

eSTChristopheU, 2022-10-18:
**Describe the bug**

The h4_close() function is not available into zephyr 3.2.0, into the bluetooth drivers (hci, h4.c)

I need to use bt_enale() then bt_disable() then bt_enable() and I need to close properly the HCI external controller.

Do you plan to improve the h4.c file to propose the h4_close() and an equivalent weak function (bt_hci_transport_setup) ?

My setup: 
zephyr 3.2.0
STM32U585 as BT peripheral connected to an Espressif ESP32-C3 mini using an HCI UARTcontroller firmware (controller_hci_uart_esp32c3_and_esp32s3)

Miceuz, 2024-09-19:
I am experiencing the same issue. I'd like to shut down bluetooth while I'm not using it.

jhedberg, 2024-09-23:
@eSTChristopheU @Miceuz there are no plans to add such functionality from my side, but the contribution is welcome if you'd like to work on it. If not, I think it's better to close this enhancement request.

closed 2024-12-16 by jhedberg