zephyrproject-rtos/zephyr #51376: Silabs WFX200 Binary Blob

gcx-dimiz, 2022-10-18:
## Personal Note

Hi,

I would like to start here the discussion if this PR is required or not. Because I'm currently not sure. Because as mentioned bellow the binary blob is part of the driver of the official Silabs repository. So I don't know if it is really necessary to distribute it through `west blob`. And even if it is. I don't know who should maintain this since it is already in an official repository of Silabs. 

## Origin

The binary blob is from https://github.com/SiliconLabs/wfx-fullMAC-driver/ and is part of https://github.com/SiliconLabs/wfx-fullMAC-driver/blob/master/wfx_fmac_driver/firmware/sl_wfx_wf200_C0.h file. This blob is required to run the WFX200 HW. 

## Type

- [ ] Precompiled library
- [x] Firmware image

## Module

The binary blob will be refrenced from driver::wifi::wfx200 driver. At least at its current state

## Purpose

This blob is uploaded during the initialization to the wfx200 hardware to make it operational (https://docs.silabs.com/wifi/wf200/rtos/latest/driver-initialization). 

## Pull Request

https://github.com/zephyrproject-rtos/zephyr/pull/37525 . This PR is currently closed due to inactivity. I will either create a new one or try to reactive it. 

## Dependencies

https://github.com/zephyrproject-rtos/hal_silabs/pull/15

## License

```
/***************************************************************************//**
Copyright (c) 2021, Silicon Laboratories, Inc.

All rights reserved.

Your use of any portion of the Software shall constitute your acceptance of the
following terms and conditions. If you do not agree to the following terms and
conditions, do not use, retain or redistribute any portion of the Software.

Limited Redistribution and Use License. Silicon Laboratories Inc. (“Silicon
Labs”) hereby grants to any person obtaining a copy of this firmware (the
“Software”) a non-exclusive right to use and distribute, in binary form, without
modification, the Software and permit persons to whom the Software is provided
to do the same, subject to the following conditions:

1) Redistributions must reproduce the above copyright notice, this license
   notice and the following disclaimer in the documentation and/or other
   materials provided with the distribution.

2) No reverse  engineering, decompilation, or disassembly of this Software is
   permitted.

3) Neither Silicon Labs nor any of the third party contributors endorse or
   promote products derived from this Software.

4) Nothing shall obligate Silicon Labs to provide you with source code for
   Software licensed and provided to you in object code.

Limited patent license. Silicon Labs grants a worldwide, non-exclusive,
no-charge, royalty-free patent license to make, have made, use, offer to sell,
sell, import, and otherwise transfer this Software, where such license applies
only to those patent claims licensable by Silicon Labs that are necessarily
infringed by this Software. This patent license shall not apply to any
combinations that include this Software.  No hardware is licensed hereunder.

If you institute patent litigation against any entity (including a cross-claim
or counterclaim in a lawsuit) alleging that the Software itself infringes your
patent(s), then your rights granted under this license shall terminate as of the
date such litigation is filed.

DISCLAIMER. THIS SOFTWARE IS PROVIDED BY SILICON LABS "AS IS." ANY EXPRESS OR
IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF
MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT
SHALL  SILICON LABS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL,
EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT
OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS
INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN
CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING
IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY
OF SUCH DAMAGE.IN NO EVENT SHALL SILICON LABS’ TOTAL LIABILITY TO YOU, WHETHER
IN CONTRACT, TORT (INCLUDING NEGLIGENCE), OR OTHERWISE, EXCEED THE PRICE PAID BY
YOU FOR THE SOFTWARE.

-------------------------------------------------------------------------------
The following software may be included in this product: wpa_supplicant

Use of any of this software is governed by the terms of the license below:

    Copyright (c) 2003-2017, Jouni Malinen <j@w1.fi> and contributors
    All Rights Reserved.

This software may be distributed, used, and modified under the terms of

    BSD license:

    Redistribution and use in source and binary forms, with or without
    modification, are permitted provided that the following conditions are
    met:

    1. Redistributions of source code must retain the above copyright
       notice, this list of conditions and the following disclaimer.

    2. Redistributions in binary form must reproduce the above copyright
       notice, this list of conditions and the following disclaimer in the
       documentation and/or other materials provided with the distribution.

    3. Neither the name(s) of the above-listed copyright holder(s) nor the
       names of its contributors may be used to endorse or promote products
       derived from this software without specific prior written permission.

    THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS
    "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
    LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR
    A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT
    OWNER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL,
    SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT
    LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE,
    DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY
    THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
    (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
    OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.

-------------------------------------------------------------------------------
The following software may be included in this product: jsmn

Use of any of this software is governed by the terms of the license below:

    Copyright (c) 2010 Serge A. Zaitsev

    Permission is hereby granted, free of charge, to any person obtaining a copy
    of this software and associated documentation files (the "Software"), to deal
    in the Software without restriction, including without limitation the rights
    to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
    copies of the Software, and to permit persons to whom the Software is
    furnished to do so, subject to the following conditions:

    The above copyright notice and this permission notice shall be included in
    all copies or substantial portions of the Software.

    THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
    IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
    FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
    AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
    LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
    OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
    THE SOFTWARE.
 ******************************************************************************/
```

carlescufi, 2022-11-02:
@gcx-dimiz could you please confirm if you will be opening a PR demonstrating the use and integration of this blob in Zephyr?

henrikbrixandersen, 2022-11-02:
> I would like to start here the discussion if this PR is required or not. Because I'm currently not sure. Because as mentioned bellow the binary blob is part of the driver of the official Silabs [...]

As this is a firmware blob for enabling a SoC feature (WiFi) this does fall under the project [policies for binary blobs](https://docs.zephyrproject.org/latest/contribute/bin_blobs.html) as far as I can tell. As @carlescufi mentioned, an up-to-date example PR for how to use this in Zephyr would be needed before we can approve this.

carlescufi, 2022-11-02:
> As this is a firmware blob for enabling a SoC feature (WiFi) this does fall under the project [policies for binary blobs](https://docs.zephyrproject.org/latest/contribute/bin_blobs.html) as far as I [...]

I'm not sure why this doesn't fit the description in the binary blobs documentation? Here's a quote from it;

> Firmware images: An image containing the executable code for a secondary processor or CPU. This can be full or partial (typically delta or patch data) and is generally copied into RAM or flash memory by the main CPU. An example could be the firmware for the core running a Bluetooth LE Controll

closed 2023-01-17 by github-actions[bot]