zephyrproject-rtos/zephyr #51402: Intergration of QSPI support into SPI bus

firebladed, 2022-10-18:
**Is your enhancement proposal related to a problem? Please describe.**
currently devices that support QSPI seem to require separate "bus" drivers from SPI devices

e.g 
the ft813 supports single, dual and quad SPI configurations but requires setting of a register to switch between to dual or quad spi
and therefore requires a write in single mode to switch to dual/quad mode 

currently this appears it would require both the standard SPI bus implementation and an one or more of the apparent variants of existing QSPI drivers as there appears to be no qspi bus.

**Describe the solution you'd like**

The SPI bus can support Quad SPI, and Dual, bus configurations and can switch modes at request of the device driver.

- device drivers can request switching to Dual or Quad spi modes and get either a successful or a mode not supported response
 
- existing controller drivers that don't implement mode switching get any automatic not supported, from a default api function 
 
- an api function to return a bus controllers supported mode list/bitmap with a default api response if single where not implemented
 
- target modes that require additional pins return "not configured" if appropriate pins are not configured

- spi controller drivers implement the specifics of (de)composing data to/from the separate lines

**Describe alternatives you've considered**

an alternate QSPI bus that be substituted for a standard SPI bus in device drivers, like can be done with I2C

e.g 
```	
COND_CODE_1(DT_INST_ON_BUS(inst, spi),			
(DEVICE_CONFIG_SPI(inst)),			
(DEVICE_CONFIG_I2C(inst)));	
```

**Additional context**

related #51344

firebladed, 2022-10-19:
after reading more extensively there appears to be some support for QSPI on the spi bus in some drivers e.g drivers/spi/spi_xec_qmspi.c

but not others, 

e.g the nordic qspi controller(drivers/flash/nrf_qspi_nor.c) seems seems to implement a flash controller only 

i cant find a driver that implements nordic,nrf-qspi anywhere even though compatible: "nordic,nrf-qspi" is used in several places
e.g. dts/arm/nordic/nrf52840.dtsi

however none ive found so far seem to provide the option to either switch mode e.g single -> quad 
or read/write in a specified mode

tbursztyka, 2024-04-03:
@firebladed please look at #39991 and way more recently #70723 

So far the issue of enabling dual/quad/octal into generic SPI API has always been the lack of hardware other than flash memory devices. And thus manufacturers come with specifically tailored controller (like nordic) only meant to work against flash memory.

I have seen only one specification of a generic SPI controller exposing dual/quad/octal modes, but never got my hand on the actual hardware.