zephyrproject-rtos/zephyr #51406: Blinky not executing on Windows

annieeinna, 2022-10-18:
**Describe the bug**
The sample program: Blinky is not able to create an image on esp32 board.
The following error is displayed in command prompt:
FAILED: CMakeFiles/app.dir/src/main.c.obj
C:\Users\kcana\zephyr-sdk-0.15.1\xtensa-espressif_esp32_zephyr-elf\bin\xtensa-espressif_esp32_zephyr-elf-gcc.exe -DCONFIG_SPI_FLASH_ROM_DRIVER_PATCH -DCONFIG_SPI_FLASH_USE_LEGACY_IMPL -DESP_PLATFORM -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -IC:/Users/kcana/zephyrproject/zephyr/include -IC:/Users/kcana/zephyrproject/zephyr/build/zephyr/include/generated -IC:/Users/kcana/zephyrproject/zephyr/soc/xtensa/esp32 -IC:/Users/kcana/zephyrproject/zephyr/soc/xtensa/esp32/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/include/bt -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/include/crypto -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../esp_shared/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../esp_shared/include/wifi -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/hal/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/hal/esp32/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/hal/platform_port/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_hw_support/include/soc -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_hw_support/port/esp32 -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_hw_support/port/esp32/private_include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_hw_support/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_common/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp32/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/soc/esp32/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/xtensa/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/xtensa/esp32/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/soc/include/soc -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/soc/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/soc/src/esp32/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/driver/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_phy/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_phy/include/esp32 -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_phy/esp32/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_wifi/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/efuse/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/efuse/private_include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/efuse/esp32/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/efuse/esp32/private_include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_system/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_wifi/esp32/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_timer/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_timer/private_include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_netif/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_event/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/log/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/wpa_supplicant/esp_supplicant/src -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/wpa_supplicant/esp_supplicant/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/wpa_supplicant/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/wpa_supplicant/include/esp_supplicant -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/wpa_supplicant/port/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/wpa_supplicant/src -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/wpa_supplicant/src/crypto -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/wpa_supplicant/src/utils -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/wpa_supplicant/src/eap_peer -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/spi_flash/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/spi_flash/include/spi_flash -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/spi_flash/private_include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/bootloader_support/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/bootloader_support/include_bootloader -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/heap/include -IC:/Users/kcana/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/mbedtls/port/include -IC:/Users/kcana/zephyrproject/zephyr/modules/hal_nxp/. -isystem C:/Users/kcana/zephyrproject/zephyr/lib/libc/minimal/include -fno-strict-aliasing -Os -imacros C:/Users/kcana/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always --sysroot=C:/Users/kcana/zephyr-sdk-0.15.1/xtensa-espressif_esp32_zephyr-elf/xtensa-espressif_esp32_zephyr-elf -imacros C:/Users/kcana/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=C:/Users/kcana/zephyrproject/zephyr/samples/basic/blinky=CMAKE_SOURCE_DIR -fmacro-prefix-map=C:/Users/kcana/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=C:/Users/kcana/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -mlongcalls -Wno-unused-variable -Wno-maybe-uninitialized -std=c99 -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles\app.dir\src\main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj -c C:/Users/kcana/zephyrproject/zephyr/samples/basic/blinky/src/main.c
In file included from C:/Users/kcana/zephyrproject/zephyr/include/zephyr/toolchain/gcc.h:88,
                 from C:/Users/kcana/zephyrproject/zephyr/include/zephyr/toolchain.h:50,
                 from C:/Users/kcana/zephyrproject/zephyr/include/zephyr/kernel_includes.h:19,
                 from C:/Users/kcana/zephyrproject/zephyr/include/zephyr/kernel.h:17,
                 from C:/Users/kcana/zephyrproject/zephyr/samples/basic/blinky/src/main.c:7:
C:/Users/kcana/zephyrproject/zephyr/include/zephyr/device.h:83:41: error: '__device_dts_ord_DT_N_ALIAS_led0_P_gpios_IDX_0_PH_ORD' undeclared here (not in a function)
   83 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_, dev_id)
      |                                         ^~~~~~~~~
C:/Users/kcana/zephyrproject/zephyr/include/zephyr/toolchain/common.h:132:26: note: in definition of macro '_DO_CONCAT'
  132 | #define _DO_CONCAT(x, y) x ## y
      |                          ^
[... cut ...]
2. See error: FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\Users\kcana\zephyrproject\zephyr\build'

**Expected behavior**
Expected the LED on ESP32 to blink.

**Impact**
The impact of this issue is not knowing if the zephyr project is installed in my system correctly. The hello world sample works just fine.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain - Zephyr SDK
 - Version used: 0.15.1

**Additional context**
ESP32 blink LED sample program is not working

nordicjm, 2022-10-19:
There is no LED defined in the boards file, therefore you cannot run blinky (if your board has an LED you'd need to define it in the boards file)

nordicjm, 2022-10-19:
@sylvioalves should there be an LED definition for this board? If so, please update and fix, otherwise please close.

annieeinna, 2022-10-19:
ESP32 DevKit V1 comes with on board red LED which is connected to GPIO2.

sylvioalves, 2022-10-26:
ESP32 devkits supported in Zephyr do not have embedded LEDs. I recommend creating an overlay file and add the gpio/led mapping for that. Please, take a look at the blink documentation: https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/basic/blinky/README.rst

closed 2022-10-26 by sylvioalves