zephyrproject-rtos/zephyr #51418: Intel CAVS: Assertion failed in tests/subsys/logging/log_links

shuai1x, 2022-10-19:
**Describe the bug**
On intel_adsp_cavs15/18/25 platform, tests/subsys/logging/log_links failed due to assertion failure.
Bisect result: c0aca6af7aa2dfae63c937e567b0c4ba5a13f571

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b intel_adsp_cavs15 tests/subsys/logging/log_links
2. west flash --remote-host=(ip address to DUT) --config-dir=/home/ztest/cavs/config/ --pty
3. see error

**Logs and console output**
```
START - test_log_domain_count
 PASS - test_log_domain_count in 0.001 seconds
===================================================================
START - test_log_domain_name_get
 PASS - test_log_domain_name_get in 0.001 seconds
===================================================================
START - test_log_runtime_level_set

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/logging/log_links/src/main.c:196: test_single_runtime_level: (*link_level not equal to 2)
1:0 Unexpected compiled level (got:4 exp:2)
 FAIL - test_log_runtime_level_set in 0.001 seconds
===================================================================
START - test_log_source_count
 PASS - test_log_source_count in 0.001 seconds
===================================================================
START - test_log_source_name_get
 PASS - test_log_source_name_get in 0.001 seconds
===================================================================
TESTSUITE log_links failed.
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain - SDK 15.0
 - Commit SHA: c0aca6af7aa2dfae63c937e567b0c4ba5a13f571

smrtos, 2022-10-20:
The multi-domain logging doc: https://docs.zephyrproject.org/latest/services/logging/index.html#multi-domain-support

smrtos, 2022-10-21:
The `test_log_runtime_level_set()` has 4 tests. Tried them one by one.

```
	test_single_runtime_level(1, 0, &domains_a[0]->sources[0].rlevel);
	test_single_runtime_level(1, 3, &domains_a[0]->sources[3].rlevel);
	test_single_runtime_level(2, 1, &domains_a[1]->sources[1].rlevel); // only this one can pass
	test_single_runtime_level(3, 1, &domains_b[0]->sources[1].rlevel);
```

smrtos, 2022-10-23:
On other platforms, there are only 2 mocked log backends for this test:
https://github.com/zephyrproject-rtos/zephyr/blob/78809549ee4cf7de3150ff4c3887011825584547/tests/subsys/logging/log_links/src/main.c#L41-L43

But on ADSP platform, there's an extra backend named `log_backend_adsp`, which messed up the `backends_link_init()` for this test.

https://github.com/zephyrproject-rtos/zephyr/blob/78809549ee4cf7de3150ff4c3887011825584547/subsys/logging/backends/CMakeLists.txt#L3-L6

https://github.com/zephyrproject-rtos/zephyr/blob/78809549ee4cf7de3150ff4c3887011825584547/subsys/logging/backends/log_backend_adsp.c#L98

If remove this backend with below config. The test can pass.

Trying to fix it with: https://github.com/zephyrproject-rtos/zephyr/pull/51540

smrtos, 2022-10-23:
BTW, the doc may need to be refined. The content is not quit following the code implementation. It seems there's no `link_offset` but `domain_offset`. 

And it would be nice if there are more detailed explanation about the multi-domain working logic.  @nordic-krch

nordic-krch, 2022-10-24:
@smrtos 
> And it would be nice if there are more detailed explanation/illustration about how the multi-domain logging works.

I will try to provide some update but may take few days.

closed 2022-10-25 by carlescufi