zephyrproject-rtos/zephyr #51420: tests: subsys: logging: log_links: logging.log_links fails

nordic-piks, 2022-10-19:
**Describe the bug**
The tests/subsys/logging/log_links/logging.log_links fails

Observed for
- nrf5340dk_nrf5340_cpunet

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf5340dk_nrf5340 connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/subsys/logging/log_links  -p nrf5340dk_nrf5340_cpunet --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
[2022-10-18T23:56:25.575Z] ERROR   - *** Booting Zephyr OS build zephyr-v3.2.0-679-g5ab62a43777d  ***
[2022-10-18T23:56:25.575Z] Running TESTSUITE log_links
[2022-10-18T23:56:25.575Z] ===================================================================
[2022-10-18T23:56:25.575Z] START - test_log_compiled_level_get
[2022-10-18T23:56:25.575Z] PASS - test_log_compiled_level_get in 0.001 seconds
[2022-10-18T23:56:25.575Z] ===================================================================
[2022-10-18T23:56:25.575Z] START - test_log_domain_count
[2022-10-18T23:56:25.575Z] PASS - test_log_domain_count in 0.001 seconds
[2022-10-18T23:56:25.575Z] ===================================================================
[2022-10-18T23:56:25.575Z] START - test_log_domain_name_get
[2022-10-18T23:56:25.575Z] Assertion failed at ../../../../../../../../../zephyr/zephyr/tests/subsys/logging/log_links/src/main.c:214: log_links_test_log_domain_name_get: (strcmp(log_domain_name_get(0), "") not equal to 0)
[2022-10-18T23:56:25.575Z] Unexpected domain name
[2022-10-18T23:56:25.575Z] FAIL - test_log_domain_name_get in 0.048 seconds
[2022-10-18T23:56:25.575Z] ===================================================================
[2022-10-18T23:56:25.575Z] START - test_log_runtime_level_set
[2022-10-18T23:56:25.575Z] PASS - test_log_runtime_level_set in 0.002 seconds
[2022-10-18T23:56:25.575Z] ===================================================================
[2022-10-18T23:56:25.575Z] START - test_log_source_count
[2022-10-18T23:56:25.575Z] PASS - test_log_source_count in 0.002 seconds
[2022-10-18T23:56:25.575Z] ===================================================================
[2022-10-18T23:56:25.575Z] START - test_log_source_name_get
[2022-10-18T23:56:25.575Z] PASS - test_log_source_name_get in 0.001 seconds
[2022-10-18T23:56:25.575Z] ===================================================================
[2022-10-18T23:56:25.575Z] TESTSUITE log_links failed.
[2022-10-18T23:56:25.575Z] ------ TESTSUITE SUMMARY START ------
[2022-10-18T23:56:25.575Z] SUITE FAIL -  83.33% [log_links]: pass = 5, fail = 1, skip = 0, total = 6 duration = 0.055 seconds
[2022-10-18T23:56:25.575Z] - PASS - [log_links.test_log_compiled_level_get] duration = 0.001 seconds
[2022-10-18T23:56:25.575Z] - PASS - [log_links.test_log_domain_count] duration = 0.001 seconds
[2022-10-18T23:56:25.575Z] - FAIL - [log_links.test_log_domain_name_get] duration = 0.048 seconds
[2022-10-18T23:56:25.575Z] - PASS - [log_links.test_log_runtime_level_set] duration = 0.002 seconds
[2022-10-18T23:56:25.575Z] - PASS - [log_links.test_log_source_count] duration = 0.002 seconds
[2022-10-18T23:56:25.575Z] - PASS - [log_links.test_log_source_name_get] duration = 0.001 seconds
[2022-10-18T23:56:25.575Z] ------ TESTSUITE SUMMARY END ------
[2022-10-18T23:56:25.575Z] ===================================================================
[2022-10-18T23:56:25.575Z] RunID: e736a4aa80599aa4dc986a08ebb4f5ca
[2022-10-18T23:56:25.575Z] PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.15
 - Commit SHA or Version used: zephyr-v3.2.0-679-g5ab62a43777d

laurenmurphyx64, 2022-10-25:
(Note: Though same name / issue, not a duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/51418 - that issue was for a different testcase and had a fix specific to Intel ADSP platforms)

nordic-piks, 2023-02-27:
@nordic-krch Any update about this issue?

PR #56509 (2023-04-04): tests: logging: log_links: Set CONFIG_LOG_DOMAIN_NAME in the test
Test was assuming that domain name is empty but on some SoC CONFIG_LOG_DOMAIN_NAME is set by default and test was failing. Add setting of domain name to test
configuration to ensure that same name is used for all platforms.

Fixes #51420.
closed 2023-04-04 by carlescufi