zephyrproject-rtos/zephyr #51430: STM32WB: Implement support for extended advertising

erwango, 2022-10-19:
**Is your enhancement proposal related to a problem? Please describe.**
Extended advertising support on STM32WB requires the use of a specific BLE controler f/w on M0 side.
It also requires some specific stack initialization (with impacts on stm32wb hci driver).

**Describe the solution you'd like**
Update stm32wb ble support to enable seamless use of extended advertising.

erwango, 2022-10-19:
@piotrParf ^^

piotrParf, 2023-04-21:
Is there anything ongoing? We are still using our patches but I saw that you have updated the hal_stm32 to a newer version. Was anything done there in this case?

erwango, 2023-04-21:
@piotrParf Sorry, but nothing done on this front yet. I'll try to plan this for next release

ericmechin, 2025-04-22:
In addition of flashing the STM32WB Copro Wireless Binary stm32wb5x_BLE_HCILayer_extended_fw, the BLE stack have to be configured for extended advertising.
This configuration is done in the zephyr\drivers\bluetooth\hci\ipm_stm32wb.c file by the stm32wb_start_ble() function.
The stm32wb_start_ble() function call the SHCI_C2_BLE_Init() function with the ble_init_cmd_packet structure parameter.
For extended advertising this structure should set the following parameters:
- Maximum number of advertising sets from 1 to 8
- Maximum advertising data length from 31 to 1650 bytes
- Options flags: bit 3 for extended advertising support
                          bit 4 for CS Algo 2 support

These parameters are set with constants defined in the zephyrproject\modules\hal\stm32\lib\stm32wb\hci\app_conf.h files:
- CFG_BLE_MAX_ADV_SET_NBR
- CFG_BLE_MAX_ADV_DATA_LEN
- CFG_BLE_OPTIONS

Maximum number of advertising sets parameter is linked to Maximum advertising data length parameter such as both compliant with allocated Total memory computed with BLE_EXT_ADV_BUFFER_SIZE based on Max Extended advertising configuration supported.
These parameters are considered by the STM32WB Copro Wireless when Options flag: bit 3 for extended advertising support is set.
These  parameters have limitations due to memory constraints, below here the payload size max by adv set count.
Payload bytes by adv set count : 

- 1 =>  1650  bytes
- 2 =>  1650 bytes
- 3 =>  1650 bytes
- 4 =>  1035 bytes  
- 5 =>   621 bytes
- 6 =>   414 bytes
- 7 =>   207 bytes
- 8 =>   207 bytes

So for a Maximum number of advertising sets to 8 the Maximum advertising data length is 207 bytes, and for a Maximum advertising data length of 1650 bytes the Maximum number of advertising sets is 3.