zephyrproject-rtos/zephyr #51436: tests-ci : drivers: drivers.watchdog: nxp-imxrt11xx series : watchdog build failure

hakehuang, 2022-10-19:
**Describe the bug**
drivers.watchdog build failure due to hal code error

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1060_evk
3. make
4. See error

```
/home/shared/disk/zephyr_project/zephyr_rt1060/zephyr/drivers/watchdog/wdt_mcux_imx_wdog.c: In function 'mcux_wdog_install_timeout':
/home/shared/disk/zephyr_project/zephyr_rt1060/zephyr/drivers/watchdog/wdt_mcux_imx_wdog.c:79:32: error: 'MSEC_PER_SEC' undeclared (first use in this function)
   79 |         if (cfg-&gt;window.max &lt; (MSEC_PER_SEC / 2)) {
      |                                ^~~~~~~~~~~~
/home/shared/disk/zephyr_project/zephyr_rt1060/zephyr/drivers/watchdog/wdt_mcux_imx_wdog.c:79:32: note: each undeclared identifier is reported only once for each function it appears in
```

**Expected behavior**
PASS

**Impact**
all watchdogs for imxrt11xx series

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

dleach02, 2022-11-02:
@hakehuang, does this also fail for the rt1060? The description references rt1060_evk but the title indicates rt11xxx?

hakehuang, 2022-11-02:
> @hakehuang, does this also fail for the rt1060? The description references rt1060_evk but the title indicates rt11xxx?

yes. also failures on rt1060

closed 2022-11-02 by hakehuang