zephyrproject-rtos/zephyr #51449: device: device_get_binding is broken for nodes with the same name

gmarull, 2022-10-19:
**Describe the bug**

With the removal of labels, DT devices get its `name` set to `DT_NODE_FULL_NAME(node_id)`. The problem is the such name is not necessarily unique, e.g. this is valid in DT:

```dts
i2c@xy {
	...
	sensora: sensor@10 {  /* DT_NODE_FULL_NAME = sensor@10 */
		...
	};
}

i2c@yx {
	...
	sensorb: sensor@10 { /* DT_NODE_FULL_NAME = sensor@10 */
		...
	};
};
```

As a result, only one of them can be obtained at runtime using `device_get_binding()`. While not a frequent case, it is a limitation that should either be fixed or documented.

**To Reproduce**

Apply this patch:
```diff
diff --git a/tests/lib/devicetree/devices/app.overlay b/tests/lib/devicetree/devices/app.overlay
index 3a6212721c..20afde23b6 100644
--- a/tests/lib/devicetree/devices/app.overlay
+++ b/tests/lib/devicetree/devices/app.overlay
@@ -82,6 +82,22 @@
 			};
 		};
 
+		test_i2c2: i2c@11112223 {
+			#address-cells = <1>;
+			#size-cells = <0>;
+			compatible = "vnd,i2c";
+			status = "okay";
+			reg = <0x11112223 0x1000>;
+			clock-frequency = <100000>;
+			test_gpioy: test-i2c-dev@11 {
+				gpio-controller;
+				#gpio-cells = <2>;
+				compatible = "vnd,gpio-expander";
+				status = "okay";
+				reg = <0x11>;
+			};
+		};
+
 		test_gpio_injected: gpio@1000 {
 			gpio-controller;
 			#gpio-cells = <0x2>;
diff --git a/tests/lib/devicetree/devices/src/main.c b/tests/lib/devicetree/devices/src/main.c
index a870253888..1affad6ec4 100644
--- a/tests/lib/devicetree/devices/src/main.c
+++ b/tests/lib/devicetree/devices/src/main.c
@@ -10,9 +10,11 @@
 #include <zephyr/drivers/gpio.h>
 
 #define TEST_GPIO DT_NODELABEL(test_gpio_0)
+#define TEST_I2C2 DT_NODELABEL(test_i2c2)
 #define TEST_I2C DT_NODELABEL(test_i2c)
 #define TEST_DEVA DT_NODELABEL(test_dev_a)
 #define TEST_GPIOX DT_NODELABEL(test_gpiox)
+#define TEST_GPIOY DT_NODELABEL(test_gpioy)
 #define TEST_DEVB DT_NODELABEL(test_dev_b)
 #define TEST_DEVC DT_NODELABEL(test_dev_c)
 #define TEST_PARTITION DT_NODELABEL(test_p0)
@@ -22,7 +24,7 @@
 static const struct device *devlist;
 static const struct device *devlist_end;
 
-static device_handle_t init_order[10];
+static device_handle_t init_order[12];
 
 static int dev_init(const struct device *dev)
 {
@@ -36,6 +38,8 @@ static int dev_init(const struct device *dev)
 
 DEVICE_DT_DEFINE(TEST_GPIO, dev_init, NULL,
 		 NULL, NULL, PRE_KERNEL_1, 90, NULL);
+DEVICE_DT_DEFINE(TEST_I2C2, dev_init, NULL,
+		 NULL, NULL, POST_KERNEL, 10, NULL);
 DEVICE_DT_DEFINE(TEST_I2C, dev_init, NULL,
 		 NULL, NULL, POST_KERNEL, 10, NULL);
 DEVICE_DT_DEFINE(TEST_DEVA, dev_init, NULL,
@@ -45,6 +49,8 @@ DEVICE_DT_DEFINE(TEST_DEVB, dev_init, NULL,
 		 NULL, NULL, POST_KERNEL, 30, NULL);
 DEVICE_DT_DEFINE(TEST_GPIOX, dev_init, NULL,
 		 NULL, NULL, POST_KERNEL, 40, NULL);
+DEVICE_DT_DEFINE(TEST_GPIOY, dev_init, NULL,
+		 NULL, NULL, POST_KERNEL, 40, NULL);
 DEVICE_DT_DEFINE(TEST_DEVC, dev_init, NULL,
 		 NULL, NULL, POST_KERNEL, 50, NULL);
 DEVICE_DT_DEFINE(TEST_PARTITION, dev_init, NULL,
@@ -73,8 +79,10 @@ ZTEST(devicetree_devices, test_init_get)
 		      DEVICE_DT_GET(TEST_DEVA), NULL);
 	zassert_equal(DEVICE_INIT_DT_GET(TEST_DEVB)->dev,
 		      DEVICE_DT_GET(TEST_DEVB), NULL);
-	zassert_equal(DEVICE_INIT_DT_GET(TEST_GPIOX)->dev,
+	zassert_equal(device_get_binding(DEVICE_DT_NAME(TEST_GPIOX)),
 		      DEVICE_DT_GET(TEST_GPIOX), NULL);
+	zassert_equal(device_get_binding(DEVICE_DT_NAME(TEST_GPIOY)),
+		      DEVICE_DT_GET(TEST_GPIOY), NULL);
 	zassert_equal(DEVICE_INIT_DT_GET(TEST_DEVC)->dev,
 		      DEVICE_DT_GET(TEST_DEVC), NULL);
 	zassert_equal(DEVICE_INIT_DT_GET(TEST_PARTITION)->dev,
```

and run `west build -b native_posix tests/lib/devicetree/devices -p`.

**Expected behavior**

I should be able to obtain all devices at runtime using `device_get_binding()`. If not, document the limitations.

**Impact**

N/A

**Logs and console output**

N/A

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.15.0
 - Commit SHA or Version used f4d372bf7d41fc5ee4337733fc0ef562ece904f7

**Additional context**
 N/A

galak, 2022-10-20:
I think it should be something we document, since it was true with `label` that you could have easily had the same value of `label` before and have the exact same issue.

gmarull, 2022-10-20:
> I think it should be something we document, since it was true with `label` that you could have easily had the same value of `label` before and have the exact same issue.

But with labels you could fix it, now you can't. I guess we could also error/warning if 2 nodes are found with the same full name.

henrikbrixandersen, 2022-10-23:
Another option could be to use DT_PATH as the unique device identifier.

gmarull, 2022-10-24:
> Another option could be to use DT_PATH as the unique device identifier.

This is an option, but:

- Device name is limited to 48 characters, we could exceed this limit more easily with paths
- Shell usage would be degraded, unless we redesign how shells interact with devices (e.g., interactive device choice, no need to type).

galak, 2022-10-28:
So here's the warnings we get  -- some 2048 lines of warnings:

```
96b_aerocore2:WARNING: Duplicate node name 'counter' between /soc/timers@40000000/counter and /soc/timers@40000400/counter
96b_aerocore2:WARNING: Duplicate node name 'pwm' between /soc/timers@40000000/pwm and /soc/timers@40000400/pwm
96b_aerocore2:WARNING: Duplicate node name 'qdec' between /soc/timers@40000000/qdec and /soc/timers@40000400/qdec
96b_aerocore2:WARNING: Duplicate node name 'counter' between /soc/timers@40000400/counter and /soc/timers@40000800/counter
96b_aerocore2:WARNING: Duplicate node name 'pwm' between /soc/timers@40000400/pwm and /soc/timers@40000800/pwm
96b_aerocore2:WARNING: Duplicate node name 'qdec' between /soc/timers@40000400/qdec and /soc/timers@40000800/qdec
96b_aerocore2:WARNING: Duplicate node name 'counter' between /soc/timers@40000800/counter and /soc/timers@40000c00/counter
96b_aerocore2:WARNING: Duplicate node name 'pwm' between /soc/timers@40000800/pwm and /soc/timers@40000c00/pwm
96b_aerocore2:WARNING: Duplicate node name 'qdec' between /soc/timers@40000800/qdec and /soc/timers@40000c00/qdec
96b_aerocore2:WARNING: Duplicate node name 'counter' between /soc/timers@40000c00/counter and /soc/timers@40001000/counter
96b_aerocore2:WARNING: Duplicate node name 'counter' between /soc/timers@40001000/counter and /soc/timers@40001400/counter
96b_aerocore2:WARNING: Duplicate node name 'counter' between /soc/timers@40001400/counter and /soc/timers@40001800/counter
96b_aerocore2:WARNING: Duplicate node name 'pwm' between /soc/timers@40000c00/pwm and /soc/timers@40001800/pwm
96b_aerocore2:WARNING: Duplicate node name 'counter' between /soc/timers@40001800/counter and /soc/timers@40001c00/counter
96b_aerocore2:WARNING: Duplicate node name 'pwm' between /soc/timers@40001800/pwm and /soc/timers@40001c00/pwm
96b_aerocore2:WARNING: Duplicate node name 'counter' between /soc/timers@40001c00/counter and /soc/timers@40002000/counter
96b_aerocore2:WARNING: Duplicate node name 'pwm' between /soc/timers@40001c00/pwm and /soc/timers@40002000/pwm
96b_aerocore2:WARNING: Duplicate node name 'pwm' between /soc/timers@40002000/pwm and /soc/timers@40010000/pwm
96b_aerocore2:WARNING: Duplicate node name 'qdec' between /soc/timers@40000c00/qdec and /soc/timers@40010000/qdec
96b_aerocore2:WARNING: Duplicate node name 'pwm' between /soc/timers@40010000/pwm and /soc/timers@40010400/pwm
96b_aerocore2:WARNING: Duplicate node name 'qdec' between /soc/timers@40010000/qdec and /soc/timers@40010400/qdec
96b_aerocore2:WARNING: Duplicate node name 'counter' between /soc/timers@40002000/counter and /soc/timers@40014000/counter
96b_aerocore2:WARNING: Duplicate node name 'pwm' between /soc/timers@40010400/pwm and /soc/timers@40014000/pwm
96b_aerocore2:WARNING: Duplicate node name 'counter' between /soc/timers@40014000/counter and /soc/timers@40014400/counter
96b_aerocore2:WARNING: Duplicate node name 'pwm' between /soc/timers@40014000/pwm and /soc/timers@40014400/pwm
96b_aerocore2:WARNING: Duplicate node name 'counter' between /soc/timers@40014400/counter and /soc/timers@40014800/counter
96b_aerocore2:WARNING: Duplicate node name 'pwm' between /soc/timers@40014400/pwm and /soc/timers@40014800/pwm
96b_argonkey:WARNING: Duplicate node name 'counter' between /soc/timers@40000000/counter and /soc/timers@40000400/counter
96b_argonkey:WARNING: Duplicate node name 'pwm' between /soc/timers@40000000/pwm and /soc/timers@40000400/pwm
96b_argonkey:WARNING: Duplicate node name 'qdec' between /soc/timers@40000000/qdec and /soc/timers@40000400/qdec
96b_argonkey:WARNING: Duplicate node name 'counter' between /soc/timers@40000400/counter and /soc/timers@40000800/counter
96b_argonkey:WARNING: Duplicate node name 'pwm' between /soc/timers@40000400/pwm and /soc/timers@40000800/pwm
96b_argonkey:WARNING: Duplicate node name 'qdec' between /soc/timers@40000400/qdec and /soc/timers@40000800/qdec
96b_argonkey:WARNING: Duplicate node name 'counter' between /soc/timers@40000800/counter and /soc/timers@40000c00/counter
96b_argonkey:WARNING: Duplicate node name 'pwm' between /soc/timers@40000800/pwm and /soc/timers@40000c00/pwm
96b_argonkey:WARNING: Duplicate node name 'qdec' between /soc/timers@40000800/qdec and /soc/timers@40000c00/qdec
96b_argonkey:WARNING: Duplicate node name 'counter' between /soc/timers@40000c00/counter and /soc/timers@40001000/counter
96b_argonkey:WARNING: Duplicate node name 'counter' between /soc/timers@40001000/counter and /soc/timers@40001400/counter
96b_argonkey:WARNING: Duplicate node name 'counter' between /soc/timers@40001400/counter and /soc/timers@40001800/counter
96b_argonkey:WARNING: Duplicate node name 'pwm' between /soc/timers@40000c00/pwm and /soc/timers@40001800/pwm
96b_argonkey:WARNING: Duplicate node name 'counter' between /soc/timers@40001800/counter and /soc/timers@40001c00/counter
96b_argonkey:WARNING: Duplicate node name 'pwm' between /soc/timers@40001800/pwm and /soc/timers@40001c00/pwm
96b_argonkey:WARNING: Duplicate node name 'counter' between /soc/timers@40001c00/counter and /soc/timers@40002000/counter
96b_argonkey:WARNING: Duplicate node name 'pwm' between /soc/timers@40001c00/pwm and /soc/timers@40002000/pwm
96b_argonkey:WARNING: Duplicate node name 'pwm' between /soc/timers@40002000/pwm and /soc/timers@40010000/pwm
96b_argonkey:WARNING: Duplicate node name 'qdec' between /soc/timers@40000c00/qdec and /soc/timers@40010000/qdec
[... cut ...]
stm32f411e_disco@B:WARNING: Duplicate node name 'pwm' between /soc/timers@40000c00/pwm and /soc/timers@40010000/pwm
stm32f411e_disco@B:WARNING: Duplicate node name 'qdec' between /soc/timers@40000c00/qdec and /soc/timers@40010000/qdec
stm32f411e_disco@B:WARNING: Duplicate node name 'counter' between /soc/timers@40000c00/counter and /soc/timers@40014000/counter
stm32f411e_disco@B:WARNING: Duplicate node name 'pwm' between /soc/timers@40010000/pwm and /soc/timers@40014000/pwm
stm32f411e_disco@B:WARNING: Duplicate node name 'counter' between /soc/timers@40014000/counter and /soc/timers@40014400/counter
stm32f

galak, 2022-10-28:
This is filtering out nodes that don't have a compatible.

henrikbrixandersen, 2022-10-31:
> So here's the warnings we get -- some 2048 lines of warnings:

Ouch.

> This is filtering out nodes that don't have a compatible.

Which can be just as affected by this, I guess, as it will cause `DT_NODELABEL()` to fail or at least behave unpredictable for these nodes?

galak, 2022-11-01:
So I was looking at doing the check in `scripts/dts/gen_defines.py`.  However we could do it in `scripts/build/gen_handles.py` and thus it will only look at actual devices that exist in someones build.

galak, 2022-11-01:
> > This is filtering out nodes that don't have a compatible.
> 
> Which can be just as affected by this, I guess, as it will cause `DT_NODELABEL()` to fail or at least behave unpredictable for these nodes?

Need to double check my filter, but that's a good point.  I think it was filtering on nodes that had a binding match.

galak, 2022-11-07:
@gmarull / @henrikbrixandersen thoughts on doing this check in scripts/build/gen_handles.py.  Means it happens when you do make/ninja instead of at cmake time.  But also means we report a lot less false positives since it will be over actual `struct device`s instead of checking the devicetree.

henrikbrixandersen, 2022-11-27:
> But also means we report a lot less false positives since it will be over actual `struct device`s instead of checking the devicetree.

But the DT_* APIs are not limited to devicetree nodes which have a corresponding `struct device`. These APIs work for any devicetree node, regardless if it has a compatible property, a corresponding driver, a corresponding driver instance etc.

I think the check needs to do the same, otherwise it will work differently depending on e.g. Kconfig settings.

galak, 2022-11-28:
> But the DT_* APIs are not limited to devicetree nodes which have a corresponding `struct device`. These APIs work for any devicetree node, regardless if it has a compatible property, a corresponding [...]

Not sure I follow since the check/issue is around `device_get_binding` which would only related to `struct device`.

galak, 2023-01-17:
@gmarull thoughts on how to proceed on this?

gmarull, 2023-02-23:
> @gmarull thoughts on how to proceed on this?

Not really. The thing is that we lost a way to reliably obtain a device reference at runtime. Before you could always set `label`, which was guaranteed to be unique. We probably need to rethink devices runtime facilities. `label` was used as a software configuration property, but now we have a similar pattern with node labels, sometimes used to refer to a node within DT but in many other cases to just take a device reference in the code. Maybe we could have an option that, when enabled, allows to have a device table that has _human-friendly_ metadata to access the device at runtime, e.g. all sensors create a `/dev/sensorN` + `compatible` entry. This could be useful when debugging, or in shell-based apps.

mmahadevan108, 2024-09-17:
@gmarull, does #71912 fix this issue?

closed 2024-12-01 by github-actions[bot]