zephyrproject-rtos/zephyr #51458: Only one instance of mcp2515

babos, 2022-10-19:
I am unable to get two mcp2515 instances using DEVICE_DT_GET because the can_cmp2515 driver creates a device only for the first found instance of "microchip, mcp2515".
The adxl362 driver, on the other hand, will iterate through all adxl362 instances, using **DT_INST_FOREACH_STATUS_OKAY**

why mcp2515 not ? would it be possible to remedy?
I'm not sure why the can_mcp2515 driver won't iterate through all the instances, if it is intentional or just a mistake. 

thank'you

Babos

nordicjm, 2022-10-20:
>  I'm not sure why the can_mcp2515 driver won't iterate through all the instances, if it is intentional or just a mistake.

Drivers have been updated over time but this one has not, it should be possible to replace the macros and add support for multiple devices.

henrikbrixandersen, 2022-10-22:
I'll add multi-instance support. This was just an oversight.

henrikbrixandersen, 2022-10-23:
@babos You can test the multi-instance support in #51539.

closed 2022-10-27 by carlescufi

babos, 2022-12-22:
awesome, thx

PR #51539 (2022-10-27): drivers: can: mcp2515: add support for multiple device instances
Add support for multiple MCP2515 CAN controller device instances.

Fixes: #51458
commit 58afbc8f (2022-10-27): drivers: can: mcp2515: add support for multiple device instances

Add support for multiple MCP2515 CAN controller device instances.

Fixes: #51458
PR #51696 (2022-11-04): [Backport v3.2-branch] drivers: can: mcp2515: add support for multiple device instances
Backport 58afbc8f26ce902844b2bcbbab39fc5f9573a17a from #51539

Fixes: #51458
commit 638735fc (2022-11-04): drivers: can: mcp2515: add support for multiple device instances

Add support for multiple MCP2515 CAN controller device instances.

Fixes: #51458

(cherry picked from commit 58afbc8f26ce902844b2bcbbab39fc5f9573a17a)