zephyrproject-rtos/zephyr #51464: samples: drivers: peci: Code doesn't build for npcx7m6fb_evb board

albertofloyd, 2022-10-19:
**Describe the bug**
Build fails during PECI API update PR https://github.com/zephyrproject-rtos/zephyr/pull/51415

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/drivers/peci
2. rm -rf build && west build -b npcx7m6fb_evb
3. See error

/home/jamezaar/ecwork/zephyr_albertofloyd/drivers/peci/peci_npcx.c:32:22: error: field 'trans_sync_sem' has incomplete type
   32 |         struct k_sem trans_sync_sem;
      |                      ^~~~~~~~~~~~~~
/home/jamezaar/ecwork/zephyr_albertofloyd/drivers/peci/peci_npcx.c:33:22: error: field 'lock' has incomplete type
   33 |         struct k_sem lock;
      |                      ^~~~
/home/jamezaar/ecwork/zephyr_albertofloyd/drivers/peci/peci_npcx.c: In function 'peci_npcx_wait_completion':
/home/jamezaar/ecwork/zephyr_albertofloyd/drivers/peci/peci_npcx.c:58:15: warning: implicit declaration of function 'k_sem_take' [-Wimplicit-function-declaration]
   58 |         ret = k_sem_take(&data->trans_sync_sem, PECI_TIMEOUT);
      |               ^~~~~~~~~~
/home/jamezaar/ecwork/zephyr_albertofloyd/drivers/peci/peci_npcx.c:20:34: warning: implicit declaration of function 'K_MSEC' [-Wimplicit-function-declaration]
   20 | #define PECI_TIMEOUT             K_MSEC(300)
      |                                  ^~~~~~
/home/jamezaar/ecwork/zephyr_albertofloyd/drivers/peci/peci_npcx.c:58:49: note: in expansion of macro 'PECI_TIMEOUT'
   58 |         ret = k_sem_take(&data->trans_sync_sem, PECI_TIMEOUT);
      |                                                 ^~~~~~~~~~~~
/home/jamezaar/ecwork/zephyr_albertofloyd/drivers/peci/peci_npcx.c: In function 'peci_npcx_configure':
/home/jamezaar/ecwork/zephyr_albertofloyd/drivers/peci/peci_npcx.c:76:33: error: 'K_FOREVER' undeclared (first use in this function)
   76 |         k_sem_take(&data->lock, K_FOREVER);
      |                                 ^~~~~~~~~
/home/jamezaar/ecwork/zephyr_albertofloyd/drivers/peci/peci_npcx.c:76:33: note: each undeclared identifier is reported only once for each function it appears in
/home/jamezaar/ecwork/zephyr_albertofloyd/drivers/peci/peci_npcx.c:97:9: warning: implicit declaration of function 'k_sem_give' [-Wimplicit-function-declaration]
   97 |         k_sem_give(&data->lock);
      |         ^~~~~~~~~~
/home/jamezaar/ecwork/zephyr_albertofloyd/drivers/peci/peci_npcx.c: In function 'peci_npcx_disable':
/home/jamezaar/ecwork/zephyr_albertofloyd/drivers/peci/peci_npcx.c:106:33: error: 'K_FOREVER' undeclared (first use in this function)
  106 |         k_sem_take(&data->lock, K_FOREVER);
      |                                 ^~~~~~~~~
/home/jamezaar/ecwork/zephyr_albertofloyd/drivers/peci/peci_npcx.c: In function 'peci_npcx_enable':
/home/jamezaar/ecwork/zephyr_albertofloyd/drivers/peci/peci_npcx.c:121:33: error: 'K_FOREVER' undeclared (first use in this function)
  121 |         k_sem_take(&data->lock, K_FOREVER);
      |                                 ^~~~~~~~~
/home/jamezaar/ecwork/zephyr_albertofloyd/drivers/peci/peci_npcx.c: In function 'peci_npcx_transfer':
/home/jamezaar/ecwork/zephyr_albertofloyd/drivers/peci/peci_npcx.c:143:33: error: 'K_FOREVER' undeclared (first use in this function)
  143 |         k_sem_take(&data->lock, K_FOREVER);
      |                                 ^~~~~~~~~
/home/jamezaar/ecwork/zephyr_albertofloyd/drivers/peci/peci_npcx.c: In function 'peci_npcx_init':
/home/jamezaar/ecwork/zephyr_albertofloyd/drivers/peci/peci_npcx.c:266:9: warning: implicit declaration of function 'k_sem_init' [-Wimplicit-function-declaration]
  266 |         k_sem_init(&data->trans_sync_sem, 0, 1);
      |         ^~~~~~~~~~

**Expected behavior**
Sample code is build for the board if unsupported.
Or
Sample code builds correctly

albertofloyd, 2022-10-19:
FYI @rgundi

albertofloyd, 2022-10-20:
FYI @ChiHuaL

ChiHuaL, 2022-10-21:
Hi,
The PR#51500 fixes the compiler error. Please take a look at it.

PR #51500 (2022-10-22): Driver: PECI: npcx: fix missing header inclusion
Add the missing zephyr/kernel.h header inclusion to fix the compiler error.
This is required due to the removal of kernel.h from init.h
    
fixes: #51464
closed 2022-10-22 by stephanosio