zephyrproject-rtos/zephyr #51468: mps3_an547:tests/lib/cmsis_dsp/filtering bus fault

manuargue, 2022-10-20:
**Describe the bug**
`tests/lib/cmsis_dsp/filtering/libraries.cmsis_dsp.filtering.misc.fpu` fails with bus fault for `mps3_an547` in test case `test_arm_correlate_f32_10_11`. Caught in an unrelated PR: https://github.com/zephyrproject-rtos/zephyr/pull/51451/checks?check_run_id=8988244380

**To Reproduce**
Build and run `tests/lib/cmsis_dsp/filtering/libraries.cmsis_dsp.filtering.misc.fpu` on `mps3_an547` board.

**Expected behavior**
Test pass

**Impact**
Blocking PR https://github.com/zephyrproject-rtos/zephyr/pull/51451

**Logs and console output**
```
START - test_arm_correlate_f32_10_11
E: ***** BUS FAULT *****
E:   Precise data bus error
E:   BFAR Address: 0x3f0f52cf
E: r0/a1:  0x30003908  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0x30000298 r14/lr:  0xbdb167ce
E:  xpsr:  0x21100000
E: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x0000002f
E: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0xffffffff  s[ 7]:  0x00000000
E: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000
E: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0x00000000
E: fpscr:  0x30000000
E: Faulting instruction address (r15/pc): 0x10007ad6
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x300007a0 (test_arm_correlate_f32_10_11)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.15.1
 - Commit SHA: 2af5779101f6f8201717b4ae6ccb38b89d47e3f5, branched from main 5906bad7c317a684a6d9a16fd8c73d275b509357

**Additional context**
N/A

manuargue, 2022-10-20:
@stephanosio

lpawelcz, 2022-10-20:
Hi, the same thing happens in other unrelated PR: https://github.com/zephyrproject-rtos/zephyr/pull/50352
CI: https://github.com/zephyrproject-rtos/zephyr/pull/50352/checks?check_run_id=9001159502

stephanosio, 2022-10-20:
A regression due to https://github.com/zephyrproject-rtos/zephyr/pull/51340.

It looks like this test fails with the newlib full variant (it was previously using the nano variant due to the defaults) for some reason.  I will investigate.

stephanosio, 2022-10-20:
This happens because the MVE `arm_correlate_f32` implementation has a bug in which it writes to a negative index of the output buffer, which leads to a corrupted memory block header and hence a fault during `free()` of the memory block.

Upstream CMSIS-DSP bug filed: https://github.com/ARM-software/CMSIS-DSP/issues/59

PR #51478 (2022-10-20): tests: cmsis_dsp: MVE correlate out-of-bounds access workaround
The MVE `arm_correlate_f32` and `arm_correlate_q31` implementations may write to negative indices of the output buffer (refer to the upstream CMSIS-DSP bug ARM-software/CMSIS-DSP#59).

This commit adds a workaround for the above bug by overallocating the output buffer memory and offsetting the output buffer supplied to the function.

Revert this commit when this bug is fixed.

Fixes #51468
closed 2022-10-20 by stephanosio