zephyrproject-rtos/zephyr #51471: Network protocol MQTT: When qos=1, there is a bug in the subscription and publication

maweicoltsmart, 2022-10-20:
This problem only exists when connecting to a specific broker。

The broker address I tested is:
mqtt.dev.brightclass.cn:1883
User: liveclass
Password: bGl2ZWNsYXNz

We subscribe to a topic, such as:/testsub

We subscribe to a topic, such as/testsub, with the qos set to 1. At the same time, use mqtt.fx to subscribe to this topic. The qos is also set to 1.
Use mqtt.fx to publish any message with QoS of 1 to this topic. Our board and mqtt Fx can subscribe to this message. At first, none of this seemed to be a problem. However, the problem occurred after mqtt. fx published 20 messages. Our board could no longer receive these messages, but mqtt Fx can. After the problem occurs, I use wireshark to capture packets and see that the connection between our board and the broker is still correct, and the message that our board sends to the broker can be sent without error, but the message that mqtt.fx sends to the broker does not send to our board. It can be determined that the message published by mqtt. fx has been received by the broker, because mqtt. fx has subscribed to this topic at the same time and can successfully receive this message.

laurenmurphyx64, 2022-10-25:
Hi @maweicoltsmart, thanks for filing this issue. Is it going to be OK that you posted your username, password and server address? If you need this information removed, editing the description won't work as the history of the comment is viewable by anyone. Instead, you would need to ask me and I would have to ask our GitHub admins to see if it is possible for them to delete an issue.

maweicoltsmart, 2022-10-26:
The user name and password are only for testing. When this problem is solved, the user name will become invalid.

-----------------------------------

rlubos, 2022-10-26:
> After the problem occurs, I use wireshark to capture packets

Could you please share the full Wireshark log then? 

Anyway, it's hard to help not knowing what device are you testing with, not seeing your application code etc. 
Could you please provide a minimal sample (ideally for `qemu_x86` which allows to reproduce the issue)?

carlescufi, 2022-11-02:
@maweicoltsmart  please provide the information required.

carlescufi, 2022-11-09:
Closing due to inactivity. @maweicoltsmart please reopen if this is still an issue.

closed 2022-11-09 by carlescufi