zephyrproject-rtos/zephyr #51475: Twister: Judge timeout as skipped

shuai1x, 2022-10-20:
**Describe the bug**
In twister test report, twister judges timeout as skipped.
The cases skipped due to timeout, such as:
tests/misc/print_format/printk.format 
tests/subsys/rtio/rtio_api/subsys.rtio.api.submit_sem
tests/subsys/pm/device_runtime_api/pm.device_runtime.api
tests/subsys/shell/shell/shell.core 
tests/subsys/portability/cmsis_rtos_v1/portability.cmsis_rtos_v1

**Logs and console output**
```
kernel.common.stack_sentinel.fatal
kernel.common
Time Taken: 0.0s
Skipped: Timeout
skipped
```

**Environment (please complete the following information):**
 - OS: Linux 
 - Toolchain - SDK 15.0

enjiamai, 2022-10-21:
This issue only happens when running twister with "--build-only and --test-only". 
When a testcase is timeout, it will be judge as "skipped". This is the output of the twister.json file for example:
```
{
    "environment":{
        "os":"posix",
        "zephyr_version":"v3.2.0-intel",
        "toolchain":"xcc"
    },
    "testsuites":[
        {
            "name":"tests/kernel/mutex/mutex_api/kernel.mutex",
            "arch":"xtensa",
            "platform":"intel_adsp_xxx",
            "run_id":"6eda5c69cf2d8b747dfb6691b5d2f2e6",
            "runnable":true,
            "status":"failed",
            "reason":"Timeout",
            "execution_time":"60.61",
            "log":............(skip)"
            "testcases":[
                {
                    "identifier":"kernel.mutex.mutex_lock_unlock",
                    "execution_time":"0.00",
                    "status":"skipped",
                    "reason":"Timeout"
                },
                {
                    "identifier":"kernel.mutex.mutex_priority_inheritance",
                    "execution_time":"0.00",
                    "status":"skipped",
                    "reason":"Timeout"
                },
                {
                    "identifier":"kernel.mutex.mutex_reent_lock_forever",
                    "execution_time":"0.00",
                    "status":"skipped",
                    "reason":"Timeout"
                },
                {
                    "identifier":"kernel.mutex.mutex_reent_lock_timeout_pass",
                    "execution_time":"0.00",
                    "status":"skipped",
                    "reason":"Timeout"
                },
                {
                    "identifier":"kernel.mutex.mutex_recursive",
                    "execution_time":"0.00",
                    "status":"skipped",
                    "reason":"Timeout"
                },
                {
                    "identifier":"kernel.mutex.mutex_reent_lock_no_wait",
                    "execution_time":"0.00",
                    "status":"skipped",
                    "reason":"Timeout"
                },
                {
                    "identifier":"kernel.mutex.mutex_reent_lock_timeout_fail",
                    "execution_time":"0.00",
                    "status":"skipped",
                    "reason":"Timeout"
                }
            ]
        }
    ]
}

```

By the way, I cannot reproduce this in QEMU platform. It seems like this issue only happens on "binary handler" with using "--build-only and --test-only".

smrtos, 2022-10-25:
Trying to reproduce it.

closed 2023-01-18 by github-actions[bot]