zephyrproject-rtos/zephyr #51476: Please add documentation or sample how to use TLS_SESSION_CACHE socket option

boaks, 2022-10-20:
I tried to enable the [TLS_SESSION_CACHE](https://docs.zephyrproject.org/apidoc/latest/group__secure__sockets__options.html#ga16943eab0c13effcbdef684cc613ee04) for TLS on a nRF9160.

Therefore I set the socket option on the client socket. Add the end of the request, the connection is closed.
I would expect, that on the next connection, if the TLS session context is cached, I would be able to execute an abbreviated handshake as in [RFC5246, 7.3., Figure 2](https://www.rfc-editor.org/rfc/rfc5246#page-37) but the new `Client_Hello` never contains the session id and so a full handshake is always executed .
 
I searched the web without success.
I searched samples without success.

Would it be possible, that either an sample is added, or if I overseen it, point me to?
Or add some documentation how it must used. 
Sure, the server may send a session id but rejects to support session resumption. 
But the first step is the follow up `Client_Hello` with session id, and that's what I miss.

rlubos, 2022-10-20:
> I tried to enable the [TLS_SESSION_CACHE](https://docs.zephyrproject.org/apidoc/latest/group__secure__sockets__options.html#ga16943eab0c13effcbdef684cc613ee04) for TLS on a nRF9160.

Please clarify if you use native TLS implementation from Zephyr on top of nRF9160 socket, or rather fully offloaded modem side TLS (default configuration)?

The TLS_SESSION_CACHE option is rather straightforward to use, just enable it on a socket, and the TLS implementation will take care of storing the session. I've used the `http_get` sample with the following changes to verify the functionality:
```
diff --git a/samples/net/sockets/http_get/src/http_get.c b/samples/net/sockets/http_get/src/http_get.c
index 475f848c35..dca9ad1697 100644
--- a/samples/net/sockets/http_get/src/http_get.c
+++ b/samples/net/sockets/http_get/src/http_get.c
@@ -87,6 +87,7 @@ void main(void)
 
        dump_addrinfo(res);
 
+       for (int i = 0; i < 2; i++) {
 #if defined(CONFIG_NET_SOCKETS_SOCKOPT_TLS)
        sock = socket(res->ai_family, res->ai_socktype, IPPROTO_TLS_1_2);
 #else
@@ -104,6 +105,11 @@ void main(void)
 
        CHECK(setsockopt(sock, SOL_TLS, TLS_HOSTNAME,
                         HTTP_HOST, sizeof(HTTP_HOST)))
+
+       int cache = TLS_SESSION_CACHE_ENABLED;
+
+       CHECK(setsockopt(sock, SOL_TLS, TLS_SESSION_CACHE,
+                       &cache, sizeof(cache)))
 #endif
 
        CHECK(connect(sock, res->ai_addr, res->ai_addrlen));
@@ -130,4 +136,5 @@ void main(void)
        printf("\n");
 
        (void)close(sock);
+       }
 }
```

The second handshake seems to work as expected, i. e. Session ID is included:
[image]

But this is TLS socket implementation from Zephyr. If you use modem-side TLS on nRF91, although it should work the same way, there might be some bug modem-side, if it doesn't work this way please report the issue on https://devzone.nordicsemi.com/ so that it reaches modem developers.

Now as for the example usage, I think the option could be added to `big_http_download` sample as it already works in a way that download is restarted after the previous one finished, but there seems to be some issue with the server the sample used, it's no longer responsive. I need to find a new one that allows to download files over TLS.

boaks, 2022-10-20:
Thanks for the response!

I use NCS v2.1.0, nRF9160 SICA B1A, mfw_nrf9160_1.3.2 with a prj.conf

```
CONFIG_NETWORKING=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_OFFLOAD=y
CONFIG_NET_NATIVE=n
```

Not sure, which mode that is. 
(For DTLS I use my own tinydtls port, therefore I'm not too common with that modes.)
 
About your snippet.

```
CHECK(setsockopt(sock, SOL_TLS, TLS_HOSTNAME, HTTP_HOST, sizeof(HTTP_HOST) ))
```

NCS v2.1.0 uses in [nrf/samples/nrf9160/https_client](https://github.com/nrfconnect/sdk-nrf/blob/main/samples/nrf9160/https_client/src/main.c#L133)

```
setsockopt(fd, SOL_TLS, TLS_HOSTNAME, HTTPS_HOSTNAME, sizeof(HTTPS_HOSTNAME) - 1);
```

Is that `-1` relevant?

rlubos, 2022-10-20:
> Not sure, which mode that is.

Using native TLS with nRF91 needs some runtime configuration, so it seems that you use modem-side TLS.

> Is that -1 relevant?

Actually, for this specific option the optlen is ignored as the option expects NULL-terminated string (the `optval` is passed as is driectly to mbed TLS). So in this case `sizeof(HTTP_HOST)` would be the right choice, but it makes no real difference what you provide as optlen.

boaks, 2022-10-20:
> so it seems that you use modem-side TLS.

So the nordic forum is the right place?

rlubos, 2022-10-20:
Yes

boaks, 2022-10-21:
I did some more analysis, but end up in more detail question, where I was not able to find some docu:

Using:

```
fd = socket(AF_INET, SOCK_STREAM, IPPROTO_TLS_1_2);
```

Does this ensure, that **at least** 1.2 is used, so 1.3 will also be possible (that's my impression)
Or does it mean, that **exactly** 1.2 is used?

Also:

Is the SESSION_CACHE feature limited to 
a resumption handshake as of RFC5246 mentioned above?
Or does it also support  session tickets as of RFC 5077?

boaks, 2022-10-21:
> the server the sample used, it's no longer responsive. 

Just to mention:
Using a S3 offer (AWS, Gcloud, Azure, DigitalOcean, ExoScale) makes it very easy to provide files via https. 
It comes with low or moderate costs.

You may also contact other open source project (e.g. [Eclipse/Hawkbit](https://www.eclipse.org/hawkbit) ), if they are interested in hosting such a file in order to verify, that the zephyr community is able to use `Hawkbit`.

boaks, 2022-10-21:
Thanks a lot. I opened also ticket in the nordic devzone, either I sticked to deprecated documentation of the nrfxlib, or that may be updated.

[devzone issue](https://devzone.nordicsemi.com/f/nordic-q-a/93106/nrf9160-mfw-1-3-2-nrfxlib-2-1-0---additional-tls-details) 

FMPOV, we can close this issue.

closed 2022-10-21 by rlubos