zephyrproject-rtos/zephyr #51505: drivers: modem: gsm: gsm_ppp_stop() does not change gsm->state

jeppenodgaard, 2022-10-21:
**Describe the bug**
`gsm_ppp_stop()` does not set `gsm->state` to `GSM_PPP_STOP`.

Please also mention any information which could help others to understand
the problem you're facing:
I want to restart gsm_ppp by calling `gsm_ppp_stop()` then `gsm_ppp_start()`. However `gsm_ppp_start()` does not run because `gsm->state` is not `GSM_PPP_STOP`.

**To Reproduce**
Steps to reproduce the behavior:
When gsm_ppp is in `gsm->state` GSM_PPP_SETUP_DONE call `gsm_ppp_stop()` then `gsm_ppp_start()`.

**Expected behavior**
`gsm_ppp_start()` runs normally.

**Impact**
It is not possible to start gsm_ppp again if e.g. NO CARRIER is received.

**Logs and console output**
```
[00:01:54.403,503] <inf> modem_gsm: GSM network interface down
[00:01:54.403,594] <err> modem_gsm: gsm_ppp is already started
[00:03:04.419,921] <wrn> modem_gsm: Failed locking modem cmds!
[00:03:04.419,921] <err> modem_gsm: gsm_ppp is already started
[00:04:14.436,279] <wrn> modem_gsm: Failed locking modem cmds!
[00:04:14.436,309] <err> modem_gsm: gsm_ppp is already started
[00:05:24.452,636] <wrn> modem_gsm: Failed locking modem cmds!
[00:05:24.452,636] <err> modem_gsm: gsm_ppp is already started

```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - v3.1.0 with cherry-pick from https://github.com/zephyrproject-rtos/zephyr/pull/48533

jeppenodgaard, 2022-10-30:
To me it looks like this line `gsm->state = GSM_PPP_STOP` is missing. Do you have any feedback @ycsin ?

ycsin, 2022-10-31:
@jeppenodgaard yes you are right, somehow the reset is missing from the FSM implementation PR.
In https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/modem/gsm_ppp.c#L1212, the state should be reset to `GSM_PPP_STOP`, would it be possible for you to submit a patch for this?

jeppenodgaard, 2022-10-31:
@ycsin Yes :) Created a PR.

closed 2022-11-01 by fabiobaltieri